davidrapan/ha-solarman #109: Battery SOH: Too many digits for Y-axis on sensor detail page

PV-Joe, 2024-08-22:
Just a small bug and probably easy to fix 😄

**Screenshots**
[image]

**Metadata:**
Version: v24.08.21

davidrapan, 2024-08-22:
That is python itself related and you can read more 'bout it here: https://docs.python.org/3/tutorial/floatingpoint.html

PV-Joe, 2024-08-23:
Oh well, never mind and thanks for explaining.
It has to remain ugly then 😂

frsantos, 2024-08-23:
Can it be rounded to 2 digits on read?

davidrapan, 2024-08-23:
Did you read the article? 😆 It is rounded, the precision error is not made during saving the value but by reading by HA from db. 😆

frsantos, 2024-08-23:
Ok, so round on presentation? I think you can set a display presentation on HA, but not sure if it affects here.

davidrapan, 2024-08-23:
> but not sure if it affects here.

It does not affects graph, just showed value.

frsantos, 2024-08-23:
> It does not affects graph, just showed value.

A pity

davidrapan, 2024-08-30:
IF solution exists it has to be implemented on the HA's side. Integration has no control over this particular thing.

closed 2024-08-30 by davidrapan