davidrapan/ha-solarman #113: Not possible to start the integration on my docker instance

distancerunner, 2024-08-27:
**Describe the bug**
I removed Stephens integration and copy&paste your integration to the custom_components directory.
I tried to add your integration via the web interface, than I faced two errors.
At first there is a debug line in api.py which thrown an exception -> I commented out this line, than this issue was "solved".
Then I came to the next issue: cannot import name 'DeviceInfo' -> is this because I'm on a docker instance?

**Attach the debug log**
`2024-08-27 10:53:24.027 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solarman: Exception importing custom_components.solarman.config_flow,
2024-08-27 10:54:17.124 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solarman.config_flow,
Traceback (most recent call last):,
  File "/usr/src/homeassistant/homeassistant/loader.py", line 779, in get_platform,
    cache[full_name] = self._import_platform(platform_name),
  File "/usr/src/homeassistant/homeassistant/loader.py", line 796, in _import_platform,
    return importlib.import_module(f"{self.pkg_path}.{platform_name}"),
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module,
    return _bootstrap._gcd_import(name[level:], package, level),
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed,
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked,
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module,
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed,
  File "/config/custom_components/solarman/__init__.py", line 14, in <module>,
    from .api import Inverter,
  File "/config/custom_components/solarman/api.py", line 250,
    ('succeeded.' if results[i] == 1 else f'attempts left: {attempts_left}{'' if attempts_left > 0 else ', aborting.'}'),
                                                                                                         ^,
SyntaxError: f-string: f-string: expecting '}',
2024-08-27 10:54:17.129 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solarman: Exception importing custom_components.solarman.config_flow,
2024-08-27 11:01:09.545 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solarman: cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py),
2024-08-27 11:01:36.505 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solarman: cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py),`

**Metadata:**
Version: HA 2023.3.0

**Additional context**
Add any other context about the problem here.

distancerunner, 2024-08-27:
I remove also 'DeviceInfo' from api.py, than I getting the next failures from discovery.py 
Something is really not working for me :(

```2024-08-27 11:22:13.437 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solarman.config_flow,
Traceback (most recent call last):,
  File "/usr/src/homeassistant/homeassistant/loader.py", line 779, in get_platform,
    cache[full_name] = self._import_platform(platform_name),
  File "/usr/src/homeassistant/homeassistant/loader.py", line 796, in _import_platform,
    return importlib.import_module(f"{self.pkg_path}.{platform_name}"),
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module,
    return _bootstrap._gcd_import(name[level:], package, level),
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed,
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked,
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module,
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed,
  File "/config/custom_components/solarman/__init__.py", line 15, in <module>,
    from .discovery import InverterDiscovery,
  File "/config/custom_components/solarman/discovery.py", line 109,
    (f'attempts left: {attempts_left}{'' if attempts_left > 0 else ', aborting.'}'),
                                                                    ^,
SyntaxError: f-string: f-string: expecting '}',
2024-08-27 11:22:13.441 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solarman: Exception importing custom_components.solarman.config_flow,```

distancerunner, 2024-08-27:
Another Update from Stephan's version:
1.5.1 -> I get also an error like above

2024-08-27 11:32:20.418 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform solarman.sensor: Platform not found (cannot import name 'SupportsResponse' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py)).

1.4.3 -> it is working well on my "old" HA Version

So I need to investigate my docker HA installation.

gedger, 2024-08-27:
Your version of HA is shown as 2023.3.0. I would suggest you need to update to the latest version.

davidrapan, 2024-08-27:
I know it's not anywhere stated but the integration is made for the "newest" version of HA.

_As you can see you will find progressively more issues with older versions such are missing parts of the HA dev framework or even used older version of python itself._

closed 2024-10-15 by github-actions[bot]