davidrapan/ha-solarman #119: Plugin has errors during setup for inverter: SUN-M80G4-EU-Q0

viertel, 2024-08-29:
Hello all, 
I installed the plugin with my Deye inverter: SUN-M80G4-EU-Q0
I am a little bit confused, which Inverter Profile i need to select. So i tried multipl, one after another for example: deye_strind or deye_hybrid. 

I always geht the error message inside of the integration menu: 
Error during setup:
Check the protocols

I am not sure which is the correct debug log and where to find it. I added the System log (Settings->System->Logs). Included are only parts that reference Solarman.

Furthermore, i am a little confused: I have not entered any credentials like Solarman App or inverter login. Did i miss something?
Greetings and thank you in advance.

[home-assistant_2024-08-28T15-38-26.048Z.log](https://github.com/user-attachments/files/16792913/home-assistant_2024-08-28T15-38-26.048Z.log)

gedger, 2024-08-29:
Looking at other threads the deye_2mppt.yaml has been used successfully by others with that inverter. You also have RO file system errors reported? Are you 100% sure the serial number, IP address etc are correct in the configuration? I'd also suggest turning debug on and getting a full log.

davidrapan, 2024-08-29:
Hi @viertel,

I see several issues:
1. OSError: [Errno 30] Read-only file system
  **Solution - run: `ha core rebuild`**
  Solution from: https://github.com/davidrapan/ha-solarman/discussions/63
2. You enabled passtrough mode: inverter_passthrough': True
  **Which is for Ethernet Stick Loggers**

davidrapan, 2024-09-02:
Duplicate of #39

viertel, 2024-09-10:
Hello all, 
thanks for all the answers. I will try your solutions at the next weekend. 
Poorly, I did not have time before.

closed 2025-02-03 by davidrapan