davidrapan/ha-solarman #123: Add sensor translations - IT

davidrapan, 2024-08-30:


Edsol, 2024-09-19:
I can help with translations

davidrapan, 2024-09-21:
Hi @Edsol, I will soon commit translations for CS and DE so it will be more clear how to do it for other languages. 😉

PV-Joe, 2024-09-21:
> Hi @Edsol, I will soon commit translations for CS and DE so it will be more clear how to do it for other languages. 😉

Great!

Completing the descriptions in the attribute sections especially for configurations the in all languages would be awesome for one of the next steps.

lcavalli, 2024-10-17:
Time ago I made a preliminary translation, but never completed the work...  https://github.com/lcavalli/ha-solarman/blob/5fe5e211483ac11adcec2d37b1c780b1c8772097/custom_components/solarman/translations/it.json

davidrapan, 2024-12-23:
File w/ translations: [IT](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/translations/it.json)

Can someone verify newly added strings?

gmcinalli, 2026-02-12:
@davidrapan Are the translations working? I've my HA in on italian language but every entity (friendly names) in your integration is in english, maybe they're not in the current stable release?