davidrapan/ha-solarman #124: Add sensor translations - PL

davidrapan, 2024-08-30:


szyx22, 2024-09-03:
How?  By translating for example  inverter_definitions/sofar_lsw3.yaml file?

davidrapan, 2024-09-03:
Hi @szyx22, I will soon commit translations for CS and DE so it will be more clear how to do it for other languages. 😉

adamusekk, 2024-10-29:
I can help with translations...

davidrapan, 2024-11-29:
Keep Alive

davidrapan, 2024-12-23:
File w/ translations: [PL](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/translations/pl.json)

Can someone verify already translated strings?

aLAN-LDZ, 2025-01-14:
I'll try to look at the translations today

pedroza26, 2025-03-13:
As native I can confirm that overall it's very well translated. :) 

I made some review and I found few lines where special character were left unchanged. In most of the code I see special characters encoded but for below lines it's not. I am not sure if it's going to cause some issues or not. 

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L24

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L25

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L35

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L60

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L61

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L84

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L87

https://github.com/davidrapan/ha-solarman/blob/b50b20602097b39154cb1683fd7db6a065cd16a5/custom_components/solarman/translations/pl.json#L102

Others lines:
- 111
- 114
- 117
- 120
- 123
- 126
- 129
- 132
- 135

davidrapan, 2025-03-14:
Hi @pedroza26, thank you for your review! I don't think those special characters are even needed. But tell me, does it display correctly in Home Assistant?

Invi017, 2025-03-17:
Hi everyone. I can confirm, that the special characters are displayed correctly.