davidrapan/ha-solarman #125: Add sensor translations - PT

davidrapan, 2024-08-30:


davidrapan, 2024-09-30:
.

davidrapan, 2024-09-30:
Keep Alive

davidrapan, 2024-11-01:
Keep Alive

davidrapan, 2024-12-05:
Keep Alive

route66por, 2025-02-11:
Great job. Do you need any help on translate sensors to PT? How can i help?

davidrapan, 2025-03-02:
Hi @route66por, for start, you can try to validate translations in [pt-BR.json](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/translations/pt-BR.json)

And then maybe try to provide some more translations for some other entities. 😉

route66por, 2025-03-02:
> Hi [@route66por](https://github.com/route66por), for start, you can try to validate translations in [pt-BR.json](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/translations/pt-BR.json)
> 
> And then maybe try to provide some more translations for some other entities. 😉

I read them and they all ok.