davidrapan/ha-solarman #136: Solis Configuration (times) not working

pyriel800, 2024-09-06:
In Solis Hyrbid inverter definition there is a difference in the Time Start Charge and Discharge sensors:

```
- name: "Timed Charge Start"
        platform: time
        state_class: measurement
        scale: 1
        rule: 9
        registers: [43143, 43144]
        icon: "mdi:wrench-clock"

      - name: "Timed Charge End"
        class: time
        state_class: "measurement"
        scale: 1
        rule: 9
        registers: [43145, 43146]
        icon: "mdi:wrench-clock"

      - name: "Timed Discharge Start"
        class: time
        state_class: "measurement"
        scale: 1
        rule: 9
        registers: [43147, 43148]
        icon: "mdi:wrench-clock"

      - name: "Timed Discharge End"
        class: time
        state_class: "measurement"
        scale: 1
        rule: 9
        registers: [43149, 43150]
        icon: "mdi:wrench-clock"
```
The state class is differently defined.

The initial implementation results in the following items in HA

[image]

Which are in the sensors category not the configuration items

[image]

Other than this, the integration is awesome and thank you very much for what you have done!

davidrapan, 2024-09-06:
Hi @pyriel800, thank you for your bug report!

[fix: Solis Hybrid platforms for Timed entities](https://github.com/davidrapan/ha-solarman/commit/fbca1c3de6e8c3c951370fd72dccb4f6ac5051d2)

**And what about that Timed Discharge Current entity? I see it doesn't have a value..**

pyriel800, 2024-09-06:
Yes, this is also empty. I'm not sure if it is configured or just using the BMS to determine the value. It seems dynamic in the system - when I discharge I peak at ~4.9kW but it drops after a short while

davidrapan, 2024-09-06:
It should be configurable value the same way as Timed Charge Current. You should be able to set Charging/Discharging current limit with them...

But I don't see an error in its entity description in the file... so I'm clueless why it wasn't able to get currently set value.

pyriel800, 2024-09-06:
This is what I see for that entity:
[image]
 Is there anything else you need me to get out of the system?

For the timed sensors that are not working this is the entity page:
[image]

pyriel800, 2024-09-06:
It also looks like the times are being defined as sensors in the setup ie. sensor. not time.solis_timed_charge_end

[image]

davidrapan, 2024-09-06:
Try [v24.09.06](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.06) it will fix that entities with incorrect descriptions.

pyriel800, 2024-09-06:
This seems to have fixed it. All time entities are now showing correctly
[image]

I will read the disharge amps register using a spare Pi that I previously used for some CLI debugging/querying. I'm not worried about that parameter but it would be good to understand what is going on there :)

Thank you very very much for looking into this!

davidrapan, 2024-09-06:
You can also try to read register: 43142 using integration service: Solarman: Read Holding Registers (Modbus Function Code 3) and see if it will give you a value or...

pyriel800, 2024-09-08:
> You can also try to read register: 43142 manually using integration

Which finally I have done and it returns the following:

sensor.solisdischamps:
  '43142': 1000

davidrapan, 2024-09-08:
Okay so now it makes sense. According to the returned value it says it's set to 100 A but docs says that the max value can be only 70 A (and that's how it's set in the profile so because it gave the value out of range it's shown as empty).

So are the docs wrong?

pyriel800, 2024-09-08:
Which docs? Are they official Solis docs?

I think I configured by discharge output to 100A to maximise the potential
of my battery stack. The BMS actually reports an output rating of 180A but
that's a little too high (and the inverter is 5k!).

I suspect that the docs that you're referring to are incorrect in terms of
what is allowed.

davidrapan, 2024-09-08:
https://github.com/davidrapan/ha-solarman/wiki/Documentation

> I suspect that the docs that you're referring to are incorrect in terms of
what is allowed.

Could be.. or at least dated.

davidrapan, 2024-09-08:
Could you maybe then try to find out what is the max value inverter let's you put in?

pyriel800, 2024-09-08:
These are the values from the app/website for allowed input. I can try exceeding that when I get back home but suspect it will error on write.

davidrapan, 2024-09-08:
I increased the range to 180 A for now in: [fix: Minor bugs in inverter profiles](https://github.com/davidrapan/ha-solarman/commit/d1ec75278df31e0c3a3364a323826076d69995b8)

closed 2024-09-24 by davidrapan