davidrapan/ha-solarman #138: Component using native unit of measurement

Soltinioo, 2024-09-08:
**Describe the bug**
using native unit of measurement

**Attach the debug log**
'Rejestrator: homeassistant.components.sensor
Źródło: components/sensor/__init__.py:569
integracja: Sensor (dokumentacja, Problemy)
Pierwsze zdarzenie: 15:14:13 (1 zdarzenia)
Ostatnio zalogowany: 15:14:13

Entity sensor.falownik_deye_output_active_power (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues'

'Rejestrator: homeassistant.components.sensor
Źródło: components/sensor/__init__.py:731
integracja: Sensor (dokumentacja, Problemy)
Pierwsze zdarzenie: 15:14:13 (2 zdarzenia)
Ostatnio zalogowany: 15:14:13

Entity sensor.falownik_deye_output_active_power (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using native unit of measurement 'W' which is not a valid unit for the device class ('energy') it is using; expected one of ['MWh', 'Wh', 'kWh', 'GJ', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues
Entity sensor.falownik_deye_output_reactive_power (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using native unit of measurement 'VAR' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues'

Version: lastest 

I use deye string

davidrapan, 2024-09-08:
Hi @Soltinioo, thanks for the report it will be fixed promptly. 😉

davidrapan, 2024-09-08:
Fixed in [fix: Minor bugs in inverter profiles](https://github.com/davidrapan/ha-solarman/commit/d1ec75278df31e0c3a3364a323826076d69995b8)

closed 2024-09-08 by davidrapan

Soltinioo, 2024-09-09:
Hi. one error not corrected and is in the logs

'Rejestrator: homeassistant.components.sensor
Źródło: components/sensor/__init__.py:731
integracja: Sensor (dokumentacja, Problemy)
Pierwsze zdarzenie: 09:25:25 (1 zdarzenia)
Ostatnio zalogowany: 09:25:25

Entity sensor.falownik_deye_output_reactive_power (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using native unit of measurement 'VAR' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha-solarman/issues'

davidrapan, 2024-09-09:
[fix: Incorrect entity descriptions: state_class](https://github.com/davidrapan/ha-solarman/commit/1878ef7fe1ea7204ca91056e09aa090a166f3fc8)

frsantos, 2024-09-09:
Hey @davidrapan A couple of things:

* First, beware of the revert in the name of the kstar profile
* Do we really want to use a total_increasing in the total amount? The `total_increasing` is thought for amounts that can reset periodically (daily, weekly, monthly), not for monotonous increasing values: https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

gedger, 2024-09-09:
From my understanding the total_increasing can be reset to zero but it doesn't have to be, so continuously increasing for ever is OK too. If your inverter failed and was replaced then this would go to zero which is one of the use cases mentioned.

frsantos, 2024-09-09:
OK, this is more up to date: https://developers.home-assistant.io/blog/2021/09/20/state_class_total/

davidrapan, 2024-09-09:
[fix: Incorrect entity descriptions: state_class](https://github.com/davidrapan/ha-solarman/commit/b051e26e8194f78105e1c10f316c0f168b132d76)

frsantos, 2024-09-09:
> When I looked into it the total is intended for net meters which can also go into negative numbers.

Yes, sorry, the first blog post was misleading to me.

davidrapan, 2024-09-09:
> Yes, sorry, the first blog post was misleading to me.

Yeah I had the same issue before.