davidrapan/ha-solarman #140: Custom Sensors

wifi75, 2024-09-09:
Hi, I wanted to try the castom sensors, but where do I add them?

davidrapan, 2024-09-09:
Hi @wifi75, what you are trying to achieve?

wifi75, 2024-09-09:
this: 
I don't understand where the code should be copied..

```
- name: Power losses
  description: Includes consumption of the inverter device itself as well AC/DC conversion losses
  class: power
  state_class: measurement
  uom: W
  rule: 1
  digits: 0
  registers: [0x024E, 0x02A0, 0x02A1, 0x02A2, 0x02A3, 0x027C, 0x02B6]
  uint: enforce
  sensors:
    - signed:
      registers: [0x024E]
    - registers: [0x02A0]
    - registers: [0x02A1]
    - registers: [0x02A2]
    - registers: [0x02A3]
    - operator: subtract
      signed:
      registers: [0x027C, 0x02B6]
```

davidrapan, 2024-09-09:
So what inverter profile are you using?