davidrapan/ha-solarman #142: My Country code keeps on switching.. Strange effects

sanderbaks, 2024-09-10:
**Describe the bug**
I own a Sofar 12KTLX-3g
Serial Number of the device: SS2ES112N3U010
I use the Solarman WiFi Logger LSW-3
Firmware: LSW3_15_270A_1.68
Serial Number starts with: 238722

I can Acces my logger using the Deye_String and the Sofar LSW3 files.
All other files show an error.

For both files I see data, but none of them make sense.
I focus on the LSW3 config file.

My Country code keeps changing between Germany and Greece. This also affects the rest of the output.
So, I try to change some register addresses. I focus on 1 to check the results: PV Power 1. In the LSW3 confic file set to 0x000A
Changing it to [0x0586] (taken from the AZZURRO KTL3 config) immediately breaks the config file. Changing it back fixes the connection.

What am I doing Wrong?

**Attach the debug log**

I am not sure what log file to copy or to attach. Please help :)

**To Reproduce**

Just tet it up and change the config files.

**Expected behavior**
I expect the values to reproduce valid numbers.

**Screenshots**
See attached screenshot
[image]

**Metadata:**
24.09.06

davidrapan, 2024-09-10:
I would suspect that maybe the register is showing something different than coutry?

sanderbaks, 2024-09-10:
Indeed, but if I change a register to another the connections breaks :(

So the problem I face is how do I changes registers in the YAML, without breaking the connection..
It seams so strange.

davidrapan, 2024-09-10:
Did you tried sofar_tlx-g3.yaml profile? I believe that one is specific for your device. 😉

sanderbaks, 2024-09-10:
Yes I tried all of them, The Sofar TLX does not connect, please see screenshot
[image]

Essentially none of them connect, except for the Deye String and the Sofar Wifi

davidrapan, 2024-09-10:
> If it won't work. Enable debug logging and attach the file here.

Please.

sanderbaks, 2024-09-10:
Here is the Log file..

I really appreciate the work you are doing a lot.

[home-assistant_solarman_2024-09-10T10-29-50.199Z.log](https://github.com/user-attachments/files/16943499/home-assistant_solarman_2024-09-10T10-29-50.199Z.log)

davidrapan, 2024-09-10:
Hmmmm

@githubDante? Could you please take a look at this?

githubDante, 2024-09-10:
Hi @davidrapan

Probably yes. The response looks like a variation of the resposne in the KeyError issue, but this time the CRC is invalid. The exception code here is 2 (Illegal Data Address).

sanderbaks, 2024-09-11:
Sounds interesting :). Why does is happen with all the different  config files except 2 ?

davidrapan, 2024-09-11:
@githubDante, thanks!

> Sounds interesting :). Why does is happen with all the different config files except 2 ?

Cause the file contains different registers, but also not the correct ones I'm sure about that as it produces nonsense data.

You will need new file which will be based on the sofar_tlx-g3.yaml but with some of the registers left out.

sanderbaks, 2024-09-11:
But, when I tried to comment out all the registers, it also gave me an error..
is it not the way to proceed ? and find the one register that breaks it?

sanderbaks, 2024-09-11:
> The file is very sensitive to proper spacing so you maybe introduced errors to the file itself and that's why it did not work.
> So according to the log this 0x0404 - 0x042B range is causing issues so I would focus on registers from that range at first.

Awesome, ill start working on this and report back the results!Just curious: is the wifi logger producing the data, and creating the format of it. Or is it the inverter, and is the logger just transferring 1:1? Or even a combination?----------------Sander BaksCommercial & editorial photographerwww.sanderbaks.comwww.hotelcreatives.com

davidrapan, 2024-09-11:
You can also try to add `min_span: 1` into this block so it looks like this:
```
default:
  update_interval: 5
  min_span: 1
  digits: 6
```
That makes it to stop producing register ranges and will ask for specific ones and that will help you locate the issue faster.

Attach log here again after this change and I'll look at it. 😉

davidrapan, 2024-09-11:
Logger is formatting the data and uses solarman proprietary protocol. 😉

sanderbaks, 2024-09-11:
So, this is my YAML:

# Sofar 3.3-12KTLX-G3
# Note: no modbus energy meter connected

info:
  manufacturer: Sofar
  model: TLX-G3

default:
  update_interval: 5
  min_span: 1
  digits: 6

parameters:
  - group: PV
    items:
      - name: "PV1 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 1
        registers: [0x0586]
        icon: "mdi:solar-power-variant"

Nothing else.  It breaks...

Using Register 0x000A gives me the value of 30.

Do you guys know any register that should give me a value that I can verify ?

davidrapan, 2024-09-11:
First of all if you put it here in this format I have no way of determining if it's correct at all cause of different formatting.

And if at least the spacing is "matching" with your file then you have it wrong so I'm not surprised it's not working. 😉

sanderbaks, 2024-09-11:
here is the log file :)
[Uploading home-assistant_solarman_2024-09-11T12-56-12.708Z.log…]()

sanderbaks, 2024-09-11:
[home-assistant_solarman_2024-09-11T12-56-12.708Z.log](https://github.com/user-attachments/files/16964328/home-assistant_solarman_2024-09-11T12-56-12.708Z.log)

one more time :)

davidrapan, 2024-09-11:
So try to remove all entities with registers in this range: 0x0404 - 0x0410

And show me to log again. 😉

sanderbaks, 2024-09-11:
Here it is! 
[home-assistant_solarman_2024-09-11T15-21-44.795Z.log](https://github.com/user-attachments/files/16965831/home-assistant_solarman_2024-09-11T15-21-44.795Z.log)

davidrapan, 2024-09-11:
There are still entities w/ registers within the range: 0x040A - 0x0410

sanderbaks, 2024-09-11:
All gone now :)

[home-assistant_solarman_2024-09-11T15-44-29.566Z.log](https://github.com/user-attachments/files/16966138/home-assistant_solarman_2024-09-11T15-44-29.566Z.log)

davidrapan, 2024-09-11:
Let's continue: 0x0418

sanderbaks, 2024-09-11:
You are aware of the fact that the first line "0x0586" already breaks it? That is what I tried to say a this morning.
But here it is!

[home-assistant_solarman_2024-09-11T16-00-46.672Z.log](https://github.com/user-attachments/files/16966333/home-assistant_solarman_2024-09-11T16-00-46.672Z.log)

sanderbaks, 2024-09-11:
(I mean, don't get me wrong, you know a lot more then I do, but I used to be a programmer myself.  (C, C++,RISC and SQL)

sanderbaks, 2024-09-11:
But i 10000% appreciate your time and effort to help me!! .. If I can make a donation somewhere ?

davidrapan, 2024-09-11:
Then you know the drill. Remove all those which are causing issues. 😉

sanderbaks, 2024-09-11:
Just for my understanding, Should I not be removing from the top down? I am sure that the top one is breaking it. Now I dont know if everything below that would be ok..

Or is that not the way it works? That I am not sure of.

sanderbaks, 2024-09-11:
Removed all of them one by one, I have the feeling I can not get of the first line.. Will now try from the start, removing from the top until I find the breaking error.

davidrapan, 2024-09-11:
It always sorts the registers from lowest to highest and then does the requests. So the order in the file does not matter. But if you what registers breaks it for sure you can already remove them. 😉

davidrapan, 2024-09-11:
But I'm kinda afraid that not even this file has any correct registers set for your device... :-/

sanderbaks, 2024-09-11:
Ah! Did not know that, that is valuable.
And, I am working the the TLX-G3 file right?
My inverter is the 12KTLX-G3

Just double checking :) none of the registers seem to work.

sanderbaks, 2024-09-11:
> But I'm kinda afraid that not even this file has any correct registers set for your device... :-/

I agree :(

davidrapan, 2024-09-11:
We had many examples before that now and then some registers were causing issues for some particular devices and removing them helped but maybe it's not the case with you.

> And, I am working the the TLX-G3 file right?
My inverter is the 12KTLX-G3

Yeah that means it should be file for your device (12K just means 12000 so we are not including the 'K' in the name)

sanderbaks, 2024-09-11:
Its just a strange issue..  I do have connection.. that is the frustrating bit.
Should I try the LAN logger? Its a different Serialnumber.  
If the result is the same we can eliminate that.

sanderbaks, 2024-09-11:
Anyway, enough for today, I will fiddle on a bit and report back!
Thanks so far! Have a great evening!

davidrapan, 2024-09-11:
It won't help stick just provides access to the registers and does not sets them.

davidrapan, 2024-09-11:
Hi @Floppe, I think you added this file right? I'm no looking at it and you named it as KTLX-G3 (stated as such also in the comment in the file) but in the PR you wrote: "Added new Sofar KTLX-G2 inverter." so for which version it is then?

Floppe, 2024-09-11:
Oh. It should be KTLX-G3.

davidrapan, 2024-09-11:
Mhm, that's weird then.. It doesn't work for our friend @sanderbaks here...

sanderbaks, 2024-09-12:
Hi!  It is me again :)

I created a less complex lookup file better be able to see what is happening. I am trying to understand the values I see and to manually match them.  The strange thing is: They change in a funny way. Please see the screen shots. With your knowlegde, what can be the reason for this? The logger can not keep up? In the mean time I did not change anything, I just keep on looking at the screen.

[image]
[image]

Floppe, 2024-09-12:
PV2 voltage insane high on the first screenshot :)

I have the same logger firmware though. Does you logger have good wifi connection? Mine has signal quality only 45% but still works okay.

sanderbaks, 2024-09-12:
> PV2 voltage insane high on the first screenshot :)
> 
> I have the same logger firmware though. Does you logger have good wifi connection? Mine has signal quality only 45% but still works okay.

The Connection is perfect. It is sitting right next to the access point. It is 100%

I am now trying values according to the attached document, nothing is breaking and I do see values. But I am trying to match them by eye now. Comparing them with the app values. There MUST be a setting that works :)

[SOFARSOLAR ModBus-RTU Communication Protocol.pdf](https://github.com/user-attachments/files/16976307/SOFARSOLAR.ModBus-RTU.Communication.Protocol.pdf)

sanderbaks, 2024-09-12:
What doe Rule:1 mean ? (ore 2 or 3 in that matter)

Floppe, 2024-09-12:
It's what kind of value type, unsigned/signed/date/time/raw/bits/version and so on. Don't remember where it was documented though. Perhaps from the original integration.

sanderbaks, 2024-09-12:
So, you have the same FW of logger, and your register 0x0586 is showing PV Power? This value breaks the integration directly. ..  I think if we solve this, we solve it all. 
I mean, Current is quite stable, Voltage and current  an the AC side is stable.  Is there a way to Search for these values?
Are they integers ? Or does the integration doe any calculations on them? Or conversions?

TEMP is also a nice value to look for I guess? Is that an Integer ? 

Then if i find a few values that match, I can search the internet for a matching list of other registers. And take it from there.

Floppe, 2024-09-12:
Yes, it shows. Here some statistics for PV1. We have many smalls clouds today here.

[image]

Here very short info how I built the yaml. Google for "sofar g3 external modbus" and you find better documented specs.

From modbus specs

[image]

```
      - name: "PV1 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 1
        registers: [0x0586]
        icon: "mdi:solar-power-variant"
```

Rule 1 because U16 - unsigned 16 (1 register and only positive values)
scale by 10 so it becomes correct value in Watt.
Or by 0.1 and use kW if you want that.

sanderbaks, 2024-09-12:
That register directly breaks the whole integration. .. So weird ? What is the difference with my Inverter and yours? The logger is the same FW, so that is not it!

sanderbaks, 2024-09-12:
Do you know this tool:  https://github.com/Nedel124/Sofar_G3_LSW3

Could this help ?

Floppe, 2024-09-12:
Well, I have 2 other devices on the same modbus. But I would like to think that have more negative than positive impact on this integration.

I have seen the tool, never tried it out.

sanderbaks, 2024-09-12:
What inverter do you have ?

See attached software versions .. 
[image]

sanderbaks, 2024-09-12:
Everywhere I search in the internet the specs say: 0x0586 = Power PV1.
In my case it directly breaks.

I am going crazy!

Floppe, 2024-09-12:
[image]
Took a while to get Solarman app working :)

sanderbaks, 2024-09-12:
Sorry, and to be sure you also have the SOFAR tlx-g3 inverter ?

Floppe, 2024-09-12:
[image]

Well, it's actually 8.8. But shouldn't it be the same only that 12 can handle more power.

sanderbaks, 2024-09-12:
Yes that is correct..  I am going to do a firmware upgrade

davidrapan, 2024-09-12:
So strange how your screenshots of basic info looks so different... Given the fact it's the same device (when we don't count the rated power).

sanderbaks, 2024-09-12:
true .. it is driving me mad .. I am trying their support now. But I am not sure if that will help..

sanderbaks, 2024-09-13:
So I contacted support. I was very sceptical. But they did a full update, remotely even!
Everything is working now. Execpt for the alerts. but that is a slightly different issue. I will open a new issue for that.

Please see the new version numbers that I have now. and everything is perfectly working.

[image]

sanderbaks, 2024-09-13:
And now also for you guys: @davidrapan @Floppe : You guys are the best.. You can not believe how much I appreciate this!

THANKS!!!

[6 more replies]

closed 2024-09-23 by sanderbaks