davidrapan/ha-solarman #143: Modbus Register System Work Mode

Mights001, 2024-09-10:
Hi, 
i add this Modbusregister under Work Mode Group for the Systemwork Mode, but it does not work and the Inverter is not Changing the Settings. The Register Number is the right one with Esphome it works.  

      - name: System Work Mode
        update_interval: 300
        platform: "select"
        state_class: ""
        uom: ""
        scale: 1
        rule: 1
        registers: [0x008E]
        options: ["Selling First", "Zero to Load", "Zero to CT"]
        lookup:
          - key: 0x0000
            value: "Selling First"
          - key: 0x0001
            value: "Zero to Load"
          - key: 0x0002
            value: "Zero to CT"

And i a also add this Registers for Solar Sell Settings. Can you add this to your Repo. because if is coming a update i lost this Registers and i wants to have also the latest changes from you, this is why i not want to make my own YAML 

      - name: "Max Power Solar Sell"
        update_interval: 300
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 1
        rule: 1
        registers: [0x008F]
        configurable:
          min: 0
          max: 15600
        icon: "mdi:transmission-tower-export"
        
      - name: "Max Solar Power"
        update_interval: 300
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 1
        rule: 1
        registers: [0x0154]
        configurable:
          min: 0
          max: 15600
        icon: "mdi:transmission-tower-export"

And one more in the deye_sg04lp3.yaml the Max. Batterie Current is now 185A can you change this in your Repo. to 240A because the most Inverters are 12KW 

THX

davidrapan, 2024-09-10:
Hey @Mights001, thanks for the info! I'll test it with my device and add it here.

toruonu, 2024-09-13:
I see that this has been committed to .11, but it's not pushed as a release and hence HACS doesn't recognize that there is an update. Can you please do that as the work mode register is really needed to make sure battery power can be exported to grid as well.

davidrapan, 2024-09-17:
[v24.09.16](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.16)

closed 2024-09-24 by davidrapan