davidrapan/ha-solarman #148: Deye Inverter connects but not getting any valid values

Milliamper, 2024-09-11:
The integration finds my Deye Hybrid Inverter (SUN-5K-SG04LP3-EU) on the network but it cannot read any valid values.
[image]
I have also tried using different yaml config files (deye_string.yaml and deye_hybrid.yaml) but none of them helped.
I also attached the log file output.
[home-assistant_solarman_2024-09-11T19-22-33.924Z.log](https://github.com/user-attachments/files/16969236/home-assistant_solarman_2024-09-11T19-22-33.924Z.log)

Version: 24.09.08

davidrapan, 2024-09-11:
You are using incorrect inverter profile.

You have to use inverter profile specific for your device which is deye_sg04lp3.yaml. 😉

Milliamper, 2024-09-11:
You were right :D
It is now perfect!

closed 2024-09-11 by Milliamper