davidrapan/ha-solarman #157: deye_4mppt.yaml (SUN-M160G4-EU-Q0) only showing production values for 1 and 2 (3+4 missing)

bmaciag, 2024-09-15:
**Describe the bug**
I use deye_4mppt.yaml (SUN-M160G4-EU-Q0) and all works fine except, that the values for production are missing for 3+4. The voltage etc for 1-4 is display but for production only 1-2.
[image]

CrazyUs3r, 2024-09-15:
try this yaml

____________________________________________________
#
# First version: 08.05.2022
# Latest update: 15.09.2024
# Microinverter SUN2000G3 (DEYE/VESDAS)
# v0.1  Added all the cool registers from Deye_Modbus.-.-.V118.pdf
# 4x MPPT, 4x inverter
# 1x Logger, 4x Module,
#

               
                                 
                                 
                                    
                         
                              
                              

default:
  update_interval: 5
  digits: 6

parameters:
  - group: PV
    items:
      - name: "PV1 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x006D]
        icon: "mdi:solar-power"

      - name: "PV2 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x006F]
        icon: "mdi:solar-power"

      - name: "PV3 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0071]
        icon: "mdi:solar-power"

      - name: "PV4 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0073]
        icon: "mdi:solar-power"

      - name: "PV1 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.1
        rule: 1
        registers: [0x006E]
[... 652 lines cut ...]
        platform: "switch"
        state_class: ""
        uom: ""
        scale: 1
        rule: 1
        registers: [0x002F]

                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           [... cut ...]                                                                                                                                                                                                                                                     
      # Over-frequency Load-shedding - On/Off Switch
      - name: "Over-frequency Load Reduction"
        update_interval: 300
        platform: "switch"
        state_class: ""
        uom: ""
        scale: 1
        rule: 1
        registers: [0x0031]

bmaciag, 2024-09-17:
I added this to a merge request

bmaciag, 2024-09-17:
Thank you @CrazyUs3r 

I changed it accordingly, one issue remains:
[image]

Seems like the register for total production 3 (0x004B) is wrong.

CrazyUs3r, 2024-09-17:
Try 4a

closed 2024-09-17 by bmaciag