davidrapan/ha-solarman #161: Integration is no longer working after update to core 2024.9.2

Islanad, 2024-09-17:
The integration no longer loads.
Restoring a full backup from a previous day has no effect.

[image]

Home Assistant now also takes very long to start up and I get this in the log:

Logger: homeassistant.bootstrap
Source: bootstrap.py:1053
First occurred: 16:27:05 (1 occurrences)
Last logged: 16:27:05

Setup timed out for stage 2 waiting on set() - moving forward

davidrapan, 2024-09-17:
Hi @Islanad, please attach log file with debugging enabled.

Islanad, 2024-09-17:
I can not enable debug logging anymore.
[image]

I tried to remove the device but get this error:

Logger: aiohttp.server
Source: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:433
First occurred: 17:20:04 (3 occurrences)
Last logged: 17:23:21

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 462, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 95, in delete
    result = await hass.config_entries.async_remove(entry_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1785, in async_remove
    await entry.async_remove(self.hass)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 868, in async_remove
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solarman/__init__.py", line 14, in <module>
    from .api import Inverter
  File "/config/custom_components/solarman/api.py", line 12, in <module>
    from pysolarmanv5 import PySolarmanV5Async, V5FrameError
ModuleNotFoundError: No module named 'pysolarmanv5'

davidrapan, 2024-09-17:
`ModuleNotFoundError: No module named 'pysolarmanv5'`

Uh it looks like that HA did not installed required dependencies for the integration.

Try to remove and then install back the integration.

Do restart of your HA before and also after the reinstallation.

Islanad, 2024-09-17:
When I try to add the integration again, I get stuck on:
[image]

davidrapan, 2024-09-17:
SSH into your HA instance and try 'ha core rebuild' maybe that will help? HA is for some reason refusing to install integration dependency.

Islanad, 2024-09-17:
already tried that multiple times. No effect.

Islanad, 2024-09-17:
I get the same for the other solarman integration
[image]

davidrapan, 2024-09-17:
I'm afraid that the issues is in your HA installation. Something did break inside which handles dependencies for python modules.

Islanad, 2024-09-17:
is there anything I can do now?

davidrapan, 2024-09-17:
I have to say that I totally forget the "after update to core 2024.9.2"...

_I also did the update just now and after restart everything came up as usual... so I'm really puzzled._

**But just to be clear you just did the update and after restart it did not work? Or you also did update or installation of the integration AFTER you did update of the core?**

Islanad, 2024-09-18:
I only updated core and after the restart it did no longer work. I then loaded a backup from before and it still did not work.

I now did a new installation of home assistant on a different SD card and loaded the backup from there.
Now it works again but I won't try to install the core update again. I wait until the next version.

Thanks for all your help.

closed 2024-09-18 by Islanad