davidrapan/ha-solarman #167: total losses not working

wifi75, 2024-09-22:
hello on deye_hybrid total losses not working 
at version 24.09.20

[image]

davidrapan, 2024-09-22:
Hi @wifi75, and other losses sensor does?

wifi75, 2024-09-22:
> Hi @wifi75, and other losses sensor does?

it is ok:

[image]

davidrapan, 2024-09-22:
And Today's Losses?

wifi75, 2024-09-22:
and this:

[image]

davidrapan, 2024-09-22:
Yes I already fixed in the repo that resetting to 0 from start of the day of the Today's Losses sensor...

But I really don't see any errors in the Total Losses entity description.. :-/

Could you please try to enable debug logging for couple of minutes and attach the file here? It proly won't tell anything but it's worth the shot.

wifi75, 2024-09-22:
here

[home-assistant_solarman_2024-09-22T08-27-09.125Z.log](https://github.com/user-attachments/files/17088808/home-assistant_solarman_2024-09-22T08-27-09.125Z.log)

davidrapan, 2024-09-22:
> Edit: And can you show me all your Total sensor values?

And also this please...

wifi75, 2024-09-22:
yes, no problem:

[image]

[image]

[image]

[image]

[image]

davidrapan, 2024-09-22:
Okay so it's because your values somehow does not makes sense...

It's:
**Import + Production + Discharge - Export - Charge - Load Consumption**

And for your case it results in:
**248.6 + 3900.4 + 1623.7 - 4007.6 - 1732.7 - 2179.2 = -2,146.8**

Which is not valid result and that's why it does not work. What looks strange to me is that really high export?

wifi75, 2024-09-22:
I don't know why I added the component and didn't do anything else, but I created some integrations, but don't take those into consideration, just look at the data from your sensors, those created by your component.
if you need anything else write to me

davidrapan, 2024-09-22:
I mean I told you in previous post why it's not working for you. It's not possible to calculate Total Losses from your values, **the result is negative number**.

davidrapan, 2024-09-22:
Hey @krasnoukhov, would you be willing to help us out a little? Could you please show me your Total sensor so I can maybe try to compare it with your values? Or maybe you could have an idea what is wrong here? Thanks!

davidrapan, 2024-09-22:
What is the difference between **Total Production** and **Total Grid Production**? Maybe they should be both factored in?

davidrapan, 2024-09-22:
It almost looks like it's production for every string separately..

terzo33, 2024-09-22:
Works for me

[image]

davidrapan, 2024-09-22:
@terzo33 can you please show me what values you have in all Total sensors? Thank you!

terzo33, 2024-09-22:
> @terzo33 can you please show me what values you have in all Total sensors? Thank you!

Sure

[image]

davidrapan, 2024-09-22:
Thanks.

So yeah your values makes way more sense... but you have on the other hand kinda low export, hm.

Anyway, do you have any idea what that **Total Grid Production** sensor shows?

wifi75, 2024-09-22:
**Hi, also this is not correct:**
[image]

show this from solarman cloud:
[image]

davidrapan, 2024-09-22:
@wifi75, could you please click into the combo box and show me all possible values? Thanks!

wifi75, 2024-09-22:
yes

[image]
[image]
[image]
[image]
[image]
[image]

[Parameters_2312028217__20240922.xlsx](https://github.com/user-attachments/files/17090440/Parameters_2312028217__20240922.xlsx)

davidrapan, 2024-09-22:
Uh.. I need you to show me all the options you have under System Work Mode in the Solarman App.

And best would be if you tried to set one by one all of them and then tell me to which items it corresponds to the value in the HA's integration cause I don't have your device... :-/

wifi75, 2024-09-22:
see file execel

davidrapan, 2024-09-22:
That file does not contain any new information for me.

I need you to try to set in the Solarman App every possible value in the System Work Mode. And then look into HA into Work Mode sensor attributes to which numeric value it corresponds to.

davidrapan, 2024-09-22:
Try to edit the **Work Mode** entity description in the `deye_hybrid.yaml` file with this:
```
      - name: Work Mode
        update_interval: 300
        platform: "select"
        rule: 1
        registers: [0x00F4]
        options: ["Export First", "Zero Export To Load", "Zero Export To CT"]
        lookup:
          - key: 0x0000
            value: "Export First"
          - key: 0x0001
            value: "Zero Export To Load"
          - key: 0x0002
            value: "Zero Export To CT"
```
And tell me if it changes anything.

wifi75, 2024-09-22:
any changes made in solarman have no influence on ha

davidrapan, 2024-09-22:
> any changes made in solarman have no influence on ha

They do for sure but yoy have to either wait some time or forcefully reload the integration.

**Try to change the file as instructed above!**

terzo33, 2024-09-22:
> any changes made in solarman have no influence on ha

Devi modificare il file yaml sulla cartella custom components

wifi75, 2024-09-22:
I waited a few minutes but nothing happened... how long should I wait?

davidrapan, 2024-09-22:
> I waited a few minutes but nothing happened... how long should I wait?

Did you looked into attributes??? **That displayed value don't have to change but you can see numeric value in attributes**.

wifi75, 2024-09-22:
> - name: Work Mode
>         update_interval: 300
>         platform: "select"
>         rule: 1
>         registers: [0x00F4]
>         options: ["Export First", "Zero Export To Load", "Zero Export To [...]

I changed it, restarted Ha, but it doesn't working

[image]

davidrapan, 2024-09-22:
It changed the entity type to selector. You will find it now under **Configuration**.

davidrapan, 2024-09-22:
I already made all the necessary changes, refactored incorrectly named entities + added some new.

[feat: Deye Hybrid Energy Pattern and Work Mode](https://github.com/davidrapan/ha-solarman/commit/29a6f3a259f382cbc38f534e887316a5e26d3ca5)

davidrapan, 2024-09-22:
[v24.09.22](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.22)

wifi75, 2024-09-22:
very nice! thank you a lot

davidrapan, 2024-09-22:
@wifi75, the thing is that the original issue you had is still unresolved and I don't really know why it's not possible to calculate losses from your values... :-/

terzo33, 2024-09-23:
I see always the same value. It is normal?

[image]

davidrapan, 2024-09-23:
Now: 600.2 + 513.8 + 255.8 - 0.3 - 995 - 267.2 = 107.3
Before: 599.1 + 508.8 + 245.3 - 0.2 - 981 - 264.2 = 107.8

The calculation oscillates because the inverter doesn't have enough precision or provides values out of sync. So there are counter measures in place which results in displaying the highest recorded value which is in your case 117.1 kWh.

**So yeah it looks okay in your case. It will increase when the result of the calculation will be higher than 117.1 kWh**.

terzo33, 2024-09-23:
Thanks for explaining it to me

wifi75, 2024-09-24:
> @wifi75, the thing is that the original issue you had is still unresolved and I don't really know why it's not possible to calculate losses from your values... :-/

the active power loss sensor works, it's the kWh losses that don't work

[image]

this not working
[image]

davidrapan, 2024-09-24:
Hey @wifi75, yeah I know and I already told you why it's not working here:

> Okay so it's because your values somehow does not makes sense...
> 
> It's: **Import + Production + Discharge - Export - Charge - Load Consumption**
> 
> And for your case it results in: **248.6 + 3900.4 + 1623.7 - 4007.6 - 1732.7 - 2179.2 = -2,146.8**

The result is negative thus displayed as **unavailable**.

wifi75, 2024-09-24:
it is negative because when the Deye inverter also recycles energy from the first inverter... I have two inverters, a solkaredg and a deye...
So when the deye is producing less than what it has to give to the battery, in this case if the first inverter (solaredge) has a surplus, the deye takes that energy and passes it to the battery.
in fact the sensopre power is negative

davidrapan, 2024-09-24:
So because this additional energy is not anyhow measured by Deye it's not possible to calculate **Total Losses**.

wifi75, 2024-09-24:
the deye knows that energy is coming, of course he knows it!

wifi75, 2024-09-24:
[image]

davidrapan, 2024-09-24:
But it's not reflected in the **Total \*** sensors from which are the losses calculated.

davidrapan, 2024-09-24:
**Could you draw me a pic how exactly you have this setup connected?**

wifi75, 2024-09-24:
> **Could you draw me a pic how exactly you have this setup connected?**

see below

[image]

wifi75, 2024-09-24:
look at this scenario, I lowered the charge to 70Ah so about 3.2Kw, as you can see the deye is producing much more than what the battery requires., so a part (about 320 watts) is exporting it together with the solaridge, another part is given to the load of the LOAD output (or also called Backup) as in the previous photo.

[image]

davidrapan, 2024-09-24:
I mean power sensors won't help us much in this case. **Total Losses (Energy losses)** are calculated from energy sensors. (Energy is power over time).

In other words it's nice that you can see it in the power sensors but if it's not also measured in some way in energy sensors you would have to measure it yourself using [Riemann sum integral](https://community.home-assistant.io/t/integration-riemann-sum-integral-helper/546395) over power sensors and then by Utility Meter over that integral.

davidrapan, 2024-09-24:
**If I understand your case correctly. Your Deye can see the total consumption of both inverters but can't see the production of your other inverter. That means you have to calculate the Losses yourself.**

wifi75, 2024-09-24:
ok don't worry, anyway it's not important come on, don't worry.
could you integrate the menu below?

[image]

davidrapan, 2024-09-24:
> could you integrate the menu below?

I'll look into it.

I can help you with that losses too I guess... Tell me do you have sensor from your other inverter which shows total produced energy? in kWh?

wifi75, 2024-09-24:
> **Se ho capito bene il tuo caso, il tuo Deye può vedere il consumo totale di entrambi gli inverter, ma non può vedere la produzione dell'altro inverter. Ciò significa che devi calcolare le perdite da solo.**

the deye, has a meter (CT) that is connected to the grid input, so the deye understands and knows if the inverter is producing, exporting etc etc

davidrapan, 2024-09-24:
> the deye, has a meter (CT) that is connected to the grid input, so the deye understands and knows if the inverter is producing, exporting etc etc

But it's not reflected in it's sensors so it does not matter if you think that. If this would be 100% true the equation would not result in negative number!

davidrapan, 2024-09-24:
**There is Total Production of second inverter missing in the equation.**

wifi75, 2024-09-24:
> **Nell'equazione manca la produzione totale del secondo inverter.**
yes it's true that one is missing...
but it's here:
[image]

davidrapan, 2024-09-24:
Great. That means you can create template sensor which can calculate it manually. The Total Losses sensor which is atached to the Deye inverter can only see his own sensor. It can't see sensor from other integration - device. You know?

_So you can create Template sensor which does following:_
\+ Deye Total Energy Import
\+ Deye Total Total Production
**\+ SolarEdge Produzione Solaredge........**
\+ Deye Total Battery Discharge
\- Deye Total Energy Export
\- Deye Total Battery Charge
\- Deye Total Load Consumption

Notice the **\+ SolarEdge Produzione Solaredge........** that is missing from the calcualtion to be in positive and thus correct state.

davidrapan, 2024-09-24:
[feat: Add Settings to Deye Hybrid profile](https://github.com/davidrapan/ha-solarman/commit/cbce4da9b1331be89159c589656f781bf4ea093b)