davidrapan/ha-solarman #168: Switch on/off for microinverter doesn't work

terzo33, 2024-09-22:
Hi, @davidrapan , I tried to switch off my microinverter but doesn't work, nothing action.

[image]

[image]

[image]

[image]

davidrapan, 2024-09-22:
Hi @terzo33, please enabled debug logging and toggle the switch few times with sufficient time separation between each toggle ~ 30 seconds and attach the log file. Thanks!

terzo33, 2024-09-22:
> Hi @terzo33, please enabled debug logging and toggle the switch few times with sufficient time separation between each toggle ~ 30 seconds and attach the log file. Thanks!

This is my log. Thanks

[image]

davidrapan, 2024-09-22:
Try to edit the switch entity description in the file you are using like this:
```
      - name: ""
        platform: "switch"
        state_class: ""
        uom: ""
        scale: 1
        rule: 1
        registers: [0x002b]
        value:
          on: 0x0001
          off: 0x0002
```
And tell me if it changes anything. 😉

terzo33, 2024-09-22:
> Try to edit the switch entity description in the file you are using like this:
> ```
>       - name: ""
>         platform: "switch"
>         state_class: ""
>         uom: ""
>         scale: 1
>         rule: [...]

In entity.py

[image]

davidrapan, 2024-09-22:
No in the inverter profile which you are using 2mppt or 4mppt.

terzo33, 2024-09-22:
> Try to edit the switch entity description in the file you are using like this:
> ```
>       - name: ""
>         platform: "switch"
>         state_class: ""
>         uom: ""
>         scale: 1
>         rule: [...]

Yesss works 😍😍

davidrapan, 2024-09-22:
Okay I will commit this changes into the repo and it will be part of next release. 😉

davidrapan, 2024-09-22:
[fix: Deye 2/4mppt switch](https://github.com/davidrapan/ha-solarman/commit/2d6f61a77e47ad363193178f4b5d7055cd9a8901)

closed 2024-09-22 by davidrapan