davidrapan/ha-solarman #170: Sensors unavailable

terzo33, 2024-09-22:
Hi @davidrapan , after update to 24.09.22 i have some sensors unavailable

[image]

[image]

[image]

[image]

davidrapan, 2024-09-22:
Yes, that's not a bug. I refactored entity names which did not fit our (soon to be) standardized scheme. And some static values (like serial number) were transferred into attributes to **Device State** entity.

terzo33, 2024-09-22:
Awesome. Thanks

closed 2024-09-22 by davidrapan

terzo33, 2024-09-22:
Little OT: hacs does not see updates to your integration. I have to download it manually everytime. Why?

davidrapan, 2024-09-22:
Not sure about that but I somehow noticed that too after HACS 2.0 was release. Maybe it just takes longer before it checks if there is anything to update?

**But it does notice the update when you click on the Update information under 3 dots menu.**

terzo33, 2024-09-22:
> Not sure about that but I somehow noticed that too after HACS 2.0 was released. Maybe it just takes longer before it checks if there is anything to update?
> 
> **But it does notice the update when you click on the Update information under 3 dots menu.**

when i press on the 3 dots nothing happens. i was stuck on version 24.09.08 after about 10 days i accidentally saw that version 24.09.18 was available. no update notification

davidrapan, 2024-09-22:
And did you click there on **Update information**?

terzo33, 2024-09-22:
> And did you click there on **Update information**?

Yes, nothing happened

davidrapan, 2024-09-22:
Weird, it did works that way for me.