davidrapan/ha-solarman #171: Wrong value on micro inverter

terzo33, 2024-09-23:
Hi @davidrapan , I always see this error message but actually there is no error, on solarman there is no warning

[image]

davidrapan, 2024-09-23:
Hi @terzo33, does the Value 1.900.546 changes over time or does it stays the same?

terzo33, 2024-09-23:
> Hi @terzo33, does the Value 1.900.546 changes over time or does it stays the same?

Alwasys the same in all 3 of my micro inverters

davidrapan, 2024-09-23:
I mean it should be 0 which is OK state. And I have no idea what could this particular value mean... Does your official Solarman App shows anything useful in the Alarm/Fault notifications?

terzo33, 2024-09-23:
Yes 4mppt profile, solarman app doesn't show me any errors

davidrapan, 2024-09-23:
And if you browse the history of the notifications? Can you find something there?

terzo33, 2024-09-23:
> And if you browse the history of the notifications? Can you find something there?

In solarman app?

[image]

davidrapan, 2024-09-23:
I could maybe add this values as an OK one for you when Solarman App does not see it as an error....

Anyway can I use this opportunity and ask you if sum of all your **PV1-4 Power** is ever higher than **Output Power**?

terzo33, 2024-09-23:
> I could maybe add this values as an OK one for you when Solarman App does not see it as an error....
> 

Yes. I think so because i can't see any errors

davidrapan, 2024-09-23:
Okay.

Could you please show me full sensor list of all your microinverters?

terzo33, 2024-09-23:
[image]

[image]

[image]

[image]

terzo33, 2024-09-23:
Where can i found register integration?

davidrapan, 2024-09-23:
You need to have Developer Tools enabled in the HA and then there under **ACTIONS** you will find `Solarman: Read Holding Registers (Modbus Function Code 3)` Action. Select Device, input Register (80) and click **PERFORM ACTION**. And tell me the result. (Repeat for register number 82)

terzo33, 2024-09-23:
> You need to have Developer Tools enabled in the HA and then there under **ACTIONS** you will find `Solarman: Read Holding Registers (Modbus Function Code 3)` Action. Select Device, input Register (80) and click **PERFORM ACTION**. And tell me the result. (Repeat for register number 82)

@davidrapan 

[image]

[image]

davidrapan, 2024-09-23:
Wonderful!

But to be 100% that they don't show any value could you maybe try to add into your profile and confirm it still does not show ay meaningful data?:
```
      - name: "Test Output"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 0.1
        rule: 3
        registers: [0x0050, 0x0051]

      - name: "Test Input"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 0.1
        rule: 3
        registers: [0x0052, 0x0053]
```

davidrapan, 2024-09-23:
And also:
```
      - name: "Test PV1 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 1
        rule: 1
        registers: [0x00BA]
        icon: "mdi:solar-power-variant"

      - name: "Test PV2 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 1
        rule: 1
        registers: [0x00BB]
        icon: "mdi:solar-power-variant"
```

terzo33, 2024-09-23:
> And also:
> ```
>       - name: "Test PV1 Power"
>         class: "power"
>         state_class: "measurement"
>         uom: "W"
>         scale: 1
>         rule: 1
>         registers: [0x00BA]
>         icon: [...]

[image]

[image]

[image]

davidrapan, 2024-09-23:
It should add the sensors into your device so show me list of the sensors please. 😉

terzo33, 2024-09-23:
> It should add the sensors into your device so show me list of the sensors please. 😉 

[image]

davidrapan, 2024-09-23:
Does it Test PV1 Power anyhow corresponds to PV1 Power?

terzo33, 2024-09-23:
> Does Test PV1 Power anyhow corresponds to PV1 Power?
> 
> Edit: Maybe if you change scale from 1 to 0.1?

[image]

davidrapan, 2024-09-23:
Okay nevermind. It was worth the shot regardless. Thank you for your help! Very appreciated!

terzo33, 2024-09-23:
It's a pleasure for me to help you

davidrapan, 2024-09-23:
@terzo33, can you try [v24.09.23](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.23)?

**And easiest way how to clean up sensors after changes is to delete and re-add device after the update to newer version.**

terzo33, 2024-09-23:
> @terzo33, can you try [v24.09.23](https://github.com/davidrapan/ha-solarman/releases/tag/v24.09.23)?
> 
> **And easiest way how to clean up sensors after changes is to delete and re-add device after the update to newer version.**

Interval update for pv information is always 5 min😔😔

davidrapan, 2024-09-23:
Create another debug log file for me from at least one hour.

davidrapan, 2024-09-23:
> Another error

That's really weird though... I wonder if is the register even correct?

terzo33, 2024-09-23:
> > Another error
> 
> That's really weird though... I wonder if is the register even correct?

How can I help you?

davidrapan, 2024-09-23:
> How can I help you?

That were just my thoughts...

Show me list of all sensors which you have there now so i can see if rest of the changes are okay.

**And record for me that debug log file from at least one hour please.** I'll check the data for PV sensors.

terzo33, 2024-09-23:
@davidrapan another error🙈

[image]

I don't understand

terzo33, 2024-09-23:
@davidrapan 

debug log

[image]

davidrapan, 2024-09-23:
Great, I'll look into it.

davidrapan, 2024-09-24:
Hi @terzo33, try through that service call for reading registers read register number 109 and set the quantity to 4 (that means it will read registers from 109 to 112). Those registers contains voltage and current of your panels.

**Try to read them in like 10 seconds interval and see if the values changes every call or not.**

davidrapan, 2024-09-24:
Little OT, could you also show me values of registers 15 and 19? Thank you!

terzo33, 2024-09-24:
> Little OT, could you also show me values of registers 15 and 19? Thank you!

[image]

davidrapan, 2024-09-24:
> Yes. Change with 1/2 minutes interval

Can you show me?

terzo33, 2024-09-24:
> > Yes. Change with 1/2 minutes interval
> 
> Can you show me?

See clock

This interval is 3 min but I saw also about 2 min

[image]

[image]

terzo33, 2024-09-24:
> **To really have concrete results could you do 3 in a row with 1 min space in between each?**
> 
> Edit: Follow up question... Is the refresh period the same for all 3? **Current - Power - Voltage**? (When you look at list of all sensors)

Interval time is about 4/5 min.

I don't understand😓... other values of microinverter update with 5/6 seconds 

[image]

[image]

[image]

[image]

[image]

terzo33, 2024-09-24:
> Edit: Follow up question... Is the refresh period the same for all 3? **Current - Power - Voltage**? (When you look at list of all sensors)

Yes the same time but only for pv information.

davidrapan, 2024-09-24:
> Yes the same time but only for pv information.

You mean Voltage and Current also updates with 5 min interval?

terzo33, 2024-09-24:
> > Yes the same time but only for pv information.
> 
> You mean Voltage and Current also updates with 5 min interval?

Yes

davidrapan, 2024-09-24:
Hey, open the folder with solarman code. Open file named api.py and on the row `210` **REPLACE** that row with following:
```
_LOGGER.debug(f"[{self.serial}] Returning {result_count} new values to the Coordinator. [Previous State: {self.get_connection_state()} ({self.state}), Values: {result}]")
```

Restart the HA and enable debug, log for a while and then attach it here, thanks.

terzo33, 2024-09-24:
> Hey, open the folder with solarman code. Open file named api.py and on the row `210` **REPLACE** that row with following:
> ```
> _LOGGER.debug(f"[{self.serial}] Returning {result_count} new values to [...]

Doesn't work

[image]

[image]

davidrapan, 2024-09-24:
You accidentally removed these rows right after 210:
```
            now = datetime.now()
            self.state_interval = now - self.state_updated
            self.state_updated = now
            self.state = 1
```

terzo33, 2024-09-24:
@davidrapan

I had to lighten the file

[image]

davidrapan, 2024-09-24:
So yeah I mean you can see in the file that the data returned from the device are simply not refreshing with every request...

davidrapan, 2024-09-24:
Every hint is pointing in that direction, yes. You have probably set from the Deye side update interval to 5m. **Try to write to Deye support and ask them to lower the update interval of your microinverters and see if that changes anything.**

terzo33, 2024-09-24:
@davidrapan yes done. Just set to 60 sec

davidrapan, 2024-09-24:
What do you mean? You have already set it to 60 s?

terzo33, 2024-09-24:
> What do you mean? You have already set it to 60 s?

Yes. With deye support

terzo33, 2024-09-24:
> What do you mean? You have already set it to 60 s?

[image]

terzo33, 2024-09-24:
I saw grid ac value with 5 sec update interval on HA. Can you check on debug?

davidrapan, 2024-09-24:
Yes there some occasions when for example Grid Voltage changes from 236V to 235V and then immediately back to 236V in 5 s interval.

terzo33, 2024-09-26:
@davidrapan i don't understand why "update interval" for grid voltage is about 5/10 sec and for other values is 4/5 minutes 🤷‍♂️

https://github.com/user-attachments/assets/9f8b3a8b-d8cd-4ff7-95e0-237b77659254

davidrapan, 2024-09-26:
I honestly don't get it either.

_The way the indication of `? seconds/minutes ago` change works is that it will update only if the value displayed changes. So because the value given by the inverter does not change it displays different update periods._

terzo33, 2024-09-28:
@davidrapan Another error :(

[image]

closed 2024-11-13 by github-actions[bot]