davidrapan/ha-solarman #173: Deye 4MPPT - missing values - 24.09.2x

Wubarius, 2024-09-23:
**Describe the bug**
Version 24.09.08 - with "deye_4mppt.yaml" - works fine
Version 24.09.20 - no values
Version 24.09.22 - no values

"Total AC Output Power (Action)" -  missing
There are no more values ​​for the current PV production.
Or the entity has changed to "Output Power"
Is that what you want?
If so - I really have to adjust a lot of things.

**To Reproduce**
Steps to reproduce the behavior:
Update Solarman from 24.09.08 to latest version

**Expected behavior**
Why were the entities changed and the old ones remained but without values.
- Total AC Output Power (Action) - unavailable
- Output Power

**Screenshots**
see 3 files from device "Garage"

**Metadata:**
Version: 24.09.08 - yaml version 12.08.2024
Version: 24.09.22 - yaml version 17.09.2024

I replaced the file "deye_4mppt.yaml" with the old one.
New entities are displayed without values
- Output Frequency
- Output Power
- Total Production 3 & 4

[image]
[image]
[image]

davidrapan, 2024-09-23:
Hi @Wubarius, yes it is. You can still use the older version of the file as custom profile if it's too much problem for you. 😉

Wubarius, 2024-09-23:
Hi @davidrapan,

thanks for the quick reply.

I'll probably keep the old ones. I'd have to change too much if the entity was renamed.

I've now integrated the new one - but I'm still using the old ones.

closed 2024-09-23 by Wubarius

davidrapan, 2024-09-23:
Yeah, sorry that I totally forgot to write about those changes in the Release Note. That's big fail from my side. That shouldn't have happened. Another fail is that i messed up the new entity names anyway. Guess I need a break, lol.

Again sorry.