davidrapan/ha-solarman #176: Inverter sofar_lsw3 zero sensors

jeffersonbernardo1981, 2024-09-23:
Hello, I'm having some difficulty with ref. integration. I use the sofar_lsw3 inverter, and I've already installed and uninstalled it several times and the sensors just stay at zero, has anyone experienced this and managed to solve it?
[image]
[image]
 version: 24.09.23

davidrapan, 2024-09-24:
Hi @jeffersonbernardo1981, what exact device and model you have?

jeffersonbernardo1981, 2024-09-24:
Hi David, follow the label data for the Sofar Solar inverter, Model 5KTLM-G3.
[image]

davidrapan, 2024-09-24:
And did you also try it with other profiles? For example: `sofar_tlx-g3.yaml` or `zcs_azzurro-ktl-v3.yaml`?

jeffersonbernardo1981, 2024-09-24:
Yes, I tried them all and the only one I have feedback on is sofar_lsw3_yaml, the others give an error as shown in the image:
[image]

davidrapan, 2024-09-24:
Could you enable debug logging and load it with `sofar_tlx-g3.yaml` and attach the debug log here? Thank you.

jeffersonbernardo1981, 2024-09-24:
David, follow the logs!
[home-assistant_solarman_2024-09-24T15-26-38.265Z.log](https://github.com/user-attachments/files/17116966/home-assistant_solarman_2024-09-24T15-26-38.265Z.log)

jeffersonbernardo1981, 2024-09-24:
David, I managed to perform a firmware update directly with the manufacturer, and I did the entire process, uninstalled the integration and reinstalled it again, I am getting feedback on some data but others do not follow the prints.
[image]
[image]
[image]

davidrapan, 2024-09-24:
Great! That's a progress! Which profile did you used in this case?

It's still possible that your device uses some specific registers for some values so you will need some new profile specifically tuned for your device.

**Anyway which sensor values do work and which does not?**

jeffersonbernardo1981, 2024-09-24:
I'm using this profile, sofar_tlx-g3.yaml.

closed 2025-02-03 by davidrapan