davidrapan/ha-solarman #180: Make some controls disabled by default

voed, 2024-09-27:
I understand this controls may be useful for someone, but i think for most people there are no any reason to change this settings from Home Assistant. But, it is pretty easy to missclick while scrolling through them which may cause damage to the battery. I suggest to make most of them disabled by default.
[image]

davidrapan, 2024-09-27:
Hi @voed, I can see where you're coming from, but the truth is that the options are pretty harmless (except for one in the SG0XXP3 profiles) and at worst it just won't work. And I say that because I've seen some BAD configurations people have had here, hah! So from this point of view, I think it is good to make those values ​​rather visible in HA.

BUT I will soon add some kind of a locking mechanism to prevent accidental changes! 😉

PV-Joe, 2024-09-28:
> BUT I will soon add some kind of a locking mechanism to prevent accidental changes!

Yes please. That would be great!

closed 2025-12-24 by davidrapan