davidrapan/ha-solarman #181: Microinverter starting voltage -> Status unknown

terzo33, 2024-09-28:
@davidrapan status unknown for "starting voltage" microinverter

[image]

davidrapan, 2024-09-28:
Hi @terzo33, show me please registers 32 and 33.

Thanks!

terzo33, 2024-09-28:
> Hi @terzo33, show me please registers 32 and 33.
> 
> Thanks!

[image]

davidrapan, 2024-09-28:
Fuh, those values does not make much sense in terms of what documentation says.. :-/

**Can you somehow find out to which values is Starting voltage Upper/Lower Limit is really set on your device?**

terzo33, 2024-09-28:
> Fuh, those values does not make much sense in terms of what documentation says.. :-/
> 
> **Can you somehow find out to which values is Starting voltage Upper/Lower Limit is really set on your device?**

On my inverter or on battery?

davidrapan, 2024-09-28:
> On my inverter or on battery?

Microinverter.

terzo33, 2024-09-28:
> > On my inverter or on battery?
> 
> Microinverter.

but I don't think you can connect the battery to the micro inverter. there is no connection

davidrapan, 2024-09-28:
This is not about the battery at all.

terzo33, 2024-09-28:
> > On my inverter or on battery?
> 
> Microinverter.

On which page can I see this value?

davidrapan, 2024-09-28:
Probably in device parameters?

terzo33, 2024-09-29:
> Probably in device parameters?

I don't know where to look for these values. even if sularman I can't find anything. but I can't find the ones you entered on the home assistant like the soft start and others.

[image]

[image]

CrazyUs3r, 2024-09-29:
At my sunm80g3 is the value all time zero

[image]

davidrapan, 2024-09-29:
@terzo33, only in Solarman App.

terzo33, 2024-09-29:
> @terzo33, only in Solarman App.

I can't see nothing

[image]

[image]

[image]

davidrapan, 2024-09-29:
There should be somewhere also configurable parameters. Which you can change from the App.

terzo33, 2024-09-29:
@davidrapan I can't find them. Do you have any suggestions where to look?

CrazyUs3r, 2024-09-29:
Right Upp the 3 ...

terzo33, 2024-09-29:
> Right Upp the 3 ...

What mean?

CrazyUs3r, 2024-09-29:
[image]

terzo33, 2024-09-29:
@davidrapan @CrazyUs3r 

[image]

[image]

[image]

davidrapan, 2024-09-29:
That's under "Controllo remoto"?

It's as Device Control in English.

terzo33, 2024-09-29:
> That's under "Controllo remoto"?
> 
> It's as Device Control in English.

https://github.com/user-attachments/assets/b4fe0b0f-e037-4676-b671-3a814c8b990f

davidrapan, 2024-09-29:
What if you look at it on the website and not the app?

davidrapan, 2024-10-02:
[fix: Removed Starting Voltage * from Deye 2/4mppt](https://github.com/davidrapan/ha-solarman/commit/cb98702055ffcb84c1a1b1cec2bd97d4891c6081)

closed 2024-10-02 by davidrapan