davidrapan/ha-solarman #187: DDZY422-D2 integration

bobybob69, 2024-10-01:
**Describe the bug**
A clear and concise description of what the bug is.

I try to install my Solarman DDZY422-D2 Smart Meter using the integration ha-solarman. 
When setting everything up, I got an error message "Échec de la configuration, nouvel essai: [SERIALNUMBER] Querying IP-ADRESS:PORT failed: [0, 0]." 

**Attach the debug log**
As a file or using the 'code' formatting.

**To Reproduce**
Steps to reproduce the behavior:

From the setting, add an integration 
set your IP address
Port 
Serial number 
select solarman_dtsd422-d3.yaml 
apply setting
Wait for the confirmation and notice the error message "Échec de la configuration, nouvel essai: [SERIALNUMBER] Querying IP-ADRESS:PORT failed: [0, 0]." 

**Expected behavior**
A clear and concise description of what you expected to happen.

I don't know if it's the right configuration path as it's the D3 on the choice list 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Metadata:**
Version: 24.09.27

**Additional context**
Add any other context about the problem here.

I don't know if it's a bug report or if it's a request feature to add the configuration .yaml for the D2 smart meter (I thought D2 / D3 where the same.. ?)

bobybob69, 2024-10-01:
about the log details, If you could explain how to extract them it would be helpful and I would be able to past the code here. thanks ! :)

davidrapan, 2024-10-02:
Hi @bobybob69, like this:
[image]

bobybob69, 2024-10-02:
it was running on my screenshot but I didn't understand the file was downloading when clicking on stop 😅

please found bellow the log file 

[home-assistant_solarman_2024-10-02T19-02-14.489Z.log.zip](https://github.com/user-attachments/files/17234790/home-assistant_solarman_2024-10-02T19-02-14.489Z.log.zip)

Let me know if you get everything 

thanks for looking at it ! and let me know if you need anything else to help

davidrapan, 2024-10-02:
Extracted flow from the log file:
```
2024-10-02 21:00:41.664 DEBUG (MainThread) [custom_components.solarman.api] [????????????] Scheduling 2 query requests. #0
2024-10-02 21:00:41.664 DEBUG (MainThread) [custom_components.solarman.api] [????????????] Querying 0001 - 0041 | 0x0001 - 0x0029 # 041 ...
2024-10-02 21:00:41.664 INFO (MainThread) [custom_components.solarman.api] [????????????] Connecting to 17.0.0.81:8899
2024-10-02 21:00:41.672 DEBUG (MainThread) [custom_components.solarman.api] [????????????8] SENT: a5 17 00 10 45 02 00 9e 8e ee 9b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 29 d5 d4 fc 15
2024-10-02 21:00:43.683 DEBUG (MainThread) [custom_components.solarman.api] [????????????] RECD: a5 10 00 10 15 02 4a 9e 8e ee 9b 02 01 8d 8b 48 00 fc 15 00 00 ce 0c b5 66 05 00 a4 15
2024-10-02 21:00:43.684 DEBUG (MainThread) [custom_components.solarman.api] [????????????] Querying 0001 - 0041 | 0x0001 - 0x0029 # 041 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
```

Hi @githubDante, Exception with code 05 (0x05) | Acknowledge. Does that says anything useful? The profile is made for D3 version so it can be just about incompatibility but I'm curious what are your thoughts?

githubDante, 2024-10-03:
Hi @davidrapan,

Incompatible profile 100% - wrong start address or quantity

davidrapan, 2024-10-03:
Thanks @githubDante,

So you will have to try probe registers using built-in service for reading and create new profile for your device @bobybob69

bobybob69, 2024-10-03:
Hi @davidrapan and @githubDante  thanks for looking at the log. 

Well, I'm not familiar with that. 
What could I do to help implement that easily ? 

After comparing D2 and D3 it seems like it's just difference of power distribution (simple-phase or triple-phase) 
D2 is a simple phase, the type of install I've at home. 

Let me know I would be more than happy to help learn and implement that with one of you guys 

Thanks for helping

bobybob69, 2024-10-09:
Let me know if there is anything I could do to help integrate the D2 guys 

Thanks !

davidrapan, 2024-10-09:
It's quite simple. You just have to copy for example inverter_definitions/solarman_dtsd422-d3.yaml into inverter_definitions/custom/solarman_ddzy422-d2.yaml

* Add device using this new custom profile
* Remove all sensors which are causing issues
* Use bult-in services for reading (Developer Tools > ACTIONS > Solarman reading services 0x3 and 0x4)

And you will have to find those register which shows matching values in the Solarman App.

Sadly without documentation there is no other way than trial and error.

bobybob69, 2024-10-12:
hi @davidrapan , thanks for the tips ! 

I try to do the copy / paste yaml parts. But the integration mark an error (see bellow) , I attach also the log. 

Since the integration failed, there is nothing more for me to do using the Developper ACTION parts .. ? 

[image]

Any tips that I could try to implement the D2 ? 

Thanks ! 

[image]

[home-assistant_solarman_2024-10-12T18-50-02.280Z.log](https://github.com/user-attachments/files/17351148/home-assistant_solarman_2024-10-12T18-50-02.280Z.log)

davidrapan, 2024-10-12:
It's not relevant during this process and you can ignore it. Using the ACTIONS will still work.

bobybob69, 2024-10-12:
Bellow are the log after adding the proper modbus register but I still have an issue.. 

Any thought @davidrapan ? 

[home-assistant_solarman_2024-10-12T19-24-08.095Z.log](https://github.com/user-attachments/files/17351246/home-assistant_solarman_2024-10-12T19-24-08.095Z.log)

davidrapan, 2024-10-12:
According to the log requests to your Smart Meter still contain invalid addresses.

bobybob69, 2024-10-12:
Humm... progress but still some errors.. any thoughts @davidrapan  from where I can found these modbus register ? any macOS tools I can use to read theses address using the IP ? I sent an email to see if they have theses for me as there is nothing on the user guide unfortunately 

I found these on the net , what do you think ? 

[solarmlan_DDZY422.yaml.txt](https://github.com/user-attachments/files/17351287/solarmlan_DDZY422.yaml.txt)

davidrapan, 2024-10-12:
Yeah it's not publicly available information and not sure if solarman will be willing to provide the docs.

> from where I can found these modbus register ? any macOS tools I can use to read theses address using the IP ?

I told you using the ACTIONS trough this integration. That is the best tool you can get.

> I found these on the net , what do you think ?

I don't have the device so I can't really tell. But take it slowly. Just start even with single sensor defined in the file with registers you verified that works using said ACTIONS. 😉

githubDante, 2024-10-12:
Found this through google
[solarmlan_DDZY422.yaml.txt](https://github.com/user-attachments/files/17351415/solarmlan_DDZY422.yaml.txt)

bobybob69, 2024-10-13:
Hey @davidrapan @githubDante 
I think I understand how the actions from HA is working.
However, how can I use it even if the Smart Meter isn't list ? 
Would it be better to use YAML config instead of UI ? And so what would be the next step ? 

If I understand, There is an action I use to read values , but how can I read values from a devices that's not listed of even setup ? From what I understand from your previous message  

[image]

Bellow is what I'm seeing from the actions section when I choose reading value code 4 : 

[image]

Here's my config for the smart meter : 

[image]

Any tips would be helpful :) 

Thanks guys !

davidrapan, 2024-10-13:
> If I understand, There is an action I use to read values , but how can I read values from a devices that's not listed of even setup ?

I told you that you have to setup the device first. And ignore when it shows errors. Then try read different registers and find those which works...

bobybob69, 2024-10-13:
I'm sorry if I don't understand but I just want to make I understand what you say @davidrapan 

- first I need to found a way to have the meter not showing any error for the first setup (it doesn't matter if for sensor from the smart meter there is error right) 
- then for the sensor that will have errors, I need to investigate using the action menus from HA right ? 
- then , when troubleshooting everything, it should works properly

Did my thinking is on the right way ? If no please let me know 

Thanks for your help I will owe you a coffee at the end 😜

davidrapan, 2024-10-13:
> first I need to found a way to have the meter not showing any error for the first setup (it doesn't matter if for sensor from the smart meter there is error right)

I saying again. it does not matter if it shows errors. IGNORE IT! Just add it so you can use the ACTIONS. Thats all.

bobybob69, 2024-10-14:
hey guys @davidrapan @githubDante , I just get an answer from solarman, does this helps ? 

[Single Phase meter DDZY422-D2 parameters_1728884136958_yiy4p.pdf](https://github.com/user-attachments/files/17366674/Single.Phase.meter.DDZY422-D2.parameters_1728884136958_yiy4p.pdf)

bobybob69, 2024-10-14:
> > first I need to found a way to have the meter not showing any error for the first setup (it doesn't matter if for sensor from the smart meter there is error right)
> 
> I'm saying again. it does not matter if it shows errors. IGNORE THEM! Just add it so you can use the ACTIONS. Thats all.

To be faire @davidrapan , I should miss something.. I'm sorry. When you say "if it show error ignore them" what do you mean ? 

Here's the status of the meter actually 

[image]

How can I ignore this error ? Even in this state, iit doesn't appear on the action menu .. :(

davidrapan, 2024-10-14:
What I mean is that even you see error ACTIONS will still work so you just can leave it as it is.

bobybob69, 2024-10-14:
Well @davidrapan , don't ask me how, I change few line on the code (helped by ChatGPT) and now the smart meter is added 

[image]

now when I'm in the action menu what would be the best to troubleshoot each register ? I think I'm not using the correct syntax for the register .. ? 

[image]

what about this screenshot, I'm on the right way ? 

thanks for helping

davidrapan, 2024-10-14:
> what about this screenshot, I'm on the right way ?

Yes you are. That means that you just can't read register number 1 so you need to continue with 2... and so on. But to be 100% sure I would also try reading using Modbus Function Code 3 (it has separate service for that).

davidrapan, 2024-10-15:
There are two ACTIONS for reading:
* Modbus function code 3
* Modbus function code 4

Did you also tried with the other one?

bobybob69, 2024-10-16:
Hey @davidrapan , yes I try with both methods. 
Still having same error message. 

How can I be sure the meter is reachable from HA after I set it up ? 

The IP if a fixed one so I'm 100% sure this is the good one. 

What I'm less sure is the YAML for the config. 
It let me add the smart meter , but the only entities I've is "connection" and it appear "disconnected" 
I don't know if it's 100% sure but I've a doubt.. 

What's your thought ? 

Thanks and have a great day 

[image]

davidrapan, 2024-10-16:
> How can I be sure the meter is reachable from HA after I set it up ?

Because you are getting replies from it?

bobybob69, 2024-10-16:
I don't know @davidrapan ..
 
how can I know if I've answer, since the beginning I'm facing error so I don't know if I'm receiving anything.. 

Nothing else to do except the action button ? 

What would you need (except the Modbus) to be able to add the smart meter into HA , even if it's without any data ?

davidrapan, 2024-10-16:
**V5 frame does not contain a valid modbus RTU frame.**

This means that the device is talking with you but you send incorrect request.

bobybob69, 2024-10-16:
oh ok , so , it's a good things .. 

But to be honest I try several number of Modbus register and none if working.. is there a way to at least have 1 modbus register that should works as it's kind of a standard ? No matter the register I always have theses error and I tried different combination .. no one works. 

[image]

Or if we compare to the solarman_dtsd422-d3.yaml , is there some "classic" register that should work 100% sure ? 

Maybe this can help to see if it's working anyway 

Thanks for your help @davidrapan as always

bobybob69, 2024-10-16:
I also saw this pages : [HERE](https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_igen_dtsd422.md)

Which register modbus should I use if I want to try ? dec or hex address ? 

Thanks!

bobybob69, 2024-10-16:
Also get theses errors from the logs when trying each time the action of modbus reading.. which setting would need to be changed ? 

[image]

bobybob69, 2024-10-25:
Hey @davidrapan how are you ?
Well, after few investigation here's what's I get from SOLARMAN 
It's in Chinese for the moment I'll need to translate and integrate it. 
If you want I can share it with you bellow if you want to take a look at the file and try isolate few data's.

davidrapan, 2024-10-25:
Sure thing, put the untranslated original here.

bobybob69, 2024-10-26:
hey @davidrapan please found bellow the file :) 

Let me know if it's a good starting point ! 

[DDZY422-D2型单相远程费控智能电能表-modbus通讯内容（HT-YF2018-0929-01）_1729757263801_10de75.xlsx](https://github.com/user-attachments/files/17530651/DDZY422-D2.-modbus.HT-YF2018-0929-01._1729757263801_10de75.xlsx)

bobybob69, 2024-10-29:
Hey @davidrapan 
Does that helps ? I try on my side but I didn't found a way to have it works I shoukd have something set wrong at some point ..

davidrapan, 2024-10-29:
If you have still issues just take it slowly. Create file with just single entity description and add more after you make sure that what you already have works.

davidrapan, 2024-11-02:
Try this:
```
#
# Solarman Smart Meter DDZY422-D2
#
# Reference: https://github.com/user-attachments/files/17586995/DDZY422-D2.-modbus.HT-YF2018-0929-01._1729757263801_10de75.xlsx
#

default:
  update_interval: 5
  digits: 3

parameters:
  - group: Variable Data 1
    items:
      - name: Voltage
        class: voltage
        uom: V
        rule: 1
        registers: [0x0001]

      - name: Current
        class: current
        uom: A
        rule: 2
        magnitude: True
        registers: [0x0003, 0x0002]

      - name: Power
        class: power
        uom: W
        scale: 0.001
        rule: 2
        magnitude: True
        registers: [0x0007, 0x0006]

      - name: Power Factor
        class: power_factor
        state_class: measurement
        scale: 0.001
        rule: 2
        magnitude: True
        registers: [0x000C]

      - name: Frequency
        class: frequency
        state_class: measurement
        uom: Hz
        scale: 0.01
        rule: 1
        registers: [0x000D]
```

bobybob69, 2024-11-02:
hey @davidrapan 

I got the same error message .. "Échec de la configuration, nouvel essai: [2616102558] V5FrameError: V5 frame does not contain a valid Modbus RTU frame"

As you did the file and you used the modbus config Solarman share with me for this model, it confirm there is something else wrong on my side.. but what could be wrong ?

Bellow is the config from the solarman 

[image]

Can it be the port that's not the good one ? 

Maybe the way I use add the config ? 

I'm using File Editor module
then I go to customer component > solarman > inverter_definitions > I create a file I named solarman_DDZY422-D2.yaml and I copy paste the code you sent. 
Then I save and I restart entirely my HA instance.. 

Ping on the IP from the smart meter is working so I know there is a response from the smart meter. 

What could be wrong here ? :/ 

thanks mate for our collaboration I appreciate !

davidrapan, 2024-11-02:
Enable debug log and upload the file.

davidrapan, 2024-11-02:
I can't see in the doc to which memory area are the registers mapped so maybe it's worth the shot to try 0x04.

Try to replace that default section with:
```
default:
  update_interval: 5
  code: 0x04
  digits: 3
```

bobybob69, 2024-11-03:
hi @davidrapan ,
I tried to adjust the default section as you mentioned but nothing changed 
Please found bellow the logs. 

So if I understand, we are just looking for the correct memory right ? and for you, all the rest of the config is ok 👍 

[home-assistant_solarman_2024-11-03T20-38-21.631Z.log](https://github.com/user-attachments/files/17612001/home-assistant_solarman_2024-11-03T20-38-21.631Z.log)

davidrapan, 2024-11-03:
Yeah well if it's not even working with this minimal yaml file (which is was made using xlsx file you provided)...

_I'm afraid that there is something else going on which is outside of our reach._

I'm out of ideas... there is proly not much we can do, right @githubDante?

githubDante, 2024-11-03:
Why don't you try with the registers from the XLS file e.g. from 0x105 to 0x107 ? It's a 6 byte Date&Time.

davidrapan, 2024-11-03:
Okay, try this one:
```
#
# Solarman Smart Meter DDZY422-D2
#
# Reference: https://github.com/user-attachments/files/17586995/DDZY422-D2.-modbus.HT-YF2018-0929-01._1729757263801_10de75.xlsx
#

default:
  update_interval: 5
  digits: 3

parameters:
  - group: Variable Data 1
    items:
      - name: Date & Time
        platform: datetime
        rule: 8
        registers: [0x0105, 0x0106, 0x0107]
```

bobybob69, 2024-11-04:
hey guys, 

just tried now, and still same error message ... "Echec de la configuration, nouvel essai: [2616102558] V5FrameError: V5 frame does not contain a valid Modbus RTU frame" 

please found bellow the logs (in the meantime It can maybe help for #203 as logs turn all day until it turn off when I restart HomeAssistant during no solar production) 

[home-assistant_solarman_2024-11-04T20-14-28.300Z.log](https://github.com/user-attachments/files/17623335/home-assistant_solarman_2024-11-04T20-14-28.300Z.log)

you confirm when I made the change from the config file, I restart HA instance and I re-set the meter ? 

thanks guys

davidrapan, 2024-11-04:
Uhm but you don't have that device (2616102558) configured w/ `solarman_DDZY422-D2.yaml`!

I see it configured with `deye_hybrid.yaml`!

bobybob69, 2024-11-04:
Hummmm strange I double checked, it's configured with the right profile.. 

See logs bellow 

[home-assistant_solarman_2024-11-04T20-28-09.595Z.log](https://github.com/user-attachments/files/17623462/home-assistant_solarman_2024-11-04T20-28-09.595Z.log)

bobybob69, 2024-11-04:
just so you know, I deleted the devices and I re-install it properly . 
And the config file with the latest logs I sent only contain what you shared bellow , is it right ? 

> Okay, try this one:
>
> ```
> #
> # Solarman Smart Meter DDZY422-D2
> #
> # Reference: https://github.com/user-attachments/files/17586995/DDZY422-D2.-modbus.HT-YF2018-0929-01._1729757263801_10de75.xlsx
> # [...]

davidrapan, 2024-11-04:
Yes now it uses correct one. Unfortunately w/ the same result.. 😞

bobybob69, 2024-11-04:
can it be the port that cause issue ? by default 8899 ? 
what could be wrong.. it's so frustrating at this level to not be able to integrate it even with the modbus xls 😞

Any troubleshoot I could do locally ? 
The errors are because the meter doesn't answer back the data we're looking for right ? 
Or it's the modbus that is not correct apparently ?

githubDante, 2024-11-04:
> can it be the port that cause issue ? by default 8899 ?
> what could be wrong.. it's so frustrating at this level to not be able to integrate it even with the modbus xls 

No, it's not the port. The smart meter is responding with an error/exception code, which means that the request is wrong (wrong address/wrong function). 

> Any troubleshoot I could do locally ?
> The errors are because the meter doesn't answer back the data we're looking for right ?
> Or it's the modbus that is not correct apparently 

The best you can do is to request the proper/full MODBUS documentation for the smart meter from IGEN. Apparently the one in the XLS file is wrong and incomplete.

bobybob69, 2024-11-13:
hi @githubDante , how are you ? I got answer form the support team of solarman, and they confirm the file is the good one.. what can we do to integrate it ? from where the issue can come from ? :/ 
thanks mate and have a pleasant day

davidrapan, 2024-11-13:
Did you mentioned what responses are we getting from the meter?

bobybob69, 2024-11-14:
hi @davidrapan , just answer back to their message adding the error message we're facing and I'm waiting for their answer.. will share it with you as soon as I got the answer

teoJam, 2024-12-03:
Hello, 
I have the same device (DDZY422-D2), the zucchetti ZH3ES non-hybrid inverter. I have a solarman logger that HA can read, but I also have the DDZY422-D2 meter, in which I can't get the integration to work.

I am new to Home Assistant, but if you can explain what tests I can do to add my logs as well, if I can be of any help, it is an issue I am interested in solving.

I have done some of the tests you mention here, but for example in Actions I can't find the Solarman action. How come?

Also in configuration I can't find the files I create from file manager.
I have created the file 
/homeassistant/custom_components/solarman/inverter_definitions/solarman_DDZY422-D2.yaml
 with the values you suggested in it, but in the device menu I do not see the file, so I copied the same values to the custom_parameters.yaml file and configured the device with this file, but of course it still always goes to error.

Despite this when I then review the device configuration it seems to be reading the zcs_blue-ktl-v3.yaml file I started from.

By reading or log I realized that I used to get an error because I was using the external HA url with *.duckdns.org, but then they resolved the moment I started using the local one.

Thank you

davidrapan, 2024-12-03:
Show me a screenshot where did you put the file.

teoJam, 2024-12-03:
[image]

davidrapan, 2024-12-03:
You are not running this integration.

[61 more replies]