davidrapan/ha-solarman #188: PV Power with Deye 4MPTT

bobybob69, 2024-10-02:
**Is your feature request related to a problem? Please describe.**

When I was using the previous solarman integration from Stephane Joubert, I was able to select total PV Power for my Inverter and set them up in the energy dashboard. 
Now, the entities for PV Power (that count the sum of all my PV production on my inverter) are not eligible to be set in the energy dashboard. 

The only entities I have eligible for the energy dashboard is "Today's production" , but it stay at 0 kWh. 

Did I made something wrong in understanding energy dashboard or their is something wrong in the setup of the integration that doesn't made it eligible for the dashboard ? 

**Describe the solution you'd like**

Make eligible PV Power (that represent the solar production per inverter) eligible for the energy dashboard.

**Describe alternatives you've considered**

I try to create entities that create a sum with Riemman for the PV power but it doesn't work as I would like to .. 

**Additional context**

If it was working in the previous integration, I'm sure it's something easy to fix on this excellent integration! :) 

Bellow is some screenshot of what I can set for the energy dashboard. And you can notice "today's production" stay at 0 

[image]

Thanks for reading :)

davidrapan, 2024-10-02:
Hi @bobybob69, I'm quite confused.. In the title you are talking about PV Power but then you are describing an issue with the energy dashboard?

First thing is **Today's \*** sensors has no place in energy dashboard. **Total \*** sensors has to be there.

Second one is so your **Today's \*** sensor does not work? What kind of a device you have? Full model type pls.

bobybob69, 2024-10-02:
hi @davidrapan , 

It's not an issue, after reading @Triple-S I think it's more a misunderstanding from me about how energy dashboard is working. 
I thought the today's sensor would work instantly but apparently there is some time before having the data showing on the energy dashboard ? 
Energy dashboard seems to work on an hourly basis right ? 
thanks for sharing those tips guys it helped me understand more this particular setup !

davidrapan, 2024-10-02:
Sure thing. 👍

Switch those **Today \*** for **Total \*** (those without ever resetting to zero) in the energy dashboard and you should be good! 😉

closed 2024-10-05 by davidrapan