davidrapan/ha-solarman #189: deye inverter on parallele

bmatti996, 2024-10-02:
hi i have 2 deye inverter 6kw with the solarman stick logger and the integration see only the master 

[image]
[home-assistant_solarman_2024-10-02T21-12-00.412Z.log](https://github.com/user-attachments/files/17236119/home-assistant_solarman_2024-10-02T21-12-00.412Z.log)

davidrapan, 2024-10-02:
Hi @bmatti996 It looks like you're using wrong serial number.

You are using correct serial number for 192.168.178.192 which is 2907XXXXXX but for 192.168.178.52 you set 2312XXXXXX which is not correct (probably serial number of the inverter instead of the stick logger) and you should use serial number which starts like this 2905XXXXXX.

😉

bmatti996, 2024-10-04:
thanks you,
now it works

closed 2024-10-04 by bmatti996