davidrapan/ha-solarman #191: Can't connect to SOFAR ME3000SP

stephenchilds76, 2024-10-05:
Hi,

I can't connect to my SOFAR ME3000SP with WiFi stick logger. I configured solarman with serial number, IP address, etc.

Get the following error:
[image]

i.e. V5 frame does not contain valid modbus frame

I have tried all the SOFAR yaml parameter files

Firmware version is MW_08_512_0501_1.94

stephenchilds76, 2024-10-05:
Have been trying different parameter files. I can connect successfully with various solis parameter files but as these are not for hybrid inverters I can't see info about the battery.

Also not sure if the values returned will be correct since it's for solis inverter

davidrapan, 2024-10-05:
Hi @stephenchilds76, turn on debug logging and attach the file.

davidrapan, 2024-10-05:
Try this profile: zcs_azzurro-1ph-tl-v3.yaml

stephenchilds76, 2024-10-05:
this is debug log with WiFi kit (azzurro yaml didn't work either)

2024-10-05 16:32:59.491 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-05T15:04:01.805965+00:00', 'data': {'name': 'ME3000SP', 'inverter_host': '192.168.1.10', 'inverter_serial': 4050986569, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'sofar_wifikit.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J9EKB1GDP5X797GQR2KQH1ND', 'minor_version': 1, 'modified_at': '2024-10-05T15:29:45.308202+00:00', 'options': {'inverter_host': '192.168.1.10', 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'sofar_wifikit.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'ME3000SP', 'unique_id': 'solarman_4050986569', 'version': 1})
2024-10-05 16:32:59.491 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-10-05 16:32:59.491 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 192.168.1.10
2024-10-05 16:32:59.491 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-10-05 16:32:59.565 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.1.10, 34EAE75D1B1E, 4050986569]
2024-10-05 16:32:59.709 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: 25, digits: 6
2024-10-05 16:32:59.710 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '34:ea:e7:5d:1b:1e')}, 'identifiers': {('solarman', 4050986569)}, 'name': 'ME3000SP', 'manufacturer': 'Sofar', 'model': 'WIFIKIT', 'serial_number': 4050986569}
2024-10-05 16:32:59.710 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-10-05 16:32:59.710 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Scheduling 4 query requests. #0
2024-10-05 16:32:59.710 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 ...
2024-10-05 16:32:59.711 INFO (MainThread) [custom_components.solarman.api] [4050986569] Connecting to 192.168.1.10:8899
2024-10-05 16:32:59.716 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] SENT: a5 17 00 10 45 aa 00 49 26 75 f1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 00 00 46 c5 80 7e 15
2024-10-05 16:33:00.866 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] RECD: a5 10 00 10 15 aa 90 49 26 75 f1 02 01 74 1a 25 02 a2 0b 00 00 b8 41 dc 64 05 00 e7 15
2024-10-05 16:33:00.866 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-05 16:33:01.069 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 attempts left: 4
2024-10-05 16:33:01.069 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] SENT: a5 17 00 10 45 ab 00 49 26 75 f1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 00 00 46 c5 80 7f 15
2024-10-05 16:33:02.352 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] RECD: a5 10 00 10 15 ab 91 49 26 75 f1 02 01 76 1a 25 02 a3 0b 00 00 b8 41 dc 64 05 00 ec 15
2024-10-05 16:33:02.353 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-05 16:33:02.754 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 attempts left: 3
2024-10-05 16:33:02.754 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] SENT: a5 17 00 10 45 ac 00 49 26 75 f1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 00 00 46 c5 80 80 15
2024-10-05 16:33:03.987 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] RECD: a5 10 00 10 15 ac 92 49 26 75 f1 02 01 78 1a 25 02 a5 0b 00 00 b8 41 dc 64 05 00 f2 15
2024-10-05 16:33:03.988 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-05 16:33:04.589 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 attempts left: 2
2024-10-05 16:33:04.590 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] SENT: a5 17 00 10 45 ad 00 49 26 75 f1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 00 00 46 c5 80 81 15
2024-10-05 16:33:05.708 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] RECD: a5 10 00 10 15 ad 93 49 26 75 f1 02 01 79 1a 25 02 a6 0b 00 00 b8 41 dc 64 05 00 f6 15
2024-10-05 16:33:05.708 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-05 16:33:06.509 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Querying 0512 - 0581 | 0x0200 - 0x0245 # 070 attempts left: 1
[... cut ...]
2024-10-05 16:33:07.775 DEBUG (MainThread) [custom_components.solarman.api] [4050986569] Request failed. [Previous State: Disconnected (-1)]
2024-10-05 16:33:07.775 INFO (MainThread) [custom_components.solarman.api] [4050986569] Disconnecting from 192.168.1.10:8899
2024-10-05 16:33:07.781 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching ME3000SP data in 8.071 seconds (success: False)
2024-10-05 16:33:07.781 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-10-05 16:33:07.781 INFO (MainThread) [custom_components.solarman.api] [4050986569] Disconnecting from 192.168.1.10:8899

davidrapan, 2024-10-05:
According to the #85 *wifikit file should work though.

You can try probe registers from 512 - 581 range using built-in service for reading them and see if you can pin-point some specific ones which are causing troubles.

closed 2024-11-19 by github-actions[bot]