davidrapan/ha-solarman #194: Datatime Problem

CrazyUs3r, 2024-10-06:
I downloaded everything new this morning and noticed that datetime was showing an error

TypeError: strptime() argument 1 must be str, not datetime.datetime

here is my suggestion

    @property
    def native_value(self) -> datetime | None:
        """Return the value reported by the datetime."""
        if not self._attr_native_value:
            return None

        if isinstance(self._attr_native_value, datetime):
            if self._attr_native_value.tzinfo is None:
                return self._attr_native_value.replace(tzinfo=ZoneInfo(self.coordinator.hass.config.time_zone))
            return self._attr_native_value

        return datetime.strptime(self._attr_native_value, '%y/%m/%d %H:%M:%S').replace(tzinfo=ZoneInfo(self.coordinator.hass.config.time_zone))

davidrapan, 2024-10-06:
[refactor: Moved Date/Time formats into constants](https://github.com/davidrapan/ha-solarman/commit/adce259cfb119b876ee45e39c9d038023ed03984) should also fix it I guess?

CrazyUs3r, 2024-10-06:
I try it later

CrazyUs3r, 2024-10-06:
it works, if I enter e.g. 4 pm it will be written 2 pm

davidrapan, 2024-10-06:
I'm not able to reproduce it.

davidrapan, 2024-10-06:
Replace method in datetime.py at line ~64 with:
```
    async def async_set_value(self, value: datetime) -> None:
        """Change the date/time."""
        _LOGGER.warning(value)
        _LOGGER.warning(value.astimezone(ZoneInfo(self.coordinator.hass.config.time_zone)))
        _LOGGER.warning(get_dt_as_list_int(value, self._multiple_registers))
        _LOGGER.warning(get_dt_as_list_int(value.astimezone(ZoneInfo(self.coordinator.hass.config.time_zone)), self._multiple_registers))
        #if await self.coordinator.inverter.call(CODE.WRITE_MULTIPLE_HOLDING_REGISTERS, self.register, get_dt_as_list_int(value, self._multiple_registers), ACTION_ATTEMPTS_MAX) > 0:
        #    self.set_state(value.strftime(DATETIME_FORMAT))
        #    self.async_write_ha_state()
```
Restart HA, try to set datetime and show me what's in the log.

davidrapan, 2024-10-06:
Do you have correct time zone setup in your pc/phone?

CrazyUs3r, 2024-10-06:
at my windows laptop is it correct and my phone the same, in ha log is the correct time from ha

davidrapan, 2024-10-06:
In my case values between rows 1 and 2 are not different... so I don't understand why they are in yours...

davidrapan, 2024-10-06:
How is your HA installed?

CrazyUs3r, 2024-10-06:
my ha core is installed under proxmox on a mini server every log from ha has the correct locale time

davidrapan, 2024-10-06:
What if you try to set datetime using Developer tools > ACTIONS > Date/Time: Set date/time?

CrazyUs3r, 2024-10-06:
I use the config from the yaml
[image]

davidrapan, 2024-10-06:
Sure but try it using ACTIONS too and tell me result.

CrazyUs3r, 2024-10-06:
There is nothing in the log that I can send you, and the time that is read does not change

davidrapan, 2024-10-06:
What? You did not do it properly then. Nvm I found the issue, it looks like it's due to a bug in the HA.

davidrapan, 2024-10-06:
[fix: Inconsistent behavior of datetime set_value](https://github.com/davidrapan/ha-solarman/commit/d409ea9255128fadc15598e742fa57a8151f9fb6)

CrazyUs3r, 2024-10-06:
I'll test it tomorrow, there's no sun today but then I'll see tomorrow morning what other errors there are. I've already loaded the datetime

CrazyUs3r, 2024-10-06:
But now i See

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

'''
Logger: homeassistant
Quelle: custom_components/solarman/datetime.py:62
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 18:42:19 (3 Vorkommnisse)
Zuletzt protokolliert: 18:45:23

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 44, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 62, in native_value
    return datetime.strptime(self._attr_native_value, DATETIME_FORMAT).replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '48/0/0 0:0:0' does not match format '%y/%m/%d %H:%M:%S'
'''

And

'''
Logger: homeassistant
Quelle: components/sensor/__init__.py:636
Erstmals aufgetreten: 18:42:13 (3 Vorkommnisse)
Zuletzt protokolliert: 18:44:19

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 44, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 636, in state
    raise ValueError(
ValueError: Sensor sensor.solarmanm80_device provides state value '', which is not in the list of options provided
'''

davidrapan, 2024-10-06:
I guess this was during shutdown of the inverter?

CrazyUs3r, 2024-10-06:
I assume so too, I just wanted to show you. It hasn't lost WiFi in the meantime and there was no other log entry either.

davidrapan, 2024-10-06:
Sure! I'll look where would be best to put some additional checks so it doesn't happen even during that. Thanks! 😉

CrazyUs3r, 2024-10-07:
So to day erros

Logger: homeassistant
Quelle: components/sensor/__init__.py:636
Erstmals aufgetreten: 07:44:55 (20 Vorkommnisse)
Zuletzt protokolliert: 16:35:26

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 44, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 636, in state
    raise ValueError(
ValueError: Sensor sensor.solarmanm80_device provides state value '', which is not in the list of options provided

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant
Quelle: custom_components/solarman/datetime.py:62
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 07:43:24 (2 Vorkommnisse)
Zuletzt protokolliert: 07:45:00

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 44, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 62, in native_value
    return datetime.strptime(self._attr_native_value, DATETIME_FORMAT).replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
[... 13 lines cut ...]
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1366, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 62, in native_value
    return datetime.strptime(self._attr_native_value, DATETIME_FORMAT).replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '48/0/0 0:0:0' does not match format '%y/%m/%d %H:%M:%S'

closed 2024-10-12 by davidrapan