davidrapan/ha-solarman #195: Avoid logging sensitive data

frsantos, 2024-10-06:
**Is your feature request related to a problem? Please describe.**
A typical request when debugging an issue is to add the logs to the github issue. However, the serial number is repeated many times in the logs.

The serial number is the unique identification of the inverter/logger and so it should be considered as sensitive data.

Is the serial number also appearing encoded in the responses? Can it be redacted/filtered too in the logs?

**Describe the solution you'd like**
Don't log the serial number in the logs, either in plain or encoded in the request . Use a different identifier if needed, maybe a HA identifier different than the SN?

**Describe alternatives you've considered**
* No log at all of the SN
* Redact the SN replacing last 6-7 digits by `X`
* Use a different ID associated to the SN

**Additional context**
Example of (redacted) log

```
2024-10-06 12:59:24.999 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-08-28T19:57:40.207144+00:00', 'data': {'name': 'KSTAR', 'inverter_discovery': True, 'inverter_host': '192.168.1.134', 'inverter_serial': 2355XXXXXX, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'kstar_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J6D91CZFTSW8Z5JXA2TJG2KE', 'minor_version': 1, 'modified_at': '2024-08-28T19:57:40.207151+00:00', 'options': {'name': 'KSTAR', 'inverter_discovery': True, 'inverter_host': '192.168.1.134', 'inverter_serial': 2355XXXXXX, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'kstar_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'KSTAR', 'unique_id': None, 'version': 1})
2024-10-06 12:59:24.999 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-10-06 12:59:25.000 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 192.168.1.134
2024-10-06 12:59:25.000 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-10-06 12:59:25.011 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.1.134, 289C6E8569EC, 2355XXXXXX]
2024-10-06 12:59:25.308 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 60, code: 4, min_span: 25, digits: 6
2024-10-06 12:59:25.309 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '28:9c:6e:85:69:ec')}, 'identifiers': {('solarman', 2355XXXXXX)}, 'name': 'KSTAR', 'manufacturer': 'KSTAR', 'model': 'Hybrid Inverter', 'serial_number': 2355XXXXXX}
2024-10-06 12:59:25.309 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-10-06 12:59:25.310 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Scheduling 7 query requests. #0
2024-10-06 12:59:25.310 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 ...
2024-10-06 12:59:25.311 INFO (MainThread) [custom_components.solarman.api] [2355XXXXXX] Connecting to 192.168.1.134:8899
2024-10-06 12:59:25.322 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] SENT: a5 17 00 10 45 82 00 e8 56 61 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 00 54 00 01 70 1a ff 15
2024-10-06 12:59:25.411 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] RECD: a5 13 00 10 15 82 90 e8 56 61 8c 02 01 eb bd 99 38 3f 26 00 00 a2 af 68 2e 01 84 02 c2 c1 47 15
2024-10-06 12:59:25.412 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 failed. [IllegalDataAddressError:  The data address received in the request is not an allowable address for the server.]
2024-10-06 12:59:25.613 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 attempts left: 4
2024-10-06 12:59:25.613 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] SENT: a5 17 00 10 45 83 00 e8 56 61 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 00 54 00 01 70 1a 00 15
2024-10-06 12:59:25.661 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] RECD: a5 13 00 10 15 83 91 e8 56 61 8c 02 01 ec bd 99 38 3f 26 00 00 a2 af 68 2e 01 84 02 c2 c1 4a 15
2024-10-06 12:59:25.661 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 failed. [IllegalDataAddressError:  The data address received in the request is not an allowable address for the server.]
2024-10-06 12:59:26.063 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 attempts left: 3
2024-10-06 12:59:26.063 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] SENT: a5 17 00 10 45 84 00 e8 56 61 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 00 54 00 01 70 1a 01 15
2024-10-06 12:59:26.161 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] RECD: a5 13 00 10 15 84 92 e8 56 61 8c 02 01 ec bd 99 38 3f 26 00 00 a2 af 68 2e 01 84 02 c2 c1 4c 15
2024-10-06 12:59:26.162 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 failed. [IllegalDataAddressError:  The data address received in the request is not an allowable address tfor he server.]
2024-10-06 12:59:26.764 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 attempts left: 2
2024-10-06 12:59:26.764 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] SENT: a5 17 00 10 45 85 00 e8 56 61 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 00 54 00 01 70 1a 02 15
2024-10-06 12:59:26.861 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] RECD: a5 13 00 10 15 85 93 e8 56 61 8c 02 01 ed bd 99 38 40 26 00 00 a2 af 68 2e 01 84 02 c2 c1 50 15
2024-10-06 12:59:26.862 DEBUG (MainThread) [custom_components.solarman.api] [2355XXXXXX] Querying 0084 - 0084 | 0x0054 - 0x0054 # 001 failed. [IllegalDataAddressError:  The data address received in the request is not an allowable address for the server.]
```

frsantos, 2024-10-06:
Not sure if categorize this as `enhancement` or as `bug`

davidrapan, 2024-10-06:
Serial number is really the best way how to identify stick logger and for example discovery would not even work without it.

**And it's not that sensitive though. One packet using UDP protocol to the network broadcast address and the stick logger will very willingly tell you it's serial number.**

Even identifying devices in the log using first 4 digits is not sufficient. I already saw even first 5 and 6 the same for multiple devices in single setup. 😉

**So the best solution is to do the privacy as postprocess.**

frsantos, 2024-10-06:
> Serial number is really the best way how to identify stick logger and for example discovery would not even work without it.

That does not mean it has to be dumped to the logs

> 
> **And it's not that sensitive though. One packet using UDP protocol to the network broadcast address and the stick logger will very willingly tell you it's serial number.**

If you have someone sniffing into your network, that's a bigger issue. But posting the SN into a public Github repository means someone may request the inverter provider things they can't do if they don't know the SN, like updates and configuration changes from the server side.

> 
> Even identifying devices in the log using first 4 digits is not sufficient. I already saw even first 5 and 6 the same for multiple devices in single setup. 😉

Maybe the integration can use an internal ID associated to the SN?

> 
> **So the best solution is to do the privacy in post-process.**
By redacting the SN as I did? Not everyone is aware that is needed.

davidrapan, 2024-10-06:
> That does not mean it has to be dumped to the logs

SENT and RECD are generated in used library so any changes in in integration would not be enough anyway.

> If you have someone sniffing into your network, that's a bigger issue. But posting the SN into a public Github repository means someone may request the inverter provider things they can't do if they don't know the SN, like updates and configuration changes from the server side.

If SN is enough for that it's issue with the manufacturers and not the integration.

> Maybe the integration can use an internal ID associated to the SN? I mean, this is just for logs, we can even use "N1", "N2", ... "Nm" to identify the devices, as long as there is a way to know which is N1 and which is N2. Even the Device name given in the integration could work, as that is the way the rest of entities are created.

That's really easy to imagine and very hard to sufficiently implement. I can easily imagine code solving this having more rows than rest of the integration. 😆

**Not even talking about the act of changing the unique id would be hell of a breaking change.**

> By redacting the SN as I did? Not everyone is aware that this is needed.

Yes.

frsantos, 2024-10-06:
> > That does not mean it has to be dumped to the logs
> 
> SENT and RECD are generated in used library so any changes in integration would not be enough anyway.
> 

Yes, I see redacting the SN in the messages would be hard to do, unless it's done by the library itself.

> If SN is enough for that it's issue with the manufacturers and not the integration.

Still, that does not mean the integration should facilitate it. It should avoid  that case as hard as it can. The less sensitive data exposed by the integration, the less risk to the user.

> That's really easy to imagine and very hard to sufficiently implement. I can easily imagine code solving this having more rows than rest of the integration. 😆

Replacing `[2355XXXXXX]` by `[KSTAR]` in the logs? I don't know the code, but it seems plain easy to me.

> 
> **Not even talking about the act of changing the unique id would be hell of a breaking change.**

Why would it be a breaking change to change how something is logged, when by definition logs are transient data not affecting the functionality of the integration? I'm not talking about changing how the integration works, but how the integration logs. You can have a different ID, generated from the SN or the device name (or even the device name) just for logging.

> 
> > By redacting the SN as I did? Not everyone is aware that this is needed.
> 
> Yes.

That's not a proper solution to me.

> Edit: Even the communication with solarman servers is not secured so for example your ISP knows the serial number too. 😉

Solarman servers should commuicate using HTTPS, so the ISP should not know the traffic between by inverter and an external company, same as they don't know your bank credentials. And even if that was the case, that does not mean that the integration should also leak that information.

davidrapan, 2024-10-06:
> Replacing [2355XXXXXX] by [KSTAR] in the logs? I don't know the code, but it seems plain easy to me.

But what about those how has 2 devices 3, or X devices? [KSTAR] [KSTAR] [DEYE], ... would render the logs useless.

davidrapan, 2024-10-06:
> Solarman servers should commuicate using HTTPS

Not really true for 100% of cases, sadly.

davidrapan, 2024-10-06:
`KSTAR_2355XXXXXX` is used to generate entities.

frsantos, 2024-10-06:
> `KSTAR_2355XXXXXX` is used to generate entities.

[image]

davidrapan, 2024-10-06:
What you see is entity_id. `KSTAR_2355XXXXXX` is used in unique_id.

davidrapan, 2024-10-06:
That just means that if you create device with same name it uses the unique_id to understand that it has to create entities as KSTAR_PV1_Power_2 (Add _2).

frsantos, 2024-10-06:
And that there are 2 entities with the same name that can't be user in automations.

davidrapan, 2024-10-06:
Yes it won't let you create two entities with the same entity_id.

davidrapan, 2024-10-06:
Also entity_id can be changed. unique_id can't.

frsantos, 2024-10-06:
> That just means that if you create device with same name it uses the unique_id to understand that it has to create entities as KSTAR_PV1_Power_2 (Add _2).

So, can we add a single incremental id to the device, same as the entities, so we log "KSTAR", "KSTAR_2" ?  As I said previously, logs are transient data, so having this changing over time seems no problem to me if it's only on the logs.

davidrapan, 2024-10-06:
You are looking at it from wrong perspective. In the point where you want to use KSTART placeholder, you, in the integration, don't know about other devices. So you don't know if KSTAR or KSTART_1 exist.

**Devices don't know about other devices.**

frsantos, 2024-10-06:
> You are looking at it from wrong perspective. In the point where you want to use KSTAR placeholder, you, in the integration, don't know about other devices. So you don't know if KSTAR or KSTAR_1 exists.
> 
> **Device don't know about other devices.**

(Bummer) :disappointed:

davidrapan, 2024-10-06:
And I was really thoroughly investigation this matter during implementation of advanced discovery (when you are adding multiple new devices in the row the configuration window is always prefilled with serial - host pair of new not yet added device). And during that process would be awesome to know about other devices... I had to solve it using serial number as device_unique_id (different from unique_id) and try to set it and if it fails then try next one. And that's how I'm able to do that in the end. Even HA in the docs suggests you to use serial number there and so on...

And really given the fact that security is really like the least problem for the solarman I also did not want to waste any more energy solving it for them, you know what I mean?

davidrapan, 2024-10-06:
@frsantos try to use wireshark and sniff the communication from the inverter directed at port 10000. You will be very (not pleasantly) surprised....

PV-Joe, 2024-10-06:
> ... try to use wireshark and sniff the communication from the inverter directed at port 10000. You will be very (not pleasantly) surprised...

I heard that communication is not secure. Is there anything we can do to have it secured?

davidrapan, 2024-10-06:
**The only thing you can do is to completely disable communication with the servers.**

frsantos, 2024-10-06:
> @frsantos try to use wireshark and sniff the communication from the inverter directed at port 10000. You will be very (not pleasantly) surprised...
> 
> As I said ISP knows your serial number... And not just that but also ssid your wifi... firmware of the stick... etc.

I'm not available to capture the traffic from the server right now, but anyway, that is unrelated to this issue. Even if the inverter communicates with the solarman server using HTTP and not HTTPS, the integration is still writing the SN to the logs and the recommendation is to post those logs into this public repository.

There are multiple levels where security can be improved. One is the inverter/server communication. Other is the network security, that is, allowing others to sniff into the local network. And a different one is to recommend people to post their SN into a public repository. We cannot do anything about the first two issues in this integration, but we can solve the last.

githubDante, 2024-10-06:
What about a tool that obfuscates the serial numbers in the log files. It won't be automatic, but for users who want to hide them, it won't be a problem (I think)  to use an extra step before uploading the log files to github.

githubDante, 2024-10-06:
Created a [Gist](https://gist.github.com/githubDante/565a59237f21aab22025a379ce80a2b3) with such tool. It was tested with several log files which can be found here.

davidrapan, 2024-10-07:
Yeah, I think that's the most appropriate solution!

closed 2025-12-24 by davidrapan