davidrapan/ha-solarman #197: INVT inverter not supported

MrLopHA, 2024-10-06:
Hello everyone!

I have just installed an INVT inverter, model XD6KTL to manage my solar installation and I need to integrate it in HA to be able to charge my car with the energy produced by my installation.

My inverter is very similar to those of the DEYE brand (they have the same configuration page, access credentials and even their logos are similar).

Right now I can only visualise the production and consumption data through the SOLARMAN app. I have tried this integration (and another one in HACS) without success (and I have tried all the profiles).

I can't believe I'm the only INVT user who wants to use Home Assistant, but I feel like I'm preaching in the wilderness.

I have written an email to SOLARMAN, and to INVT but so far no news.

I hope to get more luck through this great community.

Thank you very much in advance

githubDante, 2024-10-06:
```java
 if (ulDataAddr == 4660) {
....
                    InverterCommuTask.this.m_map.put("invt_serialno", serial);
```
Any response from register 4660?!? The length should be 6.

PV-Joe, 2024-10-07:
> Right now I can only visualise the production and consumption data through the SOLARMAN app.

That sounds promising.

Maybe it helps if you provide the logfile and post it here for the experts to analyse?

Please let us know what configuration profile you are using and if you have any documentation of the modbus addresses of the inverter.

MrLopHA, 2024-10-07:
> That sounds promising.
>
> Maybe it helps if you provide the logfile and post it here for the experts to analyse?
>
> Please let us know what configuration profile you are using and if you have any [...]

Hi again PV-Joe, thanks for continuing to try to help me also for this post.

I have used several profiles of the integration and the ones that have taken the longest to give an error (it seemed that it was going to connect) has been with the Afore investor profile. For example with the Deye_hybrid.yaml profile and with almost all the others, it gives this error:

[image]

I found on the internet a PDF with the modbus protocols of the INVT inverter (XD and XG series), I pass it to you here as well.

[INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf](https://github.com/user-attachments/files/17279845/INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf)

I've put a log below, I hope it helps.

> 2024-10-07 15:39:24.066 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-07T13:35:19.421393+00:00', 'data': {'name': 'INVT', 'inverter_host': '192.168.1.159', 'inverter_serial': 35XXXXXX18, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'solarman_dtsd422-d3.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J9KK21VXDK76YQ180XBZMTP2', 'minor_version': 1, 'modified_at': '2024-10-07T13:37:52.320132+00:00', 'options': {'inverter_host': '192.168.1.159', 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'afore_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'INVT', 'unique_id': 'solarman_35XXXXXX18', 'version': 1})
2024-10-07 15:39:24.067 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-10-07 15:39:24.067 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 192.168.1.159
2024-10-07 15:39:24.067 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-10-07 15:39:24.113 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.1.159, D427870BA20E, 35XXXXXX18]
2024-10-07 15:39:24.268 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 4, min_span: 25, digits: 6
2024-10-07 15:39:24.269 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', 'd4:27:87:0b:a2:0e')}, 'identifiers': {('solarman', 35XXXXXX18)}, 'name': 'INVT', 'manufacturer': 'Afore', 'model': 'HYBRID', 'serial_number': 35XXXXXX18}
2024-10-07 15:39:24.269 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-10-07 15:39:24.270 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Scheduling 4 query requests. #0
2024-10-07 15:39:24.270 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 ...
2024-10-07 15:39:24.270 INFO (MainThread) [custom_components.solarman.api] [35XXXXXX18] Connecting to 192.168.1.159:8899
2024-10-07 15:39:24.294 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] SENT: a5 17 00 10 45 12 00 f6 d0 ff d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 01 fb 00 36 00 11 5e 15
2024-10-07 15:39:30.610 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] RECD: a5 10 00 10 15 12 51 f6 d0 ff d1 02 01 9b b2 03 00 42 02 00 00 f7 31 00 67 05 00 59 15
2024-10-07 15:39:30.611 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-07 15:39:30.813 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 attempts left: 4
2024-10-07 15:39:30.813 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] SENT: a5 17 00 10 45 13 00 f6 d0 ff d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 01 fb 00 36 00 11 5f 15
2024-10-07 15:39:36.992 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] RECD: a5 10 00 10 15 13 52 f6 d0 ff d1 02 01 a1 b2 03 00 49 02 00 00 f7 31 00 67 05 00 68 15
2024-10-07 15:39:36.993 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-07 15:39:37.394 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 attempts left: 3
2024-10-07 15:39:37.395 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] SENT: a5 17 00 10 45 14 00 f6 d0 ff d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 01 fb 00 36 00 11 60 15
2024-10-07 15:39:43.647 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] RECD: a5 10 00 10 15 14 53 f6 d0 ff d1 02 01 a8 b2 03 00 4f 02 00 00 f7 31 00 67 05 00 77 15
2024-10-07 15:39:43.647 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 failed. [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-10-07 15:39:44.249 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 attempts left: 2
[... cut ...]
2024-10-07 15:39:59.611 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] Querying 0507 - 0560 | 0x01FB - 0x0230 # 054 ...
2024-10-07 15:39:59.612 INFO (MainThread) [custom_components.solarman.api] [35XXXXXX18] Connecting to 192.168.1.159:8899
2024-10-07 15:39:59.654 DEBUG (MainThread) [custom_components.solarman.api] [35XXXXXX18] SENT: a5 17 00 10 45 61 00 f6 d0 ff d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 01 fb 00 36 00 11 ad 15
> 

Thank you very much for your help! Hopefully we will reach a successful conclusion soon.

MrLopHA, 2024-10-07:
He visto también que en la web del inversor, existe una opción para conectarse a un servidor a través de TCP/UTP. He intentado conectarme a Home Assistant y recibir datos via Node Red con nodos de TCP in, pero no he tenido éxito. 

[image]

githubDante, 2024-10-07:
@MrLopHA the registers for this inverter (checked few from the PDF) don't match any of the definition files here, so you'll have to start from scratch and make your own YAML file.

PV-Joe, 2024-10-07:
> Right now I can only visualise the production and consumption data through the SOLARMAN app.

What profile does show production and consumption?

I suggest you start with that profile and delete all the sensors except for these two sensors.

After that add one modbus register after the other from your modbus.pdf and check if they are working.

Any other suggestion @davidrapan?

MrLopHA, 2024-10-07:
> 
> What profile does show production and consumption?
> 
> I suggest you start with that profile and delete all the sensors except for these two sensors.
> 
> After that add one modbus register after the other from your modbus.pdf and check if they are working.
> 
> Any other suggestion @davidrapan?

> Right now I can only visualise the production and consumption data through the SOLARMAN app.
> 

I don't think I've made myself clear, I don't get this data from the HA integration, but from the SOLARMAN app I have on my iPhone. 

None of the profiles collected by the HA integration work with my inverter. 

My hope right now is to decipher the information I manage to collect from the inverter via TCP. I agree with @githubDante that those codes are not from the inverter, I think they are the codes that the SOLARMAN API uses to run the app on my iPhone.

davidrapan, 2024-10-07:
It's quite simple. You just have to copy for example `inverter_definitions/deye_hybrid.yaml` into `inverter_definitions/custom/invt_hybrid.yaml`

And start changing registry address for all of the sensors acording the document you posted here before:

> I found on the internet a PDF with the modbus protocols of the INVT inverter (XD and XG series), I pass it to you here as well.
[INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf](https://github.com/user-attachments/files/17279845/INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf)

And then configure device using this new profile.

MrLopHA, 2024-10-08:
[image]

[image]

La respuesta de @davidrapan fue esclarecedora, comprendí como crear mi propio archivo customizado para mi inversor. Por otro lado, encontré un código para otro inversor de INVT aquí: 

https://community.home-assistant.io/t/ha-integration-invt-xg50ktr-lsw5/681962/4

Con ese código el inversor la conexión se produce y obtengo datos, pero debo revisarlo bien porque no todos funcionan con mi inversor, pero creo que en unos dias lo tendré listo y funcionando al 100%, Si lo consigo, naturalmente, el archivo final estará a disposición del staff por si quisiera añadirlo al repositorio.

PV-Joe, 2024-10-08:
Very good! I think adding another supported inverter will be a great enhancement to the compatible list and will help other users.

Make sure to use the standardized entityID's to stay compatible with future updates.

MrLopHA, 2024-10-10:
Hello again friends, I have created some new entities related to the battery. but I can't quite understand the Modbus code list and I can't find some values that I think are essential, for example:

Current battery discharge power; Current battery charge power and Total solar generation power.

I would be very grateful if you could have a look at the PDF and help me to find these data.

> [INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf](https://github.com/user-attachments/files/17279845/INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf)
> 

I also leave you the .yaml file I'm working on in case you want to have a look at it. Thank you very much again

[INVT_inverter.txt](https://github.com/user-attachments/files/17321188/INVT_inverter.txt)

githubDante, 2024-10-10:
> Hello again friends, I have created some new entities related to the battery. but I can't quite understand the Modbus code list and I can't find some values that I think are essential, for example:
> 
> Battery current discharge power; Battery current charge power and Total solar generation power.
> 
> I would be very grateful if you could have a look at the PDF and help me to find these data.
> 
> > [INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf](https://github.com/user-attachments/files/17279845/INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf)
> 
> I also leave you the .yaml file I'm working on in case you want to have a look at it. Thank you very much again
> 
> [INVT_inverter.txt](https://github.com/user-attachments/files/17321188/INVT_inverter.txt)

The total solar power can be calculated. It's the sum of the below:
```python
[(string.pv_current * string.pv_voltage) for string in inverter.strings]
```

As for the battery the PDF clearly states that the value for register 0x3908 is signed. When the value > 0 the battery is being charged and is discharged when the value is below zero.

MrLopHA, 2024-10-10:
> The total solar power can be calculated. It's the sum of the below:
>
> ```python
> [(string.pv_current * string.pv_voltage) for string in inverter.strings]
> ```
>
> As for the battery the PDF clearly states [...]

Exacto, la potencia resulta de aplicar intensidad por voltaje. He intentado crear dos sensores (tengo dos strings) para después, sumarlos y obtener la potencia total de producción solar. 

sensor:
  - platform: template
    sensors:
      pv01_power:
        friendly_name: "PV01 Power"
        unit_of_measurement: "kW"
        value_template: "{{ (states('sensor.inverter_pv01_current') | float * states('sensor.inverter_pv01_voltage') | float) / 1000 }}"
        icon_template: 'mdi:solar-power'

      pv02_power:
        friendly_name: "PV02 Power"
        unit_of_measurement: "kW"
        value_template: "{{ (states('sensor.inverter_pv02_current') | float * states('sensor.inverter_pv02_voltage') | float) / 1000 }}"
        icon_template: 'mdi:solar-power'
`
Pero al incorporar este código en configutation.yaml la integración deja de funcionar, da error de conexión constantemente.

[image]

Respecto a la batería el valor 0x3908 muestra un valor que no tiene lógica.

[image]

MrLopHA, 2024-10-10:
> You don't have to calculate it yourself, @githubDante told you the register which holds `Battery Power` value.

I know. Maybe I didn't explain myself well. 

`Battery Power`Is included in this file, but it doesn't work well.

>I also leave you the .yaml file I'm working on in case you want to have a look at it. Thank you very much again
[INVT_inverter.txt](https://github.com/user-attachments/files/17321188/INVT_inverter.txt)

On the other hand, I want to obtain the total power that the sun produces, and I must calculate that myself. But I can't do it either :(

davidrapan, 2024-10-10:
> `Battery Power` Is included in this file, but it doesn't work well.

Because it needs rule: 2 not a rule: 1.

githubDante, 2024-10-10:
For the battery the rule must be `2`, and the datatype `uint16` 
As for the total PV power you will need something like 
```yaml
        registers:
          [
            0x0001,
            0x0002,
            0x0003,
            0x0004,
            0x0005,
            0x0006,
            0x0007,
            0x0008,
            0x0009,
            0x000a,
          ]
        sensors:
          - registers: [0x0001]
            multiply:
              registers: [0x0002]
        - registers: [0x0003]
            multiply:
              registers: [0x0004]
   ```

githubDante, 2024-10-10:
@davidrapan why the parser has double `try_parse_unsigned`  for rules 1 & 3 ?

davidrapan, 2024-10-10:
Nooooooo idea, this is a remnant of Stephan's integration, which I took inspiration from in the very beginning.

I'm guessing that the intention at the beginning was that 3 and 4 were for > 16b and later this behavior went to 1 and 2 as well.

PV-Joe, 2024-10-31:
@MrLopHA 
Have you been successful getting your inverter to work or did you give up trying?

MrLopHA, 2024-10-31:
Hi 

> @MrLopHA Have you been successful getting your inverter to work or did you give up trying?

Hi! 

I have managed to get many parameters to work, but many others are still missing. 

For me they are enough for the moment, although I will continue to investigate. 

[image]
[image]
[image]

davidrapan, 2024-10-31:
You could then share your file and I can add it to the repository. 😉

MrLopHA, 2024-10-31:
Sure! 

How can I send it to you?

davidrapan, 2024-10-31:
You can upload it right here, just drop the file inside the textbox, it will upload. 😉

MrLopHA, 2024-11-01:
[INVT_XD6KTL.txt](https://github.com/user-attachments/files/17598672/INVT_XD6KTL.txt)

Done!

MrLopHA, 2024-11-02:
@davidrapan 

I forgot to attach the updated file with the MODBUS protocol of the inverter.

Now, you can to update the documentation folder in your integration if you want.

davidrapan, 2024-11-02:
Good! Thank you!

davidrapan, 2024-11-07:
[feat: Add INVT XD3-6KTL profile](https://github.com/davidrapan/ha-solarman/commit/3d686c7b0c3e7fe972993b85a9203fc65a843dce)

w/ some fixes, added crucial sensors and renames to meet our standards

MrLopHA, 2024-11-10:
Hi @davidrapan 

Thank you very much for taking the trouble. Your file has been very helpful to me in improving the integration of my inverter, however I have been testing and reviewing your file and I have come to these conclusions:

1- This code for LOAD POWER don't work

```
- name: "Load Power"
     class: "power"
     state_class: "measurement"
     uom: "kW"
     scale: 0.1
     rule: 1
     registers: [0x381C]
```
We have calculated (Hello, Andrés ;-) ) this data by adding the active power given by the inverter in `0x381D` and the meter measurement given in `0x3827` and it works.

```
- name: "Load Power"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 0.1
      rule: 4
      digits: 0
      registers: [0x381E, 0x381D, 0x3828, 0x3827]
      uint: enforce
      sensors:
        - signed:
          registers: [0x381E, 0x381D]
        - operator: add
          signed:
          registers: [0x3828, 0x3827]
      icon: 'mdi:home-lightning-bolt'
```
2- We have created the GRID POWER sensor with the data from the METER (in my case the METER measures the input of the electrical supply of my house). I have a Shelly installed and the data from both sensors match.

```
    - name: "Grid Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 0.1
      rule: 4
      registers: [0x3828, 0x3827]
      icon: 'mdi:meter-electric'
```

3- Calling POWER to OUTPUT ACTIVE POWER I think can lead to confusion. I prefer to leave OUTPUT ACTIVE POWER in my code, but I understand that you prefer to change it and adjust it to a common style. 

4- The POWER LOSSES sensor doesn't work, we have called this sensor INVERTER LOSSES and used this code:

```
    - name: "Inverter Losses"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 0.1
      rule: 1
      digits: 0
      registers: [0x3B01, 0x3B02, 0x381E, 0x381D, 0x3909, 0x3908]
      uint: enforce
      sensors:
        - registers: [0x3B01]
        - operator: add
          registers: [0x3B02]
        - operator: subtract
          signed:
          registers: [0x381E,0x381D]
        - operator: subtract
          signed:
          registers: [0x3909,0x3908]
      icon: 'mdi:home-lightning-bolt'
```
I have done some screenshots so you can see how both are shown:

[image]
Your sensor was renamed to appear next to ours and capture it.

[image]
Your sensor

[image]
Our sensor

This is the file I'm using now. It looks like this in the absence of giving it the standardised format of your integration.

https://github.com/user-attachments/assets/13a48aab-27c3-4073-b1ea-f1dbd9d4fdfe

[INVT_XD6KTL CUSTOM.txt](https://github.com/user-attachments/files/17692476/INVT_XD6KTL.CUSTOM.txt)

davidrapan, 2024-11-13:
I have to say I was little bit confused about some mentioned registers above cause the docs are not really clear about them.

> We have calculated (Hello, @andresindlp ;-) ) this data by adding the active power given by the inverter in 0x381D and the meter measurement given in 0x3827 and it works.

So the inverter does not have dedicated register for that?

> Calling POWER to OUTPUT ACTIVE POWER I think can lead to confusion. I prefer to leave OUTPUT ACTIVE POWER in my code, but I understand that you prefer to change it and adjust it to a common style.

Yes it's for the general scheme across inverters and also in reference to the HA. And "ACTIVE" is in this term redundant word. OUTPUT ACTIVE POWER and OUTPUT POWER are virtually synonymous terms.

> The POWER LOSSES sensor doesn't work, we have called this sensor INVERTER LOSSES and used this code:

The term INVERTER LOSSES (INVERTER & LOSSES, could be good candidate for the name though) could be little bit misleading. Power losses sensor contains Inverter Consumption and AC/DC conversion losses.

MrLopHA, 2024-11-18:
> So the inverter does not have dedicated register for that?

Of course...theoretically. But certainly, we have not been able to obtain any value from this register.

> The term INVERTER LOSSES (INVERTER & LOSSES, could be good candidate for the name though) could be little bit misleading. Power losses sensor contains Inverter Consumption and AC/DC conversion losses.

Thats right, but as in the previous case, the registry that should provide this data doesn´t work.

Changing the subject: 

Do you think it would be possible to add writing commands through this integration?

 I am thinking of being able to configure the priority of the generated energy (to the loads, to the battery or to the grid) or program the battery charging from the grid.

davidrapan, 2024-11-20:
So that means Battery Power values is inverted?

[fix: INVT XD-TL profile](https://github.com/davidrapan/ha-solarman/commit/acc150aaf944667f052186b5e7e287c8d2323742)

MrLopHA, 2024-11-22:
> So that means Battery Power value is inverted? Strange.
> 
> 
> 
> [fix: INVT XD-TL profile](https://github.com/davidrapan/ha-solarman/commit/acc150aaf944667f052186b5e7e287c8d2323742)

I don’t understand what you mean

I’ll try the code modifications and I’ll tell you

MrLopHA, 2024-11-26:
Today I have seen several errors in my registration:

The first:

>  Registrador: homeassistant.components.sensor
> Fuente: components/sensor/__init__.py:733
> integración: Sensor (documentación, problemas)
> Ocurrió por primera vez: 18:11:43 (1 ocurrencias)
> Última vez registrado: 18:11:43
>
>Entity sensor.invt_battery_temperature (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using native >unit of measurement 'ºC' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', 'K', >'°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at >https://github.com/davidrapan/ha-solarman/issues

This is what the code looks like in the configuration file:

[image]

I have changed ```uom: "ºC"``` to ```uom: "°C"``` (it looks the same but it is not). 

The second:

>Registrador: homeassistant.components.sensor
>Fuente: components/sensor/__init__.py:571
>integración: Sensor (documentación, problemas)
>Ocurrió por primera vez: 18:11:42 (2 ocurrencias)
>Última vez registrado: 18:11:43
>
>Entity sensor.invt_inverter_status (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class >'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your >configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha->solarman/issues
>Entity sensor.invt_battery_state (<class 'custom_components.solarman.sensor.SolarmanSensor'>) is using state class >'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your >configuration if your entity is manually configured, otherwise create a bug report at https://github.com/davidrapan/ha->solarman/issues

This is what the code looks like in the configuration file:

[image] 
[image]

I have deleted ```state_class: "measurement"``` in both

davidrapan, 2024-11-27:
[fix: INVT XD-TL profile](https://github.com/davidrapan/ha-solarman/commit/d7efa4d022d96bfa8635b21c6c4fd74fee679944)

closed 2025-01-11 by davidrapan