davidrapan/ha-solarman #199: New supported inverter (CHINT)

gardiol, 2024-10-07:
Hi!

just migrated from home_assistant_solarman!

It's a CHINT, a 6KW hybrid inverter, and i also have the chint-provided PDF file with the MODBUS mappings, where should i upload it to?

My inverter is supported, i manually wrote the fllowing yaml:

```
requests:
  - start: 0x1001
    end: 0x104C
    mb_functioncode: 0x03

  - start: 0x1080
    end: 0x1092
    mb_functioncode: 0x03

  - start: 0x1300
    end: 0x1362
    mb_functioncode: 0x03

  - start: 0x1A44
    end: 0x01A46
    mb_functioncode: 0x03

# Battery: 0x2000 - 0x2013

parameters:
 - group: info
   items:
    - name: "Rated Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x1A44]
      icon: 'mdi:solar-power'
    - name: "Rated Frequency"
      class: "frequency"
      state_class: "measurement"
      uom: "Hz"
      scale: 0.01
      rule: 1
      registers: [0x1A45]
      icon: 'mdi:solar-power'
    - name: "Rated Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x1A46]
      icon: 'mdi:solar-power'

 - group: solar
   items:
    - name: "Phase A Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x1001]
      icon: 'mdi:solar-power'
    - name: "Phase A Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
[... 687 lines cut ...]
      icon: 'mdi:solar-power'

    - name: "Daily energy to EPS"
      class: "energy"
      state_class: "measurement"
      uom: "Wh"
      scale: 10
      rule: 3
      registers: [0x1360]
      icon: 'mdi:solar-power'
    - name: "Accumulated energy to EPS"
      class: "energy"
      state_class: "measurement"
      uom: "Wh"
      scale: 10
      rule: 3
      registers: [0x1362]
      icon: 'mdi:solar-power'

# Battery SOC ... Battery acucmultaed discharge energy ... from 0x2000 to 0x2013
```

Now, it's pretty unclear to me HOW to integrate this to HA energy dashboard?

Some of those values are populated, others are always 0, but i am not knowledgeable enugh to understand what to do now?

PV-Joe, 2024-10-07:
> Now, it's pretty unclear to me HOW to integrate this to HA energy dashboard?

Add the Total sensors to energy dashboard and it should work.

davidrapan, 2024-10-12:
Device added to the list of [Supported Devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices)

closed 2024-10-12 by davidrapan

gardiol, 2024-10-12:
Beautiful!
Thanks!

PV-Joe, 2024-10-12:
Great work and congratulations for another supported device 👍

PV-Joe, 2024-10-12:
Is that a typo in your config file?

Frequencyr instead of Frequency?

gardiol, 2024-10-12:
Pretty sure its a typo ...

gardiol, 2024-10-12:
Since the file is going to be renamed anyway, maybe @davidrapan can directly fix it too

davidrapan, 2024-10-12:
Yeah I already noticed that. 😉