davidrapan/ha-solarman #207: Not change in deye inverter parameters

RomanLambr, 2024-10-11:
Maybe some can help me I added this SOLARMAN to my home assistance for automatically charge battery in depend from forecast weather. 
Added automatic action, but without any result nothing change on deye dashboard if I run auto or try to set soh parameters from home assistant. 
maybe I need add any addons or change some in my inverter

CrazyUs3r, 2024-10-11:
Which inverter do you have the one from deye

RomanLambr, 2024-10-12:
Deye 12 kw sg04lp03

*Roman Lambruh*
+421915529840

+380503727783

davidrapan, 2024-10-12:
You are probably doing something wrong, show me the yaml of automation you are using.

RomanLambr, 2024-10-12:
Please write me directly to my mail
***@***.***

I will be very appreciate if you will help me solve it

*Roman Lambruh*
+421915529840

+380503727783

RomanLambr, 2024-10-17:
Finaly today i found some part solution.
If i change number.inverter_program_1_soc manualy  from entery - charge parameters in Deye change well.
Problem that i can't undarstand how work automatisation, i dont have next sensore - 
sensor.solcast_pv_forecast_forecast_today, in my entery.

I made copy to Yaml for creating auto.

davidrapan, 2024-10-17:
This is my automation:
```
alias: Inverter Grid Charging Program
mode: single
triggers:
  - at: "03:00:10"
    trigger: time
  - at: "04:00:00"
    trigger: time
  - at: "05:00:00"
    trigger: time
actions:
  - sequence:
      - action: solcast.update_forecasts
        data: {}
      - action: number.set_value
        metadata: {}
        target:
          entity_id: number.inverter_program_1_soc
        data:
          value: "{{ states('sensor.inverter_battery_precharge') }}"
```
and `sensor.inverter_battery_precharge` is template sensor which has some `if - elif - else` statements according to the `sensor.solcast_pv_forecast_forecast_today`.

RomanLambr, 2024-10-17:
and for forecasts use other extension like this,

https://github.com/bjReplay/ha-solcast-solar

for whether forecast,

davidrapan, 2024-10-17:
Yes I do use solcast integration.