davidrapan/ha-solarman #210: deye hybrid time of use and smart load voltage mode

oguzatagan, 2024-10-13:
hi

First of all, thank you very much for developing this integration. I am using a deye hybrid inverter. I have installed your integration, everything is very nice and fast. Moreover, the control modes also work without using an extra RS485 to control the inverter, and thus we can control the inverter without needing extra hardware. You are great. But I have a problem. I have a handmade lithium battery. So I am using the inverter in voltage mode. I want you to support voltage mode when using smart load and time of use. Is this possible?

davidrapan, 2024-10-13:
Hi @oguzatagan, sure! If you give me a list of items you are missing I will add them.

davidrapan, 2024-10-14:
[v24.10.14](https://github.com/davidrapan/ha-solarman/releases/tag/v24.10.14)

closed 2024-10-14 by davidrapan