davidrapan/ha-solarman #211: Bug in log - Error doing job: Task exception was never retrieved (None)

Soltinioo, 2024-10-13:
Hi it's me again. Small error in the logs with the last version.

Rejestrator: homeassistant
Źródło: components/sensor/__init__.py:636
Pierwsze zdarzenie: 06:56:14 (5 zdarzenia)
Ostatnio zalogowany: 08:16:00

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 43, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 636, in state
    raise ValueError(
ValueError: Sensor sensor.falownik_deye_device provides state value '', which is not in the list of options provided

davidrapan, 2024-10-13:
Can you record debug log please?

Soltinioo, 2024-10-13:
updated to the latest version a few minutes ago. Another error in the logs. the previous one does not appear. I am attaching the log

Ten błąd pochodzi z niestandardowej integracji.

Rejestrator: homeassistant
Źródło: custom_components/solarman/entity.py:61
integracja: Solarman (dokumentacja, Problemy)
Pierwsze zdarzenie: 18:28:17 (2 zdarzenia)
Ostatnio zalogowany: 18:28:44

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 42, in _handle_coordinator_update
    self.update()
  File "/config/custom_components/solarman/entity.py", line 61, in update
    if self.sensor_name in self.coordinator.data and (data := self.coordinator.data[self.sensor_name]):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
[error_log-1.txt](https://github.com/user-attachments/files/17356408/error_log-1.txt)

davidrapan, 2024-10-13:
Thanks! I can proly see what's wrong so I'll fix it promptly! 😉

davidrapan, 2024-10-14:
[fix: Coordiantor returned NoneType on timeout](https://github.com/davidrapan/ha-solarman/commit/8b431a4443671fddc77dc1175de61ea32673a158)

closed 2024-10-18 by davidrapan