davidrapan/ha-solarman #215: Several errors since last update

psykokwak-com, 2024-10-15:
### Description

Hi all,
Since the last update, i have the following errors:

Enregistreur: homeassistant.components.sensor
Source: helpers/entity_platform.py:369
intégration: Capteur (documentation, problèmes)
S'est produit pour la première fois: 09:11:09 (1 occurrences)
Dernier enregistrement: 09:11:09

Error while setting up solarman platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 369, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 628, in async_add_entities
    for entity in new_entities:
  File "/config/custom_components/solarman/sensor.py", line 52, in <genexpr>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, battery_nominal_voltage, battery_life_cycle_rating), d) for d in descriptions if (is_platform(d, _PLATFORM) and not "configurable" in d))
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/entity.py", line 21, in create_entity
    entity = creator(description)
             ^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 52, in <lambda>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, battery_nominal_voltage, battery_life_cycle_rating), d) for d in descriptions if (is_platform(d, _PLATFORM) and not "configurable" in d))
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 37, in _create_entity
    return SolarmanSensor(coordinator, description, battery_nominal_voltage, battery_life_cycle_rating)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 86, in __init__
    if "name" in sensor and sensor["name"] == "Battery" and battery_nominal_voltage > 0 and battery_life_cycle_rating > 0:
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'

Cette erreur provient d'une intégration personnalisée

Enregistreur: custom_components.solarman.entity
Source: custom_components/solarman/entity.py:27
intégration: Solarman
S'est produit pour la première fois: 09:11:09 (1 occurrences)
Dernier enregistrement: 09:11:09

Configuring {'name': 'Battery', 'class': 'battery', 'state_class': 'measurement', 'uom': '%', 'rule': 1, 'registers': [184]} failed. [TypeError: '>' not supported between instances of 'NoneType' and 'int']

### Reproduction Steps

Just restart HA.
Default deye config.

### Log

See

### Version

24.10.14

davidrapan, 2024-10-15:
You added your device quite some time ago, am I right?

Anyway, thanks for the report!

[fix: Add default values to some configurables](https://github.com/davidrapan/ha-solarman/commit/868259d78c27a47cd05a3c339dde61801bb52b1e)

psykokwak-com, 2024-10-15:
I added it before switching on your plugin. I used the other one before.
The migration was ok but since the update it does not works.
I rollbacked on the previous update for the moment.

davidrapan, 2024-10-15:
If you remove the device and re-add it back again it will also fix the errors.

psykokwak-com, 2024-10-15:
Does it changes the names of the entities?

davidrapan, 2024-10-15:
Names of the entities depends on the profile used.

closed 2024-10-18 by davidrapan