davidrapan/ha-solarman #219: Todays totals wrong

MikealSouth, 2024-10-17:
### Description

For a couple of weeks/months the Integrations displays incorrect sums for "Today Production/Usage/etc"  for a device that's called "deye_12k".
Here is the "Today's Production" as a example.
[image]

[image]

Could someone please point me to what I am doing wrong?
The WiFi connectivity is excellent, also I am using the SG04LP3 yaml file.

### Reproduction Steps

Just gathering data, the current sensor data like production, consumption, battery SOC, seems to be fine.

### Log

[home-assistant_solarman_2024-10-17T09-18-25.668Z.log](https://github.com/user-attachments/files/17409493/home-assistant_solarman_2024-10-17T09-18-25.668Z.log)

### Version

24.10.16

davidrapan, 2024-10-17:
So that means that your device caused it.

You will have to correct your history using Developer tools >> STATISTICS >> Total Production >> Graph icon and find the time when this occurred and fix it.

MikealSouth, 2024-10-17:
OK, interesting, thank you.
Problem is that I can't seem to find the time that the error occured, pretty sure that this happened last month, then I look into the history of last month I can't find anything
[image]

I am currently trying to use the "Fix Issue" function as I seem to have tons of them.
[image]

davidrapan, 2024-10-17:
You have to correct it using that icon on the right. There you will see datetime picker and timestamps with values...

MikealSouth, 2024-10-17:
OK, thank you for all of your help but I need to bug you once more but this. 
Do I just need to find one occurance of this or the first one around a month ago?
I've tried to google this but can't really find the answer, sorry to bug you teaching me how troubleshoot HA.

Here is my example, yesterdays export to grid
[image]

I couldn't even fully charge my 14,6kWh battery, no way my system has exported +70kWh.
Do I just correct/guess these values or do I need to find the first occurance os these spikes?

Is there a way to just delete the historic values and just start from scratch?

davidrapan, 2024-10-17:
íYou need to fix all occasions when this fluke happened. So it depends of how many times it happened. If you click on that bottom left button called OUTLIERS or something like that and will be lucky it will show you all the occasions when irregular delta was detected and you fix them it should be enough.

So let's say you see for example in some time +13 kWh you correct it to 0 kWh and so on.

> Is there a way to just delete the historic values and just start from scratch?

Yes there is a way. If you delete the device from the integration and then again go into STATISTICS page and click on FIX ISSUE and then CONFIRM on all the total sensors which belong to that deleted device. You will essentially delete whole long term statistics (data used in the energy dashboard) of that particular sensor.

MikealSouth, 2024-10-17:
I used to have several of these events a day so I chose the "delete historic data" path.
Somehow after deleting the device, dev > statistic > search for device and "fix issues" the data came back as seen as I added that device again.
So I want the more hardcore route (after a full backup) and used the SQLite explorer
```
SELECT * FROM "statistics_meta" WHERE "statistic_id" = "sensor.deye_12k_today_production";
765 sensor.deye_12k_today_production;
DELETE FROM statistics WHERE metadata_id="765";
SELECT * FROM "statistics_meta" WHERE "statistic_id" = "sensor.deye_12k_total_production";
766 sensor.deye_12k_total_production;
DELETE FROM statistics WHERE metadata_id="766";
```

I will report back if that helped.
Thank you very much @davidrapan

davidrapan, 2024-10-17:
> Somehow after deleting the device, dev > statistic > search for device and "fix issues" the data came back as seen as I added that device again.

When you click on the sensor you see short term statistics. 😉 For energy dashboard are important long term statistics and those are deleted using procedure that procedure.

MikealSouth, 2024-10-17:
I totally understand your point and if I would have seen another way I would have taken that.
The only real long term statistic I really need is this:
[image]

Since April 2nd my home has used 114kWh from the grid and send 4498 kWh to the grid.