davidrapan/ha-solarman #221: "failed to set up" after HA OS update

deebeo-blip, 2024-10-18:
### Description

Hi, I updated from the older Integration from Stefan(...) to your Integration because with the old one, for whatever reason,  I could not upgrade Home Assistant OS above Version 12.1.. I migrated to your integration which worked fine. When I now upgraded my HA OS it stopped working. The Integration says "failed to set up".

log entry says:

`Log details (ERROR)
Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 1:54:00 PM (3 occurrences)
Last logged: 1:56:25 PM

Error setting up entry solarman for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 58, in async_setup_entry
    inverter = Inverter(inverter_host, serial, inverter_port, mb_slave_id)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 27, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = AUTO_RECONNECT, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 65, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system`

Is this something which can be fixed? I have already deleted the device and set it up fresh - with the same results. Please advise on what to do.

### Reproduction Steps

Upgrade HA OS from 12.1 to latest version.

### Log

Log details (ERROR)
Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 1:54:00 PM (3 occurrences)
Last logged: 1:56:25 PM

Error setting up entry solarman for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 58, in async_setup_entry
    inverter = Inverter(inverter_host, serial, inverter_port, mb_slave_id)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 27, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = AUTO_RECONNECT, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 65, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system

### Version

v24.10.14

deebeo-blip, 2024-10-18:
The old Integration was uninstalled before the new one was installed. I followed the migration guide from your github with several restarts and backups-

davidrapan, 2024-10-18:
Duplicate of #39

deebeo-blip, 2024-10-18:
Thanks David!

closed 2025-02-03 by davidrapan