davidrapan/ha-solarman #227: Query on write_holding_register

stevebuk1, 2024-10-18:
I'm currently updating Pv_opt to interface to your new Solarman repo using a Solis inverter. 

It was originally written for StephenHuberts Solarman but writes to the modbus registers were never enabled. 
Pv_opt runs in Appdaemon so uses code along the following lines:

```
data = {"register": address, "value": value}
self.host.call_service("solarman/write_holding_register", **data)

```

I'm trying a simple write to the start time minutes register at 43144 but its not being reflected in the entity that reports the value. 
Setting the value on the inverter screen does update the entity, so its not a read issue. 

I've been looking at closed issues and see that along with the register number and the value to be written, there also needs to be a "device" set as well? If that isnt set, will that cause the issue? If so, where can I look it up from to use in an AppDaemon app? 

Note, I'm doing the PV_opt update on behalf of another user and I don't run Solarman myself, so am not able to easily try out different device settings, hence this query.

davidrapan, 2024-10-18:
You need to know device id, you can discover it for example by trying run the action from developer tools (you select the device and then click on the GO TO YAML MODE button):
```
action: solarman.write_holding_register
data:
  wait_for_attempts: 30
  device: 68cc2dc7f6d70d4742dca1d660cba099
  register: 43144
  value: 1
```

davidrapan, 2024-10-19:
But after thinking about it some more I really think that the best approach for you will be to rather use built-in HA action with entity as target:
```
action: time.set_value
target:
  entity_id: time.inverter_timed_charge_start
data:
  time: "06:30:00"
```

stevebuk1, 2024-10-19:
Thanks for your support and quick responses.

Agreed for time registers, but hasn't gone down that route as still need to
write to the Solis storage control register, which in solis_hybrid.yaml is
a sensor and uses a modbus address which is read only.

Would it be simple to setup an entity to write to 43110 that uses the same
types? I think I'd need to use select rather than number?

Pyinthesky99, 2024-10-19:
Hi David, thanks, setting wait_for_attempts to 30 worked a treat.

davidrapan, 2024-10-19:
[image]

Are those bits mutually exclusive or do you need to control each separately?

stevebuk1, 2024-10-19:
Individually, but there's only about 8 combinations. There are four hex
codes already in Solis_hybrid.yaml on the read-only sensor, so replicating
those four is fine.

davidrapan, 2024-10-19:
Okay, I'll let you know when it's done. 😉

stevebuk1, 2024-10-19:
Thanks so much!!

davidrapan, 2024-10-19:
First draft:
```
      - name: Storage Control Mode
        platform: select
        rule: 1
        registers: [43110]
        lookup:
          - key: 0x0000
            value: "Disabled"
          - key: 0x0001
            value: "Self Use"
          - key: 0x0002
            value: "Optimized Revenue"
          - key: 0x0003
            value: "Time of Use"
          - key: 0x0004
            value: "Off-Grid"
          - key: 0x0008
            value: "Battery Wake-Up"
```

davidrapan, 2024-10-19:
[feat: Add Storage Control Mode to solis_hybrid](https://github.com/davidrapan/ha-solarman/commit/7ebe051ab693ca2683835a2d7bb8bd14b6b8bae2)

stevebuk1, 2024-10-19:
Thanks, that looks great.

I assume I can edit the keys to match the values for the read-only register
already defined, and add a few more for other combinations of bits that I
need?

name: "Storage Control Mode"
        rule: 1
        registers: [33132]
        icon: "mdi:battery-clock"
        lookup:
          - key: 0x21
            value: "Self Use"
          - key: 0x22
            value: "Optimized Revenue"
          - key: 0x23
            value: "Time of Use"
          - key: 0x24
            value: "Off-Grid Storage"
          - key: 0x28
            value: "Battery Wake-Up"
          - key: 0x60
            value: "Feed-In Priority"

davidrapan, 2024-10-19:
This just depends on whether the two registers work the same, and the docs I have only describe the first four bits.

**The best approach would be to first try to read the register using integration actions (And maybe try to switch the mode in the inverter and repeat,..) and see if it matches the read-only register or rather just what's in the docs.**

stevebuk1, 2024-10-19:
It's ok, I have the full bit definition.

davidrapan, 2024-10-19:
Great, report back finished entity description so I can add it to the profile. 😉

stevebuk1, 2024-10-19:
No problems, will do!

Pyinthesky99, 2024-10-21:
@stevebuk1 has done a lot of work on this and come up with this modified solis_hybrid.yaml which generates an entity select.solis_storage_control_mode  
[solis_hybrid.yaml.txt](https://github.com/user-attachments/files/17466600/solis_hybrid.yaml.txt)
I have added the entity to a dashboard and if I change the setting Run/Stop on the timings page of the inverter via the front buttons, this is reflected by the entities toggling between Time of Use and Self Use. Unfortunately, if change, for example, from Time of use in the entity select.solis_storage_control_mode  on my Dashboard to Self Use it doesn't change Run/Stop on the Inverter, the selected mode in the select entity just reverts, after a second or two, back to Time of Use.

Steve thinks the answer may be in select.py somewhere but this is all very different from Stephan's original Solarman.

Any ideas where to look or how to debug what is happening David?
TIA

davidrapan, 2024-10-22:
Hi, just turn on debug logging and try to switch the value and log should show what's wrong.

Pyinthesky99, 2024-10-22:
Hi David
Thanks for the suggestion, this is the log..

[home-assistant_solarman_2024-10-22T09-19-05.317Z.log](https://github.com/user-attachments/files/17473809/home-assistant_solarman_2024-10-22T09-19-05.317Z.log)

The relevant lines are probably.......

2024-10-22 10:18:40.817 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] call code 16: 43110 | 0xA866, arg: [35], wait_for_attempts: 30
2024-10-22 10:18:40.818 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] SENT: a5 1a 00 10 45 6d 00 ba bc 55 f2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 10 a8 66 00 01 02 00 23 c7 85 2c 15
2024-10-22 10:18:40.945 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] RECD: a5 16 00 10 15 6d 90 ba bc 55 f2 02 01 83 f2 a5 35 3e 09 00 00 6c 7b 71 31 01 10 a8 66 00 01 c1 b6 ae 15
2024-10-22 10:18:40.945 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] call code 16: 43110 | 0xA866, response: 1

Regards
Alan

davidrapan, 2024-10-22:
`response: 1` means that writing was successful.

Let's leave out what says HA entity.. what about the settings of inverter device itself? Is the change somehow reflected there?

Pyinthesky99, 2024-10-22:
If I select 'Time of Use' in HA the inverter doesn't move off Stop (Self Use) as reflected by the first Entity below. A Fraction of a second after the snip, the select entity reverts to Self Use.

First Snip
[image]

Second snip
[image]

Just out of curiosity, does Pysolarman have any say in any of this?

davidrapan, 2024-10-22:
I was talking about settings directly on the device and how switching the value affects that?

Pyinthesky99, 2024-10-22:
Sorry, If I change Run/Stop in the Inverter, by the buttons, both the entities select.solis_storage_control_mode and sensor.solis_storage_control_mode change accordingly to Time of Use and Self Use on the Dashboard. It just doesn't work the other way, Dashboard to Inverter.

davidrapan, 2024-10-23:
And the other way around? Because debug log from he writing says that the inverter confirmed the change so it had to do something...

Pyinthesky99, 2024-10-23:
Interesting, don't know what the inverter confirmed but according to the Inverter Panel nothing changed.

Just tried writing register 43110 from 'Actions' with the same result, response 1 in the log but no change on the inverter panel.
 
```
2024-10-23 10:06:34.416 DEBUG (MainThread) [custom_components.solarman.services] write_multiple_holding_registers: <ServiceCall solarman.write_multiple_holding_registers (c:01JAWA1ESFPJKMTTS03MN58FAG): device=b8ea543bfd251b373255a89d8175fa3e, register=43110, wait_for_attempts=30, values=[33]>
2024-10-23 10:06:34.416 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] call code 16: 43110 | 0xA866, arg: [33], wait_for_attempts: 30
2024-10-23 10:06:34.417 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] SENT: a5 1a 00 10 45 9b 00 ba bc 55 f2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 10 a8 66 00 01 02 00 21 46 44 96 15
2024-10-23 10:06:34.489 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] RECD: a5 16 00 10 15 9b dd ba bc 55 f2 02 01 ca 3f a7 35 b3 09 00 00 cf 7c 71 31 01 10 a8 66 00 01 c1 b6 98 15
2024-10-23 10:06:34.489 DEBUG (MainThread) [custom_components.solarman.api] [4065705146] call code 16: 43110 | 0xA866, response: 1

```
[home-assistant_solarman_2024-10-23T09-14-39.792Z.log](https://github.com/user-attachments/files/17489334/home-assistant_solarman_2024-10-23T09-14-39.792Z.log)

davidrapan, 2024-10-23:
Did you guys also tried to set the value using `write_holding_register` action?

Other than that it could really only mean that you are trying to set incorrect value (even so the inverter accepts it???) or some problem on the inverter side..

Anyway what that 6 bit means?

Pyinthesky99, 2024-10-23:
Hi David, 
Just now tried write_holding_register (Function Code 6) with the same result, a tick but nothing changed on Inverter.

So what could be the difference? We can write timings but not the mode, puzzling.

davidrapan, 2024-10-23:
And what that 6 bit means?

stevebuk1, 2024-10-23:
Hi,

Bit definitions of 43110 are:
Bit 0 = Self Use mode
Bit 1 = Timed Charge/Discharge
Bit 2 = Off Grid Mode
Bit 3 = Battery Wake
Bit 4 = Backup Mode
Bit 5 = Allow Grid Charging
Bit 6 = Feed in Priority Mode

The integration I use (https://github.com/wills106/homeassistant-solax-modbus) to control the Solis inverter provides a "select" entity to control the inverter, valid codes are here:

https://homeassistant-solax-modbus.readthedocs.io/en/latest/solis-operation-modes/

What we are trying to do is to establish the same via Solarman. All of the inverter side of things is fully tested 

Note: Pv_Opt uses 35d and 49d only.

stevebuk1, 2024-10-23:
Thought I'd add a brief screengrab from Pv_opt (https://github.com/fboundy/pv_opt) which currently interfaces with the Solax integration and also a 2nd Solis Integration https://github.com/fboundy/ha_solis_modbus where you will find a similar bit description in the README.md. 
```
04:30:48     INFO: Current inverter status:
04:30:48     INFO: ------------------------
04:30:48     INFO:   mode              : Timed Charge/Discharge
04:30:48     INFO:   code              : 35
04:30:48     INFO:   switches          :
04:30:48     INFO:     SelfUse         : True
04:30:48     INFO:     Timed           : True
04:30:48     INFO:     OffGrid         : False
04:30:48     INFO:     BatteryWake     : False
04:30:48     INFO:     Backup          : False
04:30:48     INFO:     GridCharge      : True
04:30:48     INFO:     FeedInPriority  : False
04:30:48     INFO:   charge            :
04:30:48     INFO:     start           : 23-Oct 00:00 BST
04:30:48     INFO:     end             : 23-Oct 06:30 BST
04:30:48     INFO:     current         : 11.8
04:30:48     INFO:     active          : True
04:30:48     INFO:   discharge         :
04:30:48     INFO:     start           : 23-Oct 00:00 BST
04:30:48     INFO:     end             : 23-Oct 00:00 BST
04:30:48     INFO:     current         : 100.0
04:30:48     INFO:     active          : False
04:30:48     INFO:   hold_soc          :
04:30:48     INFO:     active          : False
04:30:48     INFO:     soc             : 33.0
04:30:48     INFO: 
04:30:48     INFO: Currently in charge/discharge/hold slot
04:30:48     INFO: Current charge/discharge window ends in 119.2 minutes.
```

davidrapan, 2024-10-23:
Great, thanks for the explanation. 😉

Did you guys tried all the possible combination to see if all of them doesn't work (or maybe some does)?

And small note: If you get successful response to the writing that just simply means that register you are trying to write into is allowed and the value is allowed and doesn't say much about the operation itself.

davidrapan, 2024-10-30:
Hi @stevebuk1, I have small request for, could you help me determine why you are not able to read following request sets `code: 0x04, start: 33022, quantity: 122` or `code: 0x04, start: 33000, quantity: 121` as a whole? Respectively which registers in that range are causing issues cause I would like to get rid of that `requests_fine_control` in the file.

Thanks.

Pyinthesky99, 2024-10-30:
I didn't realise I had a problem reading these registers!
I can Action read 33022 Quantity 100 but not Quantity 101  Result ....  "The data address received in the request is not an allowable address for the server" Which would, on the face of it, indicate the problem is with register 33122.
Unfortunately, I can read 33023 for 100 but not for 101, which indicates it can't read more than 100 which seems to be so.
Hope this helps, can't imagine it does.

davidrapan, 2024-10-30:
Ah, so it's not about the individual registers, but rather about the length! The solution will be easier than I originally thought, nice! 

Thank you!

davidrapan, 2024-10-30:
Latest commit [feat: Add option to tune max modbus request size](https://github.com/davidrapan/ha-solarman/commit/5418c7b2de8c20d15e60d9f805c95509a80e67da) contains changes which removes the need for `requests_fine_control` (but you have to specify `max_size: 100` in defaults section).

closed 2024-10-31 by davidrapan