davidrapan/ha-solarman #231: For some number settings please add choose slider or box

oguzatagan, 2024-10-20:
### Is your feature request related to a problem?

Slider is not usefull for bad resolution of mouse :) İ think mode:box more better. sorry my bad english

### Describe the solution you'd like

Slider is not usefull for bad resolution of mouse :) İ think mode:box more better. sorry my bad english

### Describe alternatives you've considered

mode: box

CrazyUs3r, 2024-10-21:
For which yaml

oguzatagan, 2024-10-21:
i manually change the deye hybrid yaml but when coming the update coming back the slider

CrazyUs3r, 2024-10-21:
wrote here where you added mode:box then we can take that over for the next update

oguzatagan, 2024-10-21:
i think all settings must be box. but If possible, it would be better to have a choice for friends who do not want to join.

Thanks for your support.

davidrapan, 2024-10-29:
[refactor: Changed number input mode to box](https://github.com/davidrapan/ha-solarman/commit/d1f8440495be0d8459418483b2413bfa3113171c)

closed 2024-10-29 by davidrapan