davidrapan/ha-solarman #233: Deye_string intentionally has no Load values?

kingp0dd, 2024-10-22:
### Description

Inverter: 
Deye SUN-6K-SG05P1-EU-AM2

From the WIFI Stick status page, it has Grid and Load data (in Watts), but there's no value ni HASS

### Reproduction Steps

Used deye_string as profile

### Log

HASS just shows this for Load Power: 'This entity is no longer being provided by the solarman integration. If the entity is no longer in use, delete it in settings.'

### Version

v1.0.18

davidrapan, 2024-10-22:
I guess cause no one put it there before. I'll try to add it. 😉

kingp0dd, 2024-10-23:
Is it as simple as copying it from the other deye profiles? I will try it
and see if it works :)

kingp0dd, 2024-10-24:
I tried to load the other deye profiles which have Load and Grid items, but all of them show 0 value from my Deye SUN-6K-SG05P1-EU-AM2.

When I visit the SolarMan webpage, it does have Load and Grid items. Is there a method of decoding these from SolarMan?

davidrapan, 2024-10-24:
It depends on which one did you tried and what values it shows.

kingp0dd, 2024-10-24:
> It depends on which one did you tried and what values it shows.

i tried all the available deye profiles but all just showed unavailable values :(

davidrapan, 2024-10-25:
So that could mean that those values are exposed using different registers or not exposed at all and are maybe calculated.

The only profiles which could be compatible with your device are: `deye_hybrid.yaml`, `deye_2mppt.yaml` and `deye_4mppt.yaml`.

Can you maybe try to look for another values which do show some values when you use these other profiles (some other than load and grid power)?

kingp0dd, 2024-10-31:
> So that could mean that those values are exposed using different registers or not exposed at all and are maybe calculated.
> 
> The only profiles which could be compatible with your device are: `deye_hybrid.yaml`, `deye_2mppt.yaml` and `deye_4mppt.yaml`.
> 
> Can you maybe try to look for another sensors which do show some values when you use these other profiles (some other than load and grid power)?

I tried those profiles one by one and here's what I got.

Hybrid:

[image]

The only significant value I found here is Today's Load Consumption, but it's in kWh. Other useful items with values here are just PV (A, V, W).

For 2mppt / 4mppt:

[image]

Grid has values here, but when I try to compute W (A x V), it doesn't reflect the value of Grid Power in Solarman website.

davidrapan, 2024-10-31:
> It seems that the Today's Load Consumption from Hybrid profile is the same (or close to) PV Power (W) value.

This value also makes no sense. It's too high.

 And `238.4 * 4,3 = 1,025.12` which +- corresponds to the `PV Power: 1039 W` with `13.88 W` of losses.

Can you tell me what could `Power: 1278,9 W` mean?

kingp0dd, 2024-11-01:
I can't look at 1278.9W since the records yesterday got overwritten when I reconfigured. But I checked the graphs today (above is from HA, below is from Solarman webpage):

[image]

I'd like to thin that `Power` is `Load/Consumption` but in Solarman, `Load` is always higher than `PV Power`, which is not the case in HASS. 

Also, `Power Loss` in HASS seems to correlate to `Grid Power` in SolarMan.

davidrapan, 2024-11-02:
Oh wow, that looks so weird imo.

It looks like there is longer `update_interval` for Inverter Power for some reason... And that is probably causing this inconsistency so because you have most of the time "older" data for `Power`.

How does `Load Power` look like in comparison to all the values?

davidrapan, 2024-11-02:
Do you have an idea how is your inverter currently interconnected with your house and grid? All the inputs/outputs and CT clamps?

Would be helpful to better determine what's going on... 😉

kingp0dd, 2024-11-02:
> How does `Load Power` look like in comparison to all the values?

I assume you're referring to `Load/Consumption` from SolarMan webpage? Because I don't have that in HASS. Currently, there's no close value of SolarMan's Load value to any of the HASS, especially when I use the sample data at around 5:56am in the screenshot where their values are very contrasting:

PV Power: 8W
Grid: 1887W
Load/Consumption: 1.88kW

> Could you enable debug logging for couple of minutes and attach the debug log here?

It's night time here so the inverter turned off already 😆 I will check again tomorrow

> Do you have an idea how is your inverter currently interconnected with your house and grid? All the inputs/outputs and CT clamps?

I will try to take a picture tomorrow if those can be seen from the exposed cables. But I think the values are there and transmitted through the logger since SolarMan webpage has the correct values

davidrapan, 2024-11-02:
> It's night time here so the inverter turned off already 😆 I will check again tomorrow

😉

> I will try to take a picture tomorrow if those can be seen from the exposed cables. But I think the values are there and transmitted through the logger since SolarMan webpage has the correct values

Sure I just want to determine if the values are really there (measured by CT clams) OR are maybe calculated additionally.

kingp0dd, 2024-11-03:
> > It's night time here so the inverter turned off already 😆 I will check again tomorrow
> 
> 😉
> 
> > I will try to take a picture tomorrow if those can be seen from the exposed cables. But I think the values are there and transmitted through the logger since SolarMan webpage has the correct values
> 
> Sure I just want to determine if the values are really there (measured by CT clamps) OR are maybe calculated additionally.

Attaching debug logs:

```
2024-11-03 13:03:27.199 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.213 seconds (success: True)
2024-11-03 13:03:31.984 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Scheduling 1 query request. #18340
2024-11-03 13:03:31.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 ...
2024-11-03 13:03:31.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] SENT: a5 17 00 10 45 3b 00 1d 8c b7 b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3c 00 35 45 d1 45 15
2024-11-03 13:03:32.186 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] RECD: a5 7f 00 10 15 3b 45 1d 8c b7 b1 02 01 a4 05 1e 00 c3 05 00 00 7f fe 08 67 01 03 6a 00 2c 00 00 00 00 0e db 00 00 00 00 00 00 00 00 00 00 00 00 09 00 00 00 00 00 09 00 00 00 00 00 00 16 00 00 00 00 17 66 16 30 00 00 16 53 00 00 13 e5 00 00 17 4a 00 00 02 4b 00 00 05 98 00 00 00 00 03 e8 00 00 00 00 00 00 00 00 00 00 02 57 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 79 00 11 05 e3 00 10 8c 3c 00 00 62 15
2024-11-03 13:03:32.186 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 succeeded.
2024-11-03 13:03:32.186 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Returning 25 new values to the Coordinator. [Previous State: Connected (1)]
2024-11-03 13:03:32.186 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.202 seconds (success: True)
2024-11-03 13:03:36.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Scheduling 1 query request. #18345
2024-11-03 13:03:36.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 ...
2024-11-03 13:03:36.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] SENT: a5 17 00 10 45 3c 00 1d 8c b7 b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3c 00 35 45 d1 46 15
2024-11-03 13:03:37.195 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] RECD: a5 7f 00 10 15 3c 46 1d 8c b7 b1 02 01 a9 05 1e 00 c8 05 00 00 7f fe 08 67 01 03 6a 00 2c 00 00 00 00 0e db 00 00 00 00 00 00 00 00 00 00 00 00 09 04 00 00 00 00 09 04 00 00 00 00 00 16 00 00 00 00 17 68 16 64 00 00 16 77 00 00 13 e7 00 00 16 84 00 00 02 7d 00 00 05 98 00 00 00 00 03 e8 00 00 00 00 00 00 00 00 00 00 02 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 9d 00 11 06 4b 00 0f 58 04 00 00 37 15
2024-11-03 13:03:37.195 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 succeeded.
2024-11-03 13:03:37.196 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Returning 25 new values to the Coordinator. [Previous State: Connected (1)]
2024-11-03 13:03:37.196 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.211 seconds (success: True)
2024-11-03 13:03:41.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Scheduling 1 query request. #18350
2024-11-03 13:03:41.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 ...
2024-11-03 13:03:41.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] SENT: a5 17 00 10 45 3d 00 1d 8c b7 b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3c 00 35 45 d1 47 15
2024-11-03 13:03:42.198 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] RECD: a5 7f 00 10 15 3d 47 1d 8c b7 b1 02 01 ae 05 1e 00 cd 05 00 00 7f fe 08 67 01 03 6a 00 2c 00 00 00 00 0e db 00 00 00 00 00 00 00 00 00 00 00 00 09 04 00 00 00 00 09 04 00 00 00 00 00 16 00 00 00 00 17 68 16 64 00 00 16 77 00 00 13 e7 00 00 16 84 00 00 02 7d 00 00 05 98 00 00 00 00 03 e8 00 00 00 00 00 00 00 00 00 00 02 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 9d 00 11 06 4b 00 0f 58 04 00 00 43 15
2024-11-03 13:03:42.198 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 succeeded.
2024-11-03 13:03:42.199 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Returning 25 new values to the Coordinator. [Previous State: Connected (1)]
2024-11-03 13:03:42.199 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.214 seconds (success: True)
2024-11-03 13:03:46.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Scheduling 1 query request. #18355
2024-11-03 13:03:46.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 ...
2024-11-03 13:03:46.985 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] SENT: a5 17 00 10 45 3e 00 1d 8c b7 b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3c 00 35 45 d1 48 15
[... cut ...]
2024-11-03 13:09:27.198 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Querying 0060 - 0112 | 0x003C - 0x0070 # 053 succeeded.
2024-11-03 13:09:27.198 DEBUG (MainThread) [custom_components.solarman.api] [2981596189] Returning 25 new values to the Coordinator. [Previous State: Connected (1)]
2024-11-03 13:09:27.198 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.214 seconds (success: True)
```

I didn't get a chance to take a picture of the cables today, will try to get them tomorrow

kingp0dd, 2024-11-07:
I don't know if the clamps can be seen here:

[image]

davidrapan, 2024-11-07:
How many cables is going in/out on that right side? (It's lil bit blurry)

kingp0dd, 2024-11-08:
> How many cables is going in/out on that right side? (It's lil bit blurry)

After the Logger, there are 2 cables going in/out:

[image]

kingp0dd, 2024-11-16:
I was able to take more pictures earlier

[image]
[image]
[image]

kingp0dd, 2024-11-28:
> Sensors has been added in [feat: Load & Grid Power for Deye String](https://github.com/davidrapan/ha-solarman/commit/12c5d45704764a802a9d6c8a012e6ad28bc760dd) and [feat: Energy sensors for Deye String](https://github.com/davidrapan/ha-solarman/commit/20e04f14a7fc366831e65e61ed86812e185a0059)

Absolute banger!
May I ask how you were able to identify the registers?

I've been looking at the manuals:
https://github.com/mihaigsm2003/Deye2mqtt-Read-data-from-Deye-Sunsynk/blob/main/protocols/Modbus%E5%82%A8%E8%83%BD-%E7%BB%84%E4%B8%B2-%E5%BE%AE%E9%80%86%E5%AE%81%E6%B3%A2%E5%BE%B7%E4%B8%9AV118.pdf

```
      - name: "Load Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        rule: 1
        registers: [0x00c6, 0x00c7]
```
Those registers refer to:

[image]

davidrapan, 2024-11-28:
I added two more sensors in [feat: Energy sensors for Deye String](https://github.com/davidrapan/ha-solarman/commit/5dc38f3e44286b9dc071e03ce71ec9d2fb6b4b4e)

> Those registers refer to:

Translation of 198 and 199 is not correct, the Chinese text says: The active power of the load is low.

kingp0dd, 2024-11-28:
Confirming that Load is working now. The other line is from Solarman API:

[image]

`registers: [0x00c6, 0x00c7]`
This means registering two values but there's no math operation, right?

davidrapan, 2024-11-28:
That just means that the value is stored as 32b integer. (One register is 16b)

davidrapan, 2024-11-28:
Can you check for me if newly added energy sensors are working correctly?
* Today's Energy Import
* Total Energy Import
* Today's Energy Export
* Total Energy Export
* Today's Load Consumption
* Total Load Consumption
* Today's Losses
* Total Losses

kingp0dd, 2024-12-02:
It is working. I will check which are the corresponding items in solarman
web and compare the values too

closed 2024-12-17 by davidrapan