davidrapan/ha-solarman #235: Daily production

Ledo84, 2024-10-23:
### Description

deye_today_production is the problem

Hello I'm Having problems with daily production calculation from my panels. Everything was ok imo till 18th then the data just went wrong.
Im on v. 24.10.16 
deye_sg04lp3.yaml
12k Deye hybrid inverter
Logs attached.
Any ideas?

### Reproduction Steps

[image]

### Log

  File "/config/custom_components/solarman/entity.py", line 42, in _handle_coordinator_update
    self.update()
  File "/config/custom_components/solarman/entity.py", line 62, in update
    self.set_state(data["state"])
  File "/config/custom_components/solarman/sensor.py", line 101, in set_state
    if self._sensor_ensure_increasing and self._attr_native_value and self._attr_native_value > state > 0:

 TypeError: '>' not supported between instances of 'int' and 'NoneType'
2024-10-23 11:00:04.219 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Scheduling 1 query request. #23375
2024-10-23 11:00:04.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 ...
2024-10-23 11:00:04.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] SENT: a5 17 00 10 45 94 00 79 9d 7b b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 4a 00 7b 25 87 bb 15
2024-10-23 11:00:04.546 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] RECD: a5 09 01 10 15 94 fe 79 9d 7b b1 02 01 df a5 1d 00 9a 03 00 00 35 15 fb 66 01 03 f6 04 ab 15 29 00 26 00 00 ef 20 e3 40 02 5e 00 00 00 00 00 00 00 00 00 00 09 31 09 3b 09 42 00 00 00 00 00 00 06 d2 02 6b 02 a0 0b dd 00 00 13 83 02 d0 00 f4 01 5a 01 91 02 26 01 f8 02 47 00 40 fd ec 00 73 00 00 00 00 02 47 00 40 fd ec 00 73 00 00 09 2c 09 3d 09 40 01 d6 00 32 ff 9c 03 59 00 34 00 0d 03 9a 03 9a 13 83 00 00 0a 2b 02 9f 02 ad 0f 77 09 36 09 48 09 3a 00 00 00 00 00 00 05 a0 00 74 00 00 04 0d 04 0d 13 83 00 00 00 00 00 00 00 00 00 00 00 05 00 06 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d d7 05 d6 00 00 00 00 14 6b 00 42 09 d8 00 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 35 0b c1 15
2024-10-23 11:00:04.547 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.
2024-10-23 11:00:04.549 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Returning 72 new values to the Coordinator. [Previous State: Connected (1)]
2024-10-23 11:00:04.549 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.331 seconds (success: True)
2024-10-23 11:00:09.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Scheduling 1 query request. #23380
2024-10-23 11:00:09.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 ...
2024-10-23 11:00:09.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] SENT: a5 17 00 10 45 95 00 79 9d 7b b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 4a 00 7b 25 87 bc 15
2024-10-23 11:00:09.555 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] RECD: a5 09 01 10 15 95 ff 79 9d 7b b1 02 01 e4 a5 1d 00 9f 03 00 00 35 15 fb 66 01 03 f6 04 ab 14 f1 00 26 00 00 fc 18 ff 79 02 5e 00 00 00 00 00 00 00 00 00 00 09 35 09 3b 09 39 00 00 00 00 00 00 02 43 ff 80 02 5d 04 20 00 00 13 83 01 04 00 71 01 45 01 5f 02 13 01 e6 00 34 ff c1 ff e0 ff d5 00 00 00 00 00 34 ff c1 ff e0 ff d5 00 00 09 30 09 3c 09 3b 03 3e 01 0e 00 00 07 ef 03 15 00 51 0b 55 0b 55 13 83 00 00 0a 32 02 95 02 ae 0f 75 09 3b 09 41 09 37 00 00 00 00 00 00 08 23 02 d6 00 31 0b 2a 0b 2a 13 83 00 00 00 00 00 00 00 00 00 00 00 05 00 06 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 07 8e 03 e0 00 00 00 00 14 39 00 25 09 d8 00 26 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 ff ff ff ff ff ff 19 c2 5d 15
2024-10-23 11:00:09.556 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.
2024-10-23 11:00:09.558 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Returning 72 new values to the Coordinator. [Previous State: Connected (1)]
2024-10-23 11:00:09.558 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.340 seconds (success: True)
2024-10-23 11:00:14.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Scheduling 1 query request. #23385
2024-10-23 11:00:14.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 ...
2024-10-23 11:00:14.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] SENT: a5 17 00 10 45 96 00 79 9d 7b b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 4a 00 7b 25 87 bd 15
2024-10-23 11:00:15.838 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] RECD: a5 09 01 10 15 96 00 79 9d 7b b1 02 01 e9 a5 1d 00 a4 03 00 00 35 15 fb 66 01 03 f6 04 ab 14 e5 00 26 00 00 01 7f 01 ac 02 5e 00 00 00 00 00 00 00 00 00 00 09 37 09 3b 09 3a 00 00 00 00 00 00 01 c6 ff 69 01 f2 03 21 00 00 13 83 00 c3 00 73 01 32 01 5b 02 0e 01 e2 ff e9 ff e2 ff dc ff a7 00 00 00 00 ff e9 ff e2 ff dc ff a7 00 00 09 32 09 3e 09 3b 03 70 01 36 00 32 08 72 03 2d 00 bf 0c 5e 0c 5e 13 83 00 00 0a 38 02 96 02 b1 0f 7f 09 3b 09 49 09 36 00 00 00 00 00 00 08 5b 03 0f 00 9b 0c 05 0c 05 13 83 00 00 00 00 00 00 00 00 00 00 00 06 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 08 54 03 10 00 00 00 00 14 77 00 2b 09 d8 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 ff ff ff ff ff ff ff ff 11 34 e1 15
2024-10-23 11:00:15.839 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.
2024-10-23 11:00:15.840 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Returning 72 new values to the Coordinator. [Previous State: Connected (1)]
2024-10-23 11:00:15.841 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 1.622 seconds (success: True)
2024-10-23 11:00:20.219 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Scheduling 1 query request. #23390
2024-10-23 11:00:20.219 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 ...
2024-10-23 11:00:20.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] SENT: a5 17 00 10 45 97 00 79 9d 7b b1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 4a 00 7b 25 87 be 15
2024-10-23 11:00:20.544 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] RECD: a5 09 01 10 15 97 01 79 9d 7b b1 02 01 ef a5 1d 00 aa 03 00 00 35 15 fb 66 01 03 f6 04 ab 15 08 00 26 00 00 f5 fd eb a6 02 5e 00 00 00 00 00 00 00 00 00 00 09 37 09 3c 09 3b 00 00 00 00 00 00 05 9e 01 f6 03 86 0b 1a 00 00 13 83 02 30 00 c5 01 a2 01 83 02 23 01 e8 01 7f 00 72 ff 79 01 6a 00 00 00 00 01 7f 00 72 ff 79 01 6a 00 00 09 2d 09 3c 09 3c 02 4e 00 50 ff 9c 04 96 00 ac ff 2c 04 6e 04 6e 13 83 00 00 0a 34 02 a2 02 b2 0f 88 09 37 09 46 09 36 00 00 00 00 00 00 06 15 01 1e 00 00 05 d8 05 d8 13 83 00 00 00 00 00 00 00 00 00 00 00 05 00 06 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 21 04 8f 00 00 00 00 14 9d 00 3a 09 d9 00 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 3c 2d fa 15
2024-10-23 11:00:20.545 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 succeeded.
2024-10-23 11:00:20.547 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Returning 72 new values to the Coordinator. [Previous State: Connected (1)]
2024-10-23 11:00:20.547 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.329 seconds (success: True)
2024-10-23 11:00:25.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Scheduling 1 query request. #23395
2024-10-23 11:00:25.220 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Querying 0586 - 0708 | 0x024A - 0x02C4 # 123 ...
[... cut ...]
2024-10-23 11:00:31.394 DEBUG (MainThread) [custom_components.solarman.parser] do_validate Total Losses: 0 < 1
2024-10-23 11:00:31.397 DEBUG (MainThread) [custom_components.solarman.api] [2977668473] Returning 205 new values to the Coordinator. [Previous State: Connected (1)]
2024-10-23 11:00:31.397 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 1.178 seconds (success: True)
2024-10-23 11:00:31.399 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):

### Version

24.10.16

PV-Joe, 2024-10-23:
I know it doesn't help you much but the today's production for my Deye Hybrid 8K with sg04lp3.yaml and the identical version of ha-Solarman is working flawlessly.

Ledo84, 2024-10-23:
Ive restored a backup when everything worked fine see how it goes... 
Time is set and synced... HA System is up and running all the time... I have additional Shelly sensor on my 2nd inverter no problem there it shows correct data. I'm puzzled.

Ledo84, 2024-10-23:
Well... I can't see  the production from one of my inverters that's kind of a bummer. 
I remembered Ive added a string of panels to my inverter on PV2 but that couldn't have coused it?

davidrapan, 2024-10-23:
I'm not sure I understand what you are trying to say?

Ledo84, 2024-10-23:
I physically added panels to my inverter. Could that be a problem for the meter?  
Do You have any ideas? Reinstall solarman?

davidrapan, 2024-10-23:
> I physically added panels to my inverter. Could that be a problem for the meter?

No, I don't see how that could be a problem.

> Do You have any ideas? Reinstall solarman?

This doesn't look like it could be a problem w/ integration itself.

closed 2024-12-12 by github-actions[bot]