davidrapan/ha-solarman #244: SOLARMAN INSTALLATION ERROR -  DEYE HYBRID INVERTER  SG03LP1-EU5000W BATTERY DEYE  5 KW WITH HOME ASSISTANT GREEN

sentink, 2024-10-30:
### Description

Good afternoon, I have installed the solarman integration and after placing all the configuration it gives an error indicating "could not be configured" "check the logs"

- DEYE HYBRID INVERTER  SG03LP1 -EU5000W - DATA LOGGER
- BATTERY DEYE  5 KW
- METER CHINT
- HOME ASSISTANT GREEN WITH HA OS

[image]

[image]

### Reproduction Steps

SOFTWARE INSTALLATION
HACS 2.0.1
SOLARMAN 24.10.31

### Log

Registrador: homeassistant.config_entries
Fuente: config_entries.py:594
Ocurrió por primera vez: 06:56:19 (8 ocurrencias)
Última vez registrado: 12:23:47

Error setting up entry Inverter for solarman
Error setting up entry Inverter1 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 58, in async_setup_entry
    inverter = Inverter(inverter_host, serial, inverter_port, mb_slave_id)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 27, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = AUTO_RECONNECT, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 65, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system

### Version

24.10.31

davidrapan, 2024-10-30:
Duplicate of #39

davidrapan, 2024-10-30:
> OSError: [Errno 30] Read-only file system

ssh into your ha and run: `ha core rebuild`

sentink, 2024-10-30:
Thank you very much David, great job, I have already solved the problem, I will continue to advance in this world.

closed 2025-02-03 by davidrapan