davidrapan/ha-solarman #250: HOMEASSISTANT GREEN UPDATE VERSION 13.2 -

sentink, 2024-11-04:
### Description

Good morning, this morning HomeAssistant Green was updated to version 13.2 and Solarman has stopped working, all entities appear deactivated.

### Reproduction Steps

[image]

[image]

[image]

[image]

### Log

No debug log is generated

### Version

v24.10.31

sentink, 2024-11-04:
Solarman has been updated to version 24.11.02, but it still doesn't work

[image]

Dawid46, 2024-11-04:
will it be fixed ?

sentink, 2024-11-04:
Good afternoon David, no, it still doesn't work

[image]

davidrapan, 2024-11-04:
You need to enable debug logging, reload the integration and then attach the file here.

sentink, 2024-11-04:
Good night

Registrador: homeassistant.config_entries
Fuente: config_entries.py:594
Ocurrió por primera vez: 17:51:37 (3 ocurrencias)
Última vez registrado: 23:12:02

Error setting up entry Inverter1 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 58, in async_setup_entry
    inverter = Inverter(inverter_host, serial, inverter_port, mb_slave_id)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 26, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = AUTO_RECONNECT, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 65, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system

davidrapan, 2024-11-04:
Duplicate of #39

davidrapan, 2024-11-04:
> OSError: [Errno 30] Read-only file system

ssh into your ha and run: ha core rebuild

sentink, 2024-11-04:
Thanks for the second time David, I'm a newbie to HA, I'll record this in case it happens again in another update.

closed 2025-02-03 by davidrapan