davidrapan/ha-solarman #252: Bluesun inverter ModBUS register list

neofral, 2024-11-05:
### Describe the solution you'd like

tying to use the integration for Bluesun hybrid inverter.  My inverter model is BSE15KH3. I would very much appreciate if somebody shared the register file. I got one from Bluesun but is very old and most of the registers are not correct. I will contribute by sharing the yaml file (after tested) so it's accessible for other users. 

### Describe alternatives you've considered

Waiting from Bluesun to provide correct file with ModBUS registers

neofral, 2024-11-06:
David, where can i send you an email with the files and explanation?

davidrapan, 2024-11-06:
You can upload it right here. 😉 Just drag and drop the file in the textbox. 😉

neofral, 2024-11-06:
old file not showing all correct data, especially the battery. thought was the wrong and asked new file...
google sheets file:
https://docs.google.com/spreadsheets/d/1hLhIKdW9gyouqYl-kxwpdWvRxji5lwPeI-_4htfawfA/edit?usp=sharing

new file, not working giving error loading device
[RS485_MODBUS(ESINV-33000ID) Hybrid Inverter.pdf](https://github.com/user-attachments/files/17645636/RS485_MODBUS.ESINV-33000ID.Hybrid.Inverter.pdf)

davidrapan, 2024-11-06:
How does your yaml look like currently?

neofral, 2024-11-06:
connection not stable. disconnecting after a while.
how can i turn on the logger and where to find it?

davidrapan, 2024-11-06:
[image]

neofral, 2024-11-06:
[bluesun.txt](https://github.com/user-attachments/files/17648580/bluesun.txt)

this is a stable working file. 

need help:
with the double register values 
with parsing the state registers

davidrapan, 2024-11-06:
You cut it off (the log ends) just before it could reveal any information.

neofral, 2024-11-06:
i'm using mini-pc with HA operating system. using the wifi of the pc to connect to the wifi dongle AP.

they're right next to each other. 

the wifi dongle is connecting to solarman platform via router that is about 4 meters away with signal quality 98%

neofral, 2024-11-06:
maybe the query time of 30sec is too often, i'll try 60 sec and see what happens

davidrapan, 2024-11-06:
> maybe the query time of 30sec is too often, I'll try 60 sec and see what happens

No, that's not it.

neofral, 2024-11-06:
i was also trying modbus board with esphome connected to another comm port of the inverter trying it out.
i disconnected that device. since we got it working this way, i don't think i need that one.
let's focus and keep the connection stable.

however, i have to mention that the inverter was also disonnecting from solarman platform. i do a restart of the dongle via the webpage and it's back on.

but i'm crazy how rare is the upload interval between inverter and the solarman webpage

davidrapan, 2024-11-06:
> however, i have to mention that the inverter was also disonnecting from solarman platform. i do a restart of the dongle via the webpage and it's back on.

> but i'm crazy how rare is the upload interval between inverter and the solarman webpage

That really could indicate some problems with/on your network, imo..

neofral, 2024-11-06:
why am not able to connect via the dongle's lan ip address and not use the dongl AP?
anything settings that i've missed?

davidrapan, 2024-11-06:
I really don't think you can turn web off so it looks like the stick is just unresponsive.

neofral, 2024-11-06:
well what do you know. my mikrotik router had the upnp disabled by default. i presume the stick uses it to open ports and since it failed was behaving strangely.

just enabled upnp, reverted to 30sec update interval, and finally got connected through the lan ip address of the stick

davidrapan, 2024-11-06:
> well what do you know. my mikrotik router had the upnp disabled by default. i presume the stick uses it to open ports and since it failed was behaving strangely.

I don't think it should be like turning it on and off again repeatedly? But who knows...

> just enabled upnp, reverted to 30sec update interval, and finally got connected through the lan ip address of the stick

Yeah it should handle even once a second updates. (5 seconds is the lowest possible w/ integrations though)

neofral, 2024-11-06:
disconnected again. not a full hour working... any ideas?

davidrapan, 2024-11-06:
Run it with enabled debug logging from the time of the restart of the stick until it stops working again. Maybe the transition will reveal something.

neofral, 2024-11-06:
[home-assistant (1).log](https://github.com/user-attachments/files/17651000/home-assistant.1.log)

neofral, 2024-11-06:
do you know why the sensors are not showing in groups?

davidrapan, 2024-11-06:
> do you know why the sensors are not showing in groups?

That's not their purpose. It's only for profile configuration. You can set `update_interval` and `code` for groups.

neofral, 2024-11-06:
did a new setup with the wizard on the stick. disabled/hided the AP now no clients connecting on it.
another issue i might have is that i have a lot of switches and sensors using esp8266 and esp32 devices . i have 4 access points and about 60 wifi/lan clients. i'd say that the wifi is quite congested maybe the stick doesn't like the environment.

it's been running stable for more than 3 hours now.

by the way you have to give contact, let's see how we can cooperate

davidrapan, 2024-11-06:
> it's been running stable for more than 3 hours now.

That sounds promising!

neofral, 2024-11-08:
update: very stable and working.
moving on with setting variables.
@davidrapan, need help with time addresses

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/ee2/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/ee2/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link=blue vlink=purple>

√ | AC charging start time 1 | TimeOnAcChg1 | 2509 | U16 | 2 | H:0-23     M:0-59
-- | -- | -- | -- | -- | -- | --
√ | AC charging end   time 1 | TimeOffAcChg1 | 2510 | U16 | 2 | H:0-23     M:0-59

</body>

</html>

neofral, 2024-11-08:
log:

2024-11-08 10:00:35.056 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139921206156672] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2845, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/time/__init__.py", line 37, in _async_set_value
    return await entity.async_set_value(service_call.data[ATTR_TIME])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/time.py", line 67, in async_set_value
    await self.write(self._to_native_value(value), value.strftime(TIME_FORMAT))
  File "/config/custom_components/solarman/entity.py", line 146, in write
    if await self.coordinator.inverter.call(self.code, self.register, value, ACTION_ATTEMPTS_MAX) > 0 and state:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 271, in call
    response = await self.safe_read_write(code, start, arg)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 166, in safe_read_write
    if (response := await self.read_write(code, start, arg)) and (length := ilen(response)) and (expected := arg if code < CODE.WRITE_SINGLE_COIL else 1) and length != expected:
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 157, in read_write
    return await self.write_holding_register(start, arg)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 99, in write_holding_register
    return await super().write_holding_register(register_addr, value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 376, in write_holding_register
    value = await self._get_modbus_response(mb_request_frame)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 271, in _get_modbus_response
    modbus_values = rtu.parse_response_adu(mb_response_frame, mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/umodbus/client/serial/rtu.py", line 190, in parse_response_adu
    function = create_function_from_response_pdu(resp_pdu, req_pdu)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/umodbus/functions.py", line 132, in create_function_from_response_pdu
    function_code = pdu_to_function_code_or_raise_error(resp_pdu)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/umodbus/functions.py", line 118, in pdu_to_function_code_or_raise_error
    raise error_code_to_exception_map[error_code]
umodbus.exceptions.IllegalDataValueError:  The value contained in the request data field is not an allowable value
    for the server.

neofral, 2024-11-08:
debug logger:

2024-11-08 10:50:37.097 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 03 ~ 2504 - 2510 | 0x09C8 - 0x09CE # 007 ...
2024-11-08 10:50:37.098 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] SENT: a5 17 00 10 45 64 00 2a c1 78 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 09 c8 00 07 86 6a d2 15
2024-11-08 10:50:37.124 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] RECD: a5 21 00 10 15 64 6a 2a c1 78 d1 02 01 70 e1 08 00 96 13 00 00 7c fd 24 67 01 03 0e 01 f4 02 58 00 46 00 5f 03 e8 00 14 04 32 65 54 45 15
2024-11-08 10:50:37.125 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 03 ~ 2504 - 2510 | 0x09C8 - 0x09CE # 007 succeeded.
2024-11-08 10:50:37.125 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 0530 - 0560 | 0x0212 - 0x0230 # 031 ...
2024-11-08 10:50:37.125 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] SENT: a5 17 00 10 45 65 00 2a c1 78 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 02 12 00 1f 10 7f ce 15
2024-11-08 10:50:37.167 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] RECD: a5 51 00 10 15 65 6b 2a c1 78 d1 02 01 70 e1 08 00 96 13 00 00 7c fd 24 67 01 04 3e 00 05 00 00 00 f4 ff ff ff dc 00 00 00 d5 ff ff fe 43 00 00 01 ee 00 00 01 75 00 00 02 5e 00 00 01 3e 00 00 05 13 00 00 01 60 00 00 05 42 00 00 15 d3 11 b3 01 1e 05 a8 13 63 02 ea 10 2b bb 5a 9a 15
2024-11-08 10:50:37.168 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 0530 - 0560 | 0x0212 - 0x0230 # 031 succeeded.
2024-11-08 10:50:37.168 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 1007 - 1007 | 0x03EF - 0x03EF # 001 ...
2024-11-08 10:50:37.168 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] SENT: a5 17 00 10 45 66 00 2a c1 78 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 03 ef 00 01 00 7b 7b 15
2024-11-08 10:50:37.198 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] RECD: a5 15 00 10 15 66 6c 2a c1 78 d1 02 01 70 e1 08 00 96 13 00 00 7c fd 24 67 01 04 02 00 7c b8 d1 55 15
2024-11-08 10:50:37.198 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 1007 - 1007 | 0x03EF - 0x03EF # 001 succeeded.
2024-11-08 10:50:37.199 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 2002 - 2014 | 0x07D2 - 0x07DE # 013 ...
2024-11-08 10:50:37.199 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] SENT: a5 17 00 10 45 67 00 2a c1 78 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 07 d2 00 0d 90 82 06 15
2024-11-08 10:50:37.229 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] RECD: a5 2d 00 10 15 67 6d 2a c1 78 d1 02 01 70 e1 08 00 96 13 00 00 7c fd 24 67 01 04 1a 00 26 00 64 0c 8c fa a7 0c 84 ff ff ee da 00 7f 00 0f 00 00 05 49 00 00 06 1e 40 3d 18 15
2024-11-08 10:50:37.230 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 2002 - 2014 | 0x07D2 - 0x07DE # 013 succeeded.
2024-11-08 10:50:37.230 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 2500 - 2514 | 0x09C4 - 0x09D2 # 015 ...
2024-11-08 10:50:37.230 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] SENT: a5 17 00 10 45 68 00 2a c1 78 d1 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 09 c4 00 0f f2 6f 4c 15
2024-11-08 10:50:37.261 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] RECD: a5 31 00 10 15 68 6e 2a c1 78 d1 02 01 70 e1 08 00 96 13 00 00 7c fd 24 67 01 04 1e 00 03 00 00 00 00 00 00 00 00 00 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 01 01 78 12 96 b5 15
2024-11-08 10:50:37.261 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Querying 04 ~ 2500 - 2514 | 0x09C4 - 0x09D2 # 015 succeeded.
2024-11-08 10:50:37.262 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Returning 27 new values to the Coordinator. [Previous State: Connected (1)]
2024-11-08 10:50:37.262 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Bluesun data in 0.166 seconds (success: True)
2024-11-08 10:50:37.275 DEBUG (MainThread) [custom_components.solarman.time] SolarmanTimeEntity.native_value of Bluesun 4 AC Charge Time 1 Finish: ValueError: unconverted data remains: 4
2024-11-08 10:50:42.096 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Scheduling 0 query requests. #260
2024-11-08 10:50:42.096 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Bluesun data in 0.001 seconds (success: True)
2024-11-08 10:50:42.100 DEBUG (MainThread) [custom_components.solarman.time] SolarmanTimeEntity.native_value of Bluesun 4 AC Charge Time 1 Finish: ValueError: unconverted data remains: 4
2024-11-08 10:50:47.096 DEBUG (MainThread) [custom_components.solarman.api] [3514351914] Scheduling 0 query requests. #265

neofral, 2024-11-08:
more loggin:

2024-11-08 11:23:34.466 DEBUG (MainThread) [custom_components.solarman.time] SolarmanTimeEntity.native_value of Bluesun 3 AC Charge Time 1 Start: ValueError: time data '\x00\x14' does not match format '%H:%M'

neofral, 2024-11-08:
i tried to set 01:50

when i set this time in solarman platform, reading any value in the H place higher than 0 gives incorrect value

davidrapan, 2024-11-08:
Set it to 01:50 in the Solarman App and then read the register using built-in action in Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3) and show me the result.

neofral, 2024-11-08:
> more loggin:
> 
> 2024-11-08 11:23:34.466 DEBUG (MainThread) [custom_components.solarman.time] SolarmanTimeEntity.native_value of Bluesun 3 AC Charge Time 1 Start: ValueError: time data '\x00\x14' does not match format '%H:%M'

check above post:

more loggin:

2024-11-08 11:23:34.466 DEBUG (MainThread) [custom_components.solarman.time] SolarmanTimeEntity.native_value of Bluesun 3 AC Charge Time 1 Start: ValueError: time data '\x00\x14' does not match format '%H:%M'

davidrapan, 2024-11-08:
I did, do what I asked in previous post please.

davidrapan, 2024-11-08:
No.

neofral, 2024-11-08:
[image]

davidrapan, 2024-11-08:
Set it in Solarman App to 23:59 and read again.

neofral, 2024-11-08:
response: 

"2509": 5947

davidrapan, 2024-11-08:
I'm not sure how is the value encoded in the register...

neofral, 2024-11-08:
is it possible to be a string in the format: H:xxM:xx  ?

neofral, 2024-11-08:
can we at least try to give it command to start charging and leave the times set on solarman ?
how to arrange this:
[image]

davidrapan, 2024-11-08:
It could have been just a switch but sure.

neofral, 2024-11-08:
https://github.com/davidrapan/ha-solarman/issues/252#issuecomment-2464836229

your opinion?

neofral, 2024-11-08:
i think is more important to find out how to control the inverter.
last night inverter didn't charge even though i didn't touch these values. these values were set on solarman.
however, i created an automation to set charging power and SoC based on sun availability. not to charge the batteries max if there will be generation from solar.

no mater what i did on solarman, the inverter wouldn't go into charging the batteries. a while ago i came across registers: 206:207
but these are even more complicated for me to solve.

<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>
Run function enable | FuncEn | 206-207 | U32 | 4 | Run function enable
-- | -- | -- | -- | -- | --

</body>
</html> 
<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>
B0:switch machine
--
B1:Soft start enable
B2:Power soft
B3:EPS Mode enable
B4:Timing AC charge enable
B5:Timing  charge enable
B6:Timing discharge enable
B7:JET simulating test
B8:DRMS enable
B9:ripple control
B10: Grid high voltage drop
B11: Grid Low Voltage Rise
B12: Grid over-frequency and load reduction
B13: Grid under-frequency load increase
B14:HVRT
B15:LVRT
B16:Timing Forbid discharge enable
B17: other17
B18:other18
B19:other19
B20:DisselGenFunction
B21:DieselWorkMode
B22:other22
B23:EcoEn
B24~B31 Reseved

</body>
</html>

neofral, 2024-11-08:
can you please help with formatting switches for all of these options?

neofral, 2024-11-08:
been reading for hours about parsing time from modbus rtu... 
i found this regarding date:
https://stackoverflow.com/questions/20018623/hex-decimal-to-date-and-time-from-modbus

any hints on direction to solution?

davidrapan, 2024-11-09:
> can you please help with formatting switches for all of these options?

What do you mean? you just do `platform: select` do the `lookup` table using `bit: {index}` instead of `key`.

neofral, 2024-11-09:
in this case to have an on/off switch for every option i need to configure a different switch for all of them?
sorry but i'm just a hobby programmer, i need an example to see the syntax and then adopt.

neofral, 2024-11-09:
i found this, but as i said before - is way above my abilities...

**Reading individual bits from a 16-bit register**
Some manufacturers use 16-bit registers to store individual boolean values (bits), so with a single read from a single address, 16 booleans could be retrieved. This is sometimes called a flag register.

You need to read the register as an integer, and then extract the bit you are interested in. For example to extract the third bit from right:

registervalue = instrument.read_register(4143)
is_my_bit_set = (registervalue & 0b0000000000000100) > 0
or if using hexadecimal numbers in your code instead:

is_my_bit_set = (registervalue & 0x0004) > 0
More information on bit manipulation in Python, see the “Single bits” section of https://wiki.python.org/moin/BitManipulation

davidrapan, 2024-11-09:
You could do it using switches, yes.

But if you want maybe I could try to extend select platform to be able to set multiple bits (if you really need that).

neofral, 2024-11-09:
> You could do it using switches, yes. (Switch platform do support set of individual bits)
> 
> But if you want maybe I could try to extend select platform to be able to set multiple bits. (If you really need that)

yes, please.
i need functions under registers 206-207.
especially these functions:
B4:Timing AC charge enable
B5:Timing charge enable
B6:Timing discharge enable

and also for me is a must have the timing.
the issue where we don't know how the time is coded.
i have written to the chinese also, let's see what they come up with as answer.

i use these functions to control the charging/discharging based on weather forecast.
how much and when to charge the batteries from low/night tariff.
based on cloud/sun percentage we set charging between 50 to 100%.
it contributes to fully utilizing sun energy, when there is no sun - we need the batteries fully charged at night to decrease electricity cost

neofral, 2024-11-10:
got a reply from the tech support.

he said to parse the 2 bytes separately.
if 1458 is parsed together will give 3642, separately should give the right values
14 58=OE 3A

davidrapan, 2024-11-10:
Okay I will try to implement it.

neofral, 2024-11-17:
@davidrapan how is the implementation coming along?

davidrapan, 2024-11-18:
Time parsing

[feat: Time platform dec/hex parsing/displaying](https://github.com/davidrapan/ha-solarman/commit/da6687cff0214e54fa27c368a3f76a03ac52528e)

Add to the entity description:
```
dec: 0x100
```

davidrapan, 2024-11-18:
> can you please provide an example of the code/syntax?

It will look exactly the same as in:

https://github.com/davidrapan/ha-solarman/blob/60a95b402ed59eb82eca1acd024d9c63084a014f/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml#L1801

And you only add:
```
platform: select
```

neofral, 2024-11-18:
> Time parsing
> 
> [feat: Time platform dec/hex parsing/displaying](https://github.com/davidrapan/ha-solarman/commit/da6687cff0214e54fa27c368a3f76a03ac52528e)
> 
> Add to the entity description:
> 
> ```
> dec: 0x100
> ```

Thank you for your work.
however, still not reading the time correctly.

this is the code:
      - name: AC Charge Time1 End
        platform: time
        rule: 9
        dec: 0x100
        registers: [2510]
        code:
          read: 0x03

time is: 05:30
it reads as: 13:10

davidrapan, 2024-11-19:
And did you updated the code (manually cause there is no new version yet) and restarted HA?

neofral, 2024-11-19:
nope, sorry - didn't restart.
restarted, time reading correctly.
not writing. response in logs

[home-assistant_2024-11-19T08-52-24.172Z.log](https://github.com/user-attachments/files/17812310/home-assistant_2024-11-19T08-52-24.172Z.log)

davidrapan, 2024-11-19:
And what time you were trying to set?

davidrapan, 2024-11-19:
You can try latest commit.

neofral, 2024-11-19:
> You can try latest commit.

damn, it worked!!!
your Brilliant!!!
thank you so much

the only thing remaining is to set the 206, 207 registers and working modes.
i posted a code above, please correct it however you see fit.

after this, the file will be complete with all functions and i will share all the sensors and configuration so you can include it in the repository

[55 more replies]