davidrapan/ha-solarman #253: Wrong data battery charge On Deye Ibrid inverter

wifi75, 2024-11-05:
### Description

Hello, for several days I have encountered a problem in the battery charge data readings.
I have a 15.3 Kwh 300Ah battery, this morning it was completely flat, with a (20% residual about 3.2Kwh). As soon as the production of my PV began it started charging. In home assistant it shows the current charge and it is about 6.2 Kwh. Incorrect value, because connecting directly to the BMS it is almost charged about 285Ah.

see last scrernshot, only the sensor sensor.solar_battery It seems to give a true value

### Reproduction Steps

[image]
[image]

### Log

[home-assistant_solarman_2024-11-05T10-46-16.064Z.log](https://github.com/user-attachments/files/17631086/home-assistant_solarman_2024-11-05T10-46-16.064Z.log)

### Version

v24.11.02

davidrapan, 2024-11-05:
But does the Total Battery Charge work?

wifi75, 2024-11-05:
I'll wait for the charge to finish and then I'll show you the values..

wifi75, 2024-11-05:
Ok Davide, sorry, but after your last update the problem seems to have gone away!
the values ​​all seem to be correct

davidrapan, 2024-11-05:
Sure no worries.

closed 2024-11-05 by davidrapan