davidrapan/ha-solarman #262: Grid peak shaving value problem

terzo33, 2024-11-14:
Hi, when i change grid peak shaving value, give me an error

[image]

I'm using logger LAN adapter

davidrapan, 2024-11-14:
Hi, can you try the change with debug logging enabled? Thanks.

terzo33, 2024-11-14:
> Hi, can you try the change with debug logging enabled? Thanks.

yes, I'm starting to record now

terzo33, 2024-11-14:
> Hi, can you try the change with debug logging enabled? Thanks.

@davidrapan see log... thanks

[image]

those spikes that go to 0 are due to the integration stopping working for 5 seconds and then working again

terzo33, 2024-11-15:
@davidrapan 12h log

[image]

davidrapan, 2024-11-15:
Hi @githubDante, could you take a look at it?

davidrapan, 2024-11-15:
@terzo33 can you please try to set the value using built-in ACTIONS in Developer tools. Use Modbus function code 0x06.

githubDante, 2024-11-15:
@terzo33  can you confirm that the inverter is working according to the selected value despite the error which you see at the bottom of the screen

terzo33, 2024-11-15:
> @terzo33  can you confirm that the inverter is working according to the selected value despite the error which you see at the bottom of the screen 

Yes confirm... works but I see error a the bottom of the screen. Need to fix it

davidrapan, 2024-11-15:
So it does change the value in your inverter? Then ignore the error for now and I'll try to do something about it later. 😉 

Thanks @githubDante!

terzo33, 2024-11-15:
@davidrapan when I set the value, I get the error, however after 10/20 seconds the value is set correctly on the inverter. Before with the wifi logger, everything was immediate. Also every now, the solarman integration stops working for a few seconds using the LAN logger.

davidrapan, 2024-11-15:
> when I set the value, I get the error, however after 10/20 seconds the value is set correctly on the inverter.

The writing into inverter works correctly. The error you see is from parsing of the response from inverter. Nothing serious.

> Also every now, the solarman integration stops working for a few seconds using the LAN logger.

This is worse... Maybe slight adjustments to the timings of the requesting flow could improve it so you don't see disconnects.

terzo33, 2024-11-15:
@davidrapan I hope you can fix it

davidrapan, 2024-11-15:
In const.py you can try to change rows:
```
TIMINGS_UPDATE_TIMEOUT = TIMINGS_INTERVAL * 6
TIMINGS_SOCKET_TIMEOUT = TIMINGS_INTERVAL * 4 - 1
```

To:
```
TIMINGS_UPDATE_TIMEOUT = TIMINGS_INTERVAL * 8
TIMINGS_SOCKET_TIMEOUT = TIMINGS_INTERVAL * 6 - 1
```

terzo33, 2024-11-15:
@davidrapan done

davidrapan, 2024-11-15:
Don't forget to restart HA to apply changes and let me know if it had any effect on the disconnects.

terzo33, 2024-11-16:
> Don't forget to restart HA to apply changes and let me know if it had any effect on the disconnects.

@davidrapan done but it's the same

[image]

terzo33, 2024-11-18:
> Well in that case I don't think we can do anything about it in integration... but it's really strange that it goes to 0W...

@davidrapan I checked better and there is no connection problem but simply the parameters all go to 0, then after a few minutes they return to the correct values. there will surely be a bug

[image]

davidrapan, 2024-11-18:
And all the sensors goes to 0 too?

terzo33, 2024-11-18:
> And all the sensors goes to 0 too?

@davidrapan only program time, program power, program soc and program voltage

davidrapan, 2024-11-18:
So the sensors shows what? Value or Unavailable?

terzo33, 2024-11-19:
> So the sensors shows what? Value or Unavailable?

@davidrapan show 0 values

[image]

davidrapan, 2024-11-19:
Those are not sensors.

terzo33, 2024-11-19:
> Those are not sensors.

Sorry, all sensors works fine, show alwasys correct value

davidrapan, 2024-11-20:
I have honestly no idea why is it happening. It would make sense if also the sensor values did become Unavailable during periods with zeroed configurables...

You switched from Wifi Logger to Ethernet Logger?

terzo33, 2024-12-05:
Hi @davidrapan , after upgrade solarman integration I see always this error

[image]

davidrapan, 2024-12-05:
What does the log say?

terzo33, 2024-12-05:
> What does the log say?
[image]

I see some error for "Buffer of 4 bytes error" and then I see on dashboard all editabile value go to 0 already :(

[image]

davidrapan, 2024-12-05:
No I mean i want to see the debug log during changing the value. 😉

terzo33, 2024-12-05:
> No I mean i want to see the debug log during changing the value. 😉

I sent you

[image]

davidrapan, 2024-12-05:
> I sent you

Ah my bad, did not noticed that.

davidrapan, 2024-12-05:
Do you have this issue with all of the number configurables?

terzo33, 2024-12-05:
> Do you have this issue with all of the number configurables?

@davidrapan yes :(

davidrapan, 2024-12-05:
I'm not sure but it could be specific to Ethernet Loggers... :-/ Cause I'm not able to replicate it on my device.

Are you familiar with python? Cause you proly gonna have to debug this on your installation.

githubDante, 2024-12-05:
I think this is not needed at all. The request is correct, but the response from the datalogger after write is truncated. 

Here an example:

```
2024-12-05 09:47:40.406 DEBUG (MainThread) [custom_components.solarman.api] [2110011481] SENT: 3e 21 00 00 00 09 01 10 01 25 00 01 02 0a f0
2024-12-05 09:47:40.670 DEBUG (MainThread) [custom_components.solarman.api] [2110011481] RECD: 3e 21 00 00 00 04 01 10 01 25
```

```python
>>> pdu = bytes.fromhex('10 01 25 00 01 02 0a f0')
>>> func = create_function_from_request_pdu(pdu)
>>> func
<umodbus.functions.WriteMultipleRegisters object at 0x7349e5708e80>
>>> func.starting_address
293
>>> func.values
[2800]
>>> func.create_response_pdu().hex(' ')
'10 01 25 00 01'
```
As you can see two bytes are missing from the end.

terzo33, 2024-12-05:
yes indeed solarman gives the error but then it sets the value correctly

davidrapan, 2024-12-05:
Ah I did not notice that! I immediately assumed the error was in my code. 😆 Thanks!

terzo33, 2024-12-05:
> Ah I did not notice that! I immediately assumed the error was in my code. 😆 Thanks!

Awesome. Could you fix for next release?

davidrapan, 2024-12-05:
I totally forgot about and thought it's new. Guess I'm overworked. 😆

davidrapan, 2024-12-05:
Do we have debug log with that happening?

githubDante, 2024-12-05:
> So anyway, I was right and it's specific to Ethernet Loggers.

I wouldn't count on that. Can be a firmware bug :smiley: 

@terzo33 sorry but no, I don't use HA at all.

davidrapan, 2024-12-05:
> I wouldn't count on that. Can be a firmware bug 😃

Sure, what I mean is we are not talking solarman protocol (it's in a different league all together and adds bytes instead, haha) here.

davidrapan, 2024-12-05:
So this is how the reply should look like right?
```
8e 39 00 00 00 06 01 10 01 25 00 01
```

githubDante, 2024-12-05:
Yes, that's the correct response. Here a small patch which should fix this

```patch
diff --git a/custom_components/solarman/api.py b/custom_components/solarman/api.py
index d59adaf..1ad82f5 100644
--- a/custom_components/solarman/api.py
+++ b/custom_components/solarman/api.py
@@ -101,7 +101,11 @@ class PySolarmanV5AsyncWrapper(PySolarmanV5Async):
     async def write_multiple_holding_registers(self, register_addr, values):
         if not self._passthrough:
             return await super().write_multiple_holding_registers(register_addr, values)
-        return await self._tcp_parse_response_adu(write_multiple_registers(self.mb_slave_id, register_addr, values))
+        try:
+            r = await self._tcp_parse_response_adu(write_multiple_registers(self.mb_slave_id, register_addr, values))
+        except struct.error:
+            r = len(values)
+        return r
 
 class Inverter(PySolarmanV5AsyncWrapper):
     def __init__(self, address, serial, port, mb_slave_id):
```
Don't know why I can't attach it...

davidrapan, 2024-12-05:
I made [fix: Workaround for modbus tcp responses w/o quantity](https://github.com/davidrapan/ha-solarman/commit/e68423ebdc2c822ab20325dd607a3f237127fbdd), try it and if it fails I'll got w/ @githubDante's suggestion. 😉

Anyway, really thanks, Modbus would be sometimes complete madness for me without your help. 😆

davidrapan, 2024-12-06:
Hi @terzo33, please try latest version and let me know if that bug of printing errors is gone. 😉

terzo33, 2024-12-06:
> Hi @terzo33, please try latest version and let me know if that bug of printing errors is gone. 😉

@davidrapan yes works 😍😍 but in grid power i see something wrong 😂😂

[image]

davidrapan, 2024-12-06:
Please, can you tell me your exact device and model?

terzo33, 2024-12-06:
> Please, can you tell me your exact device and model?

Deye Hybrid 6kW

davidrapan, 2024-12-06:
I need the model number exactly.

terzo33, 2024-12-06:
> I need the model number exactly.
> 
> Edit1: It's something like SUN-12K-SG04LP3 (this is mine)
> 
> Edit2: And can you show me the same screshoots like in https://github.com/davidrapan/ha-solarman/issues/281#issuecomment-2520017035 togerther with screeshots of whole sensor list?

SUN-6K-SG03LP1-EU

[image]

davidrapan, 2024-12-06:
And screenshot of Grid and Load sensors? (Voltage, current, power)

davidrapan, 2024-12-06:
Mhmmm, gonna have to find out how to detect which device needs scaling * 10 for Grid and Load power and which do not...

Thanks!

terzo33, 2024-12-06:
> Mhmmm, gonna have to find out how to detect which device needs scaling * 10 for Grid and Load power and which do not...
> 
> Thanks!

I'm glad to help. If can help you. before the update the value was correct

davidrapan, 2024-12-06:
I know, it was requested change because some devices do need it multiplied by 10.