davidrapan/ha-solarman #263: DEYE SUN-8K-G05 yaml file

folny, 2024-11-15:
Hi I have a deye sun-8k-g05 inverter and I can't read the power on PV1 and PV2, what are the addresses to read the power ?. Does anyone have a yaml file for this inverter where all the parameters can be read ?.

davidrapan, 2024-11-15:
Hi @folny, so any of the already present `deye_*` profiles does not work?

davidrapan, 2024-11-15:
Enable debug logging and attach the file here.

folny, 2024-11-15:
Okay, I'll add the debug file tomorrow.

folny, 2024-11-16:
Hi

So I played around a bit today and it seems the data is now correct but I would like to know how I can read the values ​​from the smart meter connected to the inverter so I can find out the household consumption and the energy that goes from the grid and into the grid.
[image]

davidrapan, 2024-11-17:
Try it with `deye_hybrid.yaml` profile then. String and Hybrid profiles are essentially the same but the String has fewer registers read.

folny, 2024-11-18:
I have already tried this option, but the data I need is incorrect.

[image]

davidrapan, 2024-11-18:
What about 203, 204?

folny, 2024-11-18:
Yes, it should be those registers, but I didn't have time to try it today, I'll try it tomorrow and let you know.

folny, 2024-11-19:
So today I added more registers to the file and it seems that everything is working as it should. So I'm adding my file here and I hope it helps someone in the future.

[deye_sun_g05_power_flow_card.zip](https://github.com/user-attachments/files/17818607/deye_sun_g05_power_flow_card.zip)

[image]

davidrapan, 2024-11-20:
I think that Load Power should be unsigned.

[feat: Load & Grid Power for Deye String](https://github.com/davidrapan/ha-solarman/commit/12c5d45704764a802a9d6c8a012e6ad28bc760dd)

folny, 2024-11-22:
There are more irrelevant values ​​but I left them there.

closed 2024-12-17 by davidrapan