davidrapan/ha-solarman #265: Support of Deye SG05LP1

BoberMod, 2024-11-22:
### Is your feature request related to a problem?

I have Deye SUN-6K-SG05LP1-EU 6KW inverter. Currently, it's not listed in [supported devices](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices).

### Describe the solution you'd like

I want to try the `deye_hybrid.yaml` config. As I understand it, it's a generic profile and can work with any Deye inverter. But according to [message in wiki](https://github.com/davidrapan/ha-solarman/wiki#-changes), I can't just choose `deye_hybrid.yaml` because if it's wrong, it will overwrite my settings:

> Way more dangerous is when you load wrong profile (e.g. deye_hybrid.yaml instead of deye_sg04lp3.yaml). It will mess up with your values...

Is there a read-only mode to try?

### Describe alternatives you've considered

I considered the [original implementation](https://github.com/StephanJoubert/home_assistant_solarman) of this integration by StephanJoubert.

davidrapan, 2024-11-22:
Hi @BoberMod, this warning is for existing users to be careful when they (for any reason) delete and re-adding back their device. So they don't add it with incorrect profile which could populate existing history with nonsensical data.

If you as a new user do add device with profile which works but display incorrect data you can just delete it and remove history of the sensors using Developer tools -> STATISTICS -> Fix Issue on all entities which will delete it.

List with supported devices contains those which are confirmed from users and I'm sure that the integration works for many more which are not listed.

davidrapan, 2024-11-25:
Can you show me reading using built-in ACTION in Developer tools of register number 274?

BoberMod, 2024-11-27:
Action: `Solarman: Read Holding Registers (Modbus Function Code 3)`\
Register: 274\
Response:
```
"274": 5
```

davidrapan, 2024-11-29:
* [feat: Select platform control specific bits using mask](https://github.com/davidrapan/ha-solarman/commit/15cde65905f970501a6f93df9d418ee9ec85d6a3)
* [fix: Select platform control specific bits using mask](https://github.com/davidrapan/ha-solarman/commit/d1ac39028fcfd5f7a535a882833341b069426ec7)

Those commits should fix it. 😉

davidrapan, 2024-12-05:
Is it fixed in the latest release?

closed 2025-01-20 by davidrapan