davidrapan/ha-solarman #266: Error when adding integration

rp2019433, 2024-11-23:
### Description

After installing ha-solarman through HACS, adding integration brings this error:

[image]

### Reproduction Steps

1.Had [StephanJoubert/home_assistant_solarman](https://github.com/StephanJoubert/home_assistant_solarman) installed previously. 
2. Removed [StephanJoubert/home_assistant_solarman](https://github.com/StephanJoubert/home_assistant_solarman) installation.
3. Installed https://github.com/davidrapan/ha-solarman.

### Log

[image]

### Version

v24.11.02

CrazyUs3r, 2024-11-23:
ssh into your ha and run: ha core rebuild

rp2019433, 2024-11-23:
> ssh into your ha and run: ha core rebuild

Hi. Did it but the error persist. Tried to remove, reinstall and rebuild as well, but it didn't work.

```
Logger: homeassistant.config_entries
Source: config_entries.py:3249
First occurred: 4:50:26 PM (1 occurrences)
Last logged: 4:50:26 PM

Error occurred loading flow for integration solarman: No module named 'pysolarmanv5'
```

davidrapan, 2024-11-25:
It's a problem with HA OS or core, don't remember which one exactly, upgrade or downgrade should fix that.

rp2019433, 2024-11-25:
> It's a problem with HA OS or core, don't remember which one exactly, upgrade or downgrade should fix that.

Hi. I've tested in two HA instances, both with same software version as below:

```
Core 2024.11.3
Supervisor 2024.11.4
Operating System 13.2
Front-end 20241106.2
```

One of the instances had Solarman StephanJoubert installed, and  is the one that is experiencing problem with "Config flow could not be loaded".

The other one, fresh install, ha-solarman was installed correctly.

davidrapan, 2024-11-25:
Yes, I told you it's bug in HA and the act of downgrade/upgrade will fix it.

rp2019433, 2024-11-25:
> Yes, I told you it's bug in HA and the act of downgrade/upgrade will fix it.

Got it now. Thanks.

`ha core update --version XXX` to downgrade and later upgraded.

closed 2024-11-25 by rp2019433