davidrapan/ha-solarman #267: I am using Imars MG5KTL inverter with monitoring on solarman, please help me put this device on home assistant, thank you

thaibacgiang, 2024-11-24:
### Description

I am using Imars MG5KTL inverter with monitoring on solarman, please help me put this device on home assistant, thank you

### Reproduction Steps

I am using Imars MG5KTL inverter with monitoring on solarman, please help me put this device on home assistant, thank you

### Log

I am using Imars MG5KTL inverter with monitoring on solarman, please help me put this device on home assistant, thank you

### Version

v24.11.2

davidrapan, 2024-11-25:
Try already present profile for different INVT inverter and if it does not work you have to contact your manufacturer to give you modbus documentation for your model.

thaibacgiang, 2024-11-28:
> Try already present profile for different INVT inverter and if it does not work you have to contact your manufacturer to give you modbus documentation for your model.

I contacted the provider and they sent me the API but I don't know how to use it. Can you help me?

thaibacgiang, 2024-11-28:
> I don't know what do you mean by "API".

means the manufacturer sent me the API

davidrapan, 2024-11-28:
For what? Be specific.

davidrapan, 2024-11-28:
This integration focus is on direct communication with the inverter via modbus using solarman protocol (cloud-free).

thaibacgiang, 2024-11-28:
> This integration focus is on direct communication with the inverter via modbus using solarman protocol (cloud-free).

Yes! Now I want to connect the inverter to the hass for convenient monitoring and real-time updates.

davidrapan, 2024-11-28:
I'm afraid we don't understand each other.

Anyway, maybe it's worth the shot to just try install this integration and configure device w/ `invt_xd-tl.yaml` profile. Maybe your device uses the same register addresses. 😉

davidrapan, 2024-11-28:
It's not yet part of released version. You can manually copy the file from repository to appropriate folder of your installation.

brainstormi, 2024-11-29:
Testing invt_xd-tl integration for this model (MG5TKL-2M and stick LSW-3) doesn't seems to work due to some wrong modbus addresses, unfortunately I was not able to find in Internet any documentation regarding this model.  Here it's the log

2024-11-29 12:16:48.483 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-11-29 12:16:48.483 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 10.69.1.254
2024-11-29 12:16:48.483 DEBUG (MainThread) [custom_components.solarman.discovery] _discover
2024-11-29 12:16:48.485 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [10.69.1.254, E8FDF849C3B0, 888888888]
2024-11-29 12:16:48.523 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 3, min_span: -1, max_size: 125, digits: 6
2024-11-29 12:16:48.523 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', 'XX:XX:XX:XX:XX:XX')}, 'identifiers': {('solarman', 888888888)}, 'serial_number': 888888888, 'manufacturer': 'Invt', 'model': 'XD-TL', 'name': 'Inverter'}
2024-11-29 12:16:48.523 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-11-29 12:16:48.523 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Scheduling 4 query requests. #0
2024-11-29 12:16:48.523 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 ...
2024-11-29 12:16:48.523 INFO (MainThread) [custom_components.solarman.api] [888888888] Connecting to 10.69.1.254:8899
2024-11-29 12:16:48.525 DEBUG (MainThread) [custom_components.solarman.api] [888888888] SENT: a5 17 00 10 45 f4 00 91 13 18 a3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 38 06 00 34 a9 7c 5c 15
2024-11-29 12:16:48.579 DEBUG (MainThread) [custom_components.solarman.api] [888888888] RECD: a5 13 00 10 15 f4 95 91 13 18 a3 02 01 be 50 44 35 d3 0f 00 00 e1 51 05 32 01 83 02 c0 f1 2c 15
2024-11-29 12:16:48.579 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 failed, attempts left: 4 [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
2024-11-29 12:16:48.780 DEBUG (MainThread) [custom_components.solarman.api] [888888888] SENT: a5 17 00 10 45 f5 00 91 13 18 a3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 38 06 00 34 a9 7c 5d 15
2024-11-29 12:16:48.829 DEBUG (MainThread) [custom_components.solarman.api] [888888888] RECD: a5 13 00 10 15 f5 96 91 13 18 a3 02 01 be 50 44 35 d3 0f 00 00 e1 51 05 32 01 83 02 c0 f1 2e 15
2024-11-29 12:16:48.829 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 failed, attempts left: 3 [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
2024-11-29 12:16:49.231 DEBUG (MainThread) [custom_components.solarman.api] [888888888] SENT: a5 17 00 10 45 f6 00 91 13 18 a3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 38 06 00 34 a9 7c 5e 15
2024-11-29 12:16:49.279 DEBUG (MainThread) [custom_components.solarman.api] [888888888] RECD: a5 13 00 10 15 f6 97 91 13 18 a3 02 01 bf 50 44 35 d3 0f 00 00 e1 51 05 32 01 83 02 c0 f1 31 15
2024-11-29 12:16:49.279 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 failed, attempts left: 2 [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
2024-11-29 12:16:49.880 DEBUG (MainThread) [custom_components.solarman.api] [888888888] SENT: a5 17 00 10 45 f7 00 91 13 18 a3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 38 06 00 34 a9 7c 5f 15
2024-11-29 12:16:49.929 DEBUG (MainThread) [custom_components.solarman.api] [888888888] RECD: a5 13 00 10 15 f7 98 91 13 18 a3 02 01 bf 50 44 35 d4 0f 00 00 e1 51 05 32 01 83 02 c0 f1 34 15
2024-11-29 12:16:49.929 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 failed, attempts left: 1 [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
2024-11-29 12:16:50.730 DEBUG (MainThread) [custom_components.solarman.api] [888888888] SENT: a5 17 00 10 45 f8 00 91 13 18 a3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 38 06 00 34 a9 7c 60 15
2024-11-29 12:16:50.779 DEBUG (MainThread) [custom_components.solarman.api] [888888888] RECD: a5 13 00 10 15 f8 99 91 13 18 a3 02 01 c0 50 44 35 d5 0f 00 00 e1 51 05 32 01 83 02 c0 f1 38 15
2024-11-29 12:16:50.779 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Querying 03 ~ 14342 - 14393 | 0x3806 - 0x3839 # 052 failed, attempts left: 0, aborting. [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
2024-11-29 12:16:50.779 DEBUG (MainThread) [custom_components.solarman.api] [888888888] Fetching failed. [Previous State: Disconnected (-1)]
2024-11-29 12:16:50.779 INFO (MainThread) [custom_components.solarman.api] [888888888] Disconnecting from 10.69.1.254:8899
2024-11-29 12:16:50.780 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 2.257 seconds (success: False)
2024-11-29 12:16:50.780 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown

davidrapan, 2024-11-29:
This is unfortunate, but maybe it would be worth trying some of the other profiles too?

**Try to contact iNVT and ask them for modbus rtu documentation.**

davidrapan, 2024-11-29:
Yes, too many requests in really short time and the logger will stop responding for a moment...

brainstormi, 2024-12-06:
A little update just in case it's useful for somebody. Vendor obviously didn't reply me, but I found in Internet some images (no text format, sorry) about reg addresses for INVT IMars models. I tried to check some of these regs with pysolarmanv5 and my Inverters randomly, and seems to work for the few that I tried and could make sense to me. I attached here for documentation purposes:
 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

davidrapan, 2024-12-07:
I would start with voltage, current and power values as those containing info don't have to be that straight forward. 😉

brainstormi, 2024-12-13:
Those values are properly defined and working in the doc attached in my previos post:
voltage -> U phase voltage (0x1698)
current -> U phase current (0x1699)
power  -> Input Power (0x16A2... 2 regs, only 0x16A3 fulfilled)
                 Output Power (0x16A4... 2 regs, only 0x16A4 fulfilled)

davidrapan, 2024-12-13:
It's not about which reg are populated when it uses 2 but which are higher and lower bytes. 😉 Those "unpopulated" are higher bytes and will be used w/ number > 32767.

mnadalc, 2025-01-07:
Hey 👋 

I have an [**INVT BD6KTL-RL1**](https://www.enfsolar.com/pv/inverter-datasheet/13969) inverter with a [Stick Data Logger LSW-3](https://d2l38nissjun1p.cloudfront.net/wp-content/uploads/2022/10/Stick-Logger-WiFi-LSW-3.pdf) where unfortunately the [invt_xd-tl.yaml](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/invt_xd-tl.yaml) does not work (I get the same issue as @brainstormi ,
```[IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]```
).

But with the register addresses from https://github.com/davidrapan/ha-solarman/issues/267#issuecomment-2522709013 and creating a custom yaml file, I am able to log data, although it seems I won't be able to show as much information as using a **Stick Data Logger LSW5** like shown at [Modbuts.RTU.Protocol.3.pdf](https://github.com/user-attachments/files/17279845/INVT.XG.XD.Inverter.Modbus.RTU.Protocol.3.pdf) from this post https://github.com/davidrapan/ha-solarman/issues/197#issuecomment-2397055456

The funny thing is that I have contacted another person who has the same **INVT BD6KTL-RL1** inverter, but uses a [Stick Data Logger LSW5](https://d2l38nissjun1p.cloudfront.net/wp-content/uploads/2022/10/Stick-Logger-WiFi-LSW-5.pdf ) and he can use the `invt_xd-tl.yaml` configuration without any issue, works out of the box.

@brainstormi Do you remember where did you get all screenshots from? I'm unable to find any more information.

I'll attach as CSV all information from the pictures in case anyone wants to copy/paste something.

[INVT_LSW3_Parameters_Address.csv](https://github.com/user-attachments/files/18331339/INVT_LSW3_Parameters_Address.csv)

davidrapan, 2025-02-09:
Hi @brainstormi and @mnadalc,

what about https://community.home-assistant.io/t/ha-integration-invt-xg50ktr-lsw5/681962/3 thread? Does it contain any new information?

thaibacgiang, 2025-02-12:
> Hi [@brainstormi](https://github.com/brainstormi) and [@mnadalc](https://github.com/mnadalc),
> 
> what about https://community.home-assistant.io/t/ha-integration-invt-xg50ktr-lsw5/681962/3 thread? Does it contain any new information?

Is there any way to refresh the 10s inverter data? Please help me. Thanks a lot.

thaibacgiang, 2025-02-12:
> I'm not sure what are you talking about?

i have successfully configured solarman intergrate but all sensors it does not update fast automatically although i have installed in yalm. i want it to update continuously every 10s/time

davidrapan, 2025-02-12:
Update interval for INVT profile is set to 5s.

davidrapan, 2025-02-13:
That makes no sense… enable debug logging and attach the file here.

thaibacgiang, 2025-02-14:
> A little update just in case it's useful for somebody. Vendor obviously didn't reply me, but I found in Internet some images (no text format, sorry) about reg addresses for INVT IMars models like this one. I tried to check some of these regs with pysolarmanv5 and my Inverter randomly, and seems to work for the few that I tried and could make sense to me. I attached here for documentation purposes:
> 
> [image] [image] [image] [image] [image] [image] [image]

Thank you! I tried with the register codes it got 2 registers right. Do you have the complete yaml file that you can send me? Thank you very much!

thaibacgiang, 2025-02-15:
> That makes no sense… enable debug logging and attach the file here.

[home-assistant_solarman_2025-02-15T01-45-58.231Z.log](https://github.com/user-attachments/files/18807425/home-assistant_solarman_2025-02-15T01-45-58.231Z.log) 
This is the log, author please check again!

davidrapan, 2025-03-02:
Great, I'll look into it. 😉

davidrapan, 2025-05-08:
Hi @thaibacgiang, upload your `thai.yaml` here. It looks like you have some errors there.